Portisch/RF-Bridge-EFM8BB1 #74: RX\TX seems to stop working after some time.

Pre10der, 2019-01-07:
I am experiencing that the sending of RfRaw and RfCode commands from the Sonoff Bridge stops working after some time. When this happens, the device also stops receiving RF commands. 

The fix is to physically power off the Sonoff and power it on again. The tasmota web-gui is still responsive, however the RF side of the unit seems to freeze\lock up. A restart from the web GUI does not work and requires a physical power off/on again. Is this something that anyone else is experiencing?

I have a Sonoff R2 v1.0 with the current version of the Portisch firmware installed. In my case, I have not physically modified the copper traces on the PCB by cutting them (I opted to use jumper leads to get the firmware on).

haegar33, 2019-01-07:
With V2 and above the RF Chip sometimes hangs when sending B0 codes. If you dont get a answer like
` RESULT = {"RfRaw":{"Data":"AAA055"}}`

or similar in the console after sending a command your (RF)system is frozen. Only help is to power off/on the Bridge

Portisch, 2019-01-08:
Post your B0 codes!

Pre10der, 2019-01-08:
These are what I have calculated. I will be calculating more in coming days. Are these what you need?

Light On, Fan Off:
RfRaw AAB01E0301016802C6265C0010101100101100101010101011001010101101010255

Fan High, Light Off:
RfRaw AAB01E0301015402BC26660010101100101100101010101010101100101100101255

Fan Medium, Light Off:
RfRaw AAB01E0301016802BC268E0010101100101100101010101010110010101100110255

Fan Low, Light Off:
RfRaw AAB01E0301015E02C626700010101100101100101010101010110100101101001255

Turn All Off: 
RfRaw AAB01E0301016802C626340010101100101100101010101010101010101011010255

Fan Low, Light On: 
RfRaw 
AAB01E0301015E02D0262A0010101100101100101010101011010100110011001255

jancolle, 2019-01-08:
I was also seeing the RF part freezing when running the latest master, but after I changed the firmware to version 0x02 everything works without any lockups. The only problem is, with that firmware I cannot sniff rf codes of my rf switches, but then I just load up the newest firmware for that part, and then change back to 0x02 when finished.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch