Portisch/RF-Bridge-EFM8BB1 #75: Failed to decode buckets

mewild, 2019-01-09:
I have used the general buckets branch of software to obtain the following codes via 'rfraw 177'

12:36:16 CMD: rfraw 177
12:36:16 MQT: tasmota/stat/sonoff-rfbridge/RESULT = {"RfRaw":"ON"}
12:36:16 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:36:17 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0276 00DC 008C 13C4 481929292929292929292929292929292929292B192B1B1B1B192A1A1929292B1A1929292929292B192B19292B192B192929 55"}}
12:36:20 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0460 1C7A 0104 0974 3462 481A3A 55"}}
12:36:21 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 1374 0276 00DC 0096 13A6 481929292929292929292929292929292929292B192B1B1A1A192B1A1929292A1A1929292929292A192A19292A192A192929 55"}}
12:36:25 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0208 0118 0064 02B2 381A08 55"}}
12:36:25 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 1374 026C 00D2 00A0 1392 481929292929292929292929292929292929292B192B1B1A1A192B1A1929292B1A1929292929292A192A19292B192A192929 55"}}
12:36:31 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0280 00F0 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
12:36:34 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0280 00F0 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
12:36:37 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 137E 0276 00E6 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
12:36:37 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 137E 026C 00DC 00A0 1388 481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A192929 55"}}

This is from pressing one button multiple times

The result from running this through the bitbucket converter tool is

###### Processing line 12:36:17 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0276 00DC 008C 13C4 481929292929292929292929292929292929292B192B1B1B1B192A1A1929292B1A1929292929292B192B19292B192B192929 55"}}
 ######
Filtered 0xB1 data: AAB105136A027600DC008C13C4481929292929292929292929292929292929292B192B1B1B1B192A1A1929292B1A1929292929292B192B19292B192B19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 136A 0276 00DC 008C 13C4 9481929292929292929292929292929292929292B192B1B1B1B192A1A1929292B1A1929292929292B192B19292B192B19292 55
###### Processing line 12:36:20 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0460 1C7A 0104 0974 3462 481A3A 55"}}
 ######
Filtered 0xB1 data: AAB10504601C7A010409743462481A3A55
Something is wrong with the RF data!
Resulting 0xB0 data: AA B0 0F 05 04 0460 1C7A 0104 0974 3462 481A3A 55
###### Processing line 12:36:21 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 1374 0276 00DC 0096 13A6 481929292929292929292929292929292929292B192B1B1A1A192B1A1929292A1A1929292929292A192A19292A192A192929 55"}}
 ######
Filtered 0xB1 data: AAB1051374027600DC009613A6481929292929292929292929292929292929292B192B1B1A1A192B1A1929292A1A1929292929292A192A19292A192A19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 1374 0276 00DC 0096 13A6 9481929292929292929292929292929292929292B192B1B1A1A192B1A1929292A1A1929292929292A192A19292A192A19292 55
###### Processing line 12:36:25 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0208 0118 0064 02B2 381A08 55"}}
 ######
Filtered 0xB1 data: AAB10402080118006402B2381A0855
Something is wrong with the RF data!
Resulting 0xB0 data: AA B0 0D 04 04 0208 0118 0064 02B2 381A08 55
###### Processing line 12:36:25 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 1374 026C 00D2 00A0 1392 481929292929292929292929292929292929292B192B1B1A1A192B1A1929292B1A1929292929292A192A19292B192A192929 55"}}
 ######
Filtered 0xB1 data: AAB1051374026C00D200A01392481929292929292929292929292929292929292B192B1B1A1A192B1A1929292B1A1929292929292A192A19292B192A19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 1374 026C 00D2 00A0 1392 9481929292929292929292929292929292929292B192B1B1A1A192B1A1929292B1A1929292929292A192A19292B192A19292 55
###### Processing line 12:36:31 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0280 00F0 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
 ######
Filtered 0xB1 data: AAB105136A028000F0009613C4481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 136A 0280 00F0 0096 13C4 9481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292 55
###### Processing line 12:36:34 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 136A 0280 00F0 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
 ######
Filtered 0xB1 data: AAB105136A028000F0009613C4481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 136A 0280 00F0 0096 13C4 9481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292 55
###### Processing line 12:36:37 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 137E 0276 00E6 0096 13C4 481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B192929 55"}}
 ######
Filtered 0xB1 data: AAB105137E027600E6009613C4481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 137E 0276 00E6 0096 13C4 9481929292929292929292929292929292929292B192B1B1B1B192B1B1929292B1B1929292929292B192B19292B192B19292 55
###### Processing line 12:36:37 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 137E 026C 00DC 00A0 1388 481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A192929 55"}}
 ######
Filtered 0xB1 data: AAB105137E026C00DC00A01388481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A19292955
Failed to decode buckets...
Resulting 0xB0 data: AA B0 3E 05 04 137E 026C 00DC 00A0 1388 9481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A19292 55
###### Processing line 12:36:51 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00F0 09EC 0406 3DFE 1388 481A38 55"}}
 ######
Filtered 0xB1 data: AAB10500F009EC04063DFE1388481A3855
Something is wrong with the RF data!
Resulting 0xB0 data: AA B0 0F 05 04 00F0 09EC 0406 3DFE 1388 481A38 55

All of these contain the error failed to decode buckets...

None the less i have tried sending them via rfraw and nothing is happenening.

Any ideas why this is happening?

BrassyPanache, 2019-01-09:
Are you using version 0.3?  I had a similar issue and found that flashing the version provided in the [Tasmota source repository](https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181127.hex) gave me valid output.

mewild, 2019-01-09:
No luck. Using that firmware rfraw 177 shows no codes sniffed. @BrassyPanache

mewild, 2019-01-14:
Using a different version of the converter I get outputs like the following for all the codes posted above

```
Enter B1 line: 12:36:37 MQT: tasmota/tele/sonoff-rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 137E 026C 00DC 00A0 1388 481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A192929 55"}}
AA B1 05 137E 026C 00DC 00A0 1388 481929292929292929292929292929292929292B192A1B1A1A192A1B1929292A1A1929292929292A192B19292A192A192929 55
Repeat: 20
Sync: 94    Data: 81 92 92 92 92 92 92 92 92 92 92 92 92 92 92 92 92 92 92 B1 92 A1 B1 A1 A1 92 A1 B1 92 92 92 A1 A1 92 92 92 92 92 92 A1 92 B1 92 92 A1 92 A1 92 92 
Sync               
Hex: 

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 00A0
SYNC_LOW: bucket 0: 137E
BIT_HIGH_TIME: bucket 2: 00DC
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x1a (73%)
BIT_LOW_TIME: bucket 1: 026C
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x49 (73%)
BIT_COUNT + SYNC_BIT_COUNT: 0x0 (0, SYNC_BIT_COUNT = 0)
Data: 
7F00A0137E00DC1a026C490
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 00A0 137E 00DC 1a 026C 49 0 ' == 0x0B

'AA A8 0B 7F 00A0 137E 00DC 1a 026C 49 0  55'
```

BrassyPanache, 2019-01-14:
Can you link to the version of the script which you used? I'd like to give it a try also.

Does the result work with your Sonoff and the device you're trying to interface with?

mewild, 2019-01-15:
@BrassyPanache 

https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2374903/BitBucketConverter.txt

Unfortunately it still didn't work. I think it is something to do with the Protocol being unknown.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

Portisch, 2019-01-31:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h

mewild, 2019-01-31:
@Portisch I have used the converter tool and have B0 codes that work. Is that wiki page showing me how to save them to buttons on the bridge?

Portisch, 2019-01-31:
No. Offtopic: ask at Tasmota