Portisch/RF-Bridge-EFM8BB1 #82: Cleverio.se remote and sockets - Multiple lines of RfRaw output for each keypress

b1r63r, 2019-01-20:
I have a remote and 3 switching 220V sockets from cleverio (swedish company but products made in china)
When I run rfraw aab155 I get multiple lines of output for each button press.

21:49:39 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 04EC 01FE 0938 38181819281819292929281819292928181819292818181818 55"}}
21:49:39 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 051E 051E 281808 55"}}
21:49:39 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 09CE 04E2 29EA 381828080828082808280828082828080828280828080828280808282808280808280828082808280828082808280828082808280828082828080828082808282808 55"}}
21:49:39 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 03 010E 0500 04F6 280818 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 09CE 04EC 29CC 381828080828082808280828082828080828280828080828280808282808280808280828082808280828082808280828082808280828082828080828082808282808 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 03 010E 04F6 04EC 280818 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 04 0212 1BD0 05DC 29C2 38182A082A082A0A082A0A0A0828282A082A0A0A0A0A082A082A0A082A08282A0A08182A082A082A0A082A0A0A0828282A082A0A0A0A0A082A082A0A082A08282A0A 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 04 058C 0258 019A 0654 381A08181A08 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 03 01B8 062C 1C3E 281908190819090819090908181819081909090909081908190908190818190908 55"}}
21:49:40 MQT: tele/rf433gw/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 0640 0640 28181908 55"}}

This is from one press. Should I be able to make this work by converting only one line of the above? Should I try converting and sending a whole block (line by line)? Or is this some unsupported format/protocol?

b1r63r, 2019-01-23:
I opened the remote and found the following text on the rear of the pcb
50074.01-11
FLF-2015-10-23
1811

Some searching turned out 50074 as popping up for some other brands using a protocol called kangtai.

Tellstick server definition: https://github.com/telldus/tellstick-server/blob/master/rf433/src/rf433/ProtocolKangtai.py
rtl_433 definition: https://github.com/merbanan/rtl_433/blob/a8f438a5fb52f24f854368b581a08aa413bb7654/src/devices/kangtai.c

If this is kangtai then it seems to use a lookup table based encryption scheme. How to attack this one?

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

vegovs, 2020-04-10:
@b1r63r Did you get this to work?

pachacamac, 2020-08-29:
I need to know as well 😃

psych0d0g, 2020-11-25:
Did a sniff with RF-Bridge-EFM8BB1-20190220.hex on the same remote:

On Key Channel A:

```
{"RfRaw":{"Data":"AA B1 03 015E 049C 049C 281818 55"}}
{"RfRaw":{"Data":"AA B1 04 03E8 021C 0BD6 1C34 3819090818181908181909081819090819090818190909090A 55"}}
{"RfRaw":{"Data":"AA B1 04 03E8 0212 0BE0 1C2A 3819090818181909090909081818190819081909090909090A 55"}}
{"RfRaw":{"Data":"AA B1 04 03F2 021C 0BD6 1C2A 3819090818190819090908190909090819081818190909090A 55"}}
{"RfRaw":{"Data":"AA B1 04 03F2 021C 0BD6 1C3E 3819090819081908190819081909081908190818190909090A 55"}}
```

Off Key Channel A:
```
{"RfRaw":{"Data":"AA B1 03 015E 0488 05D2 A09090 55"}}
{"RfRaw":{"Data":"AA B1 03 01D6 041A 0424 2819090908 55"}}
{"RfRaw":{"Data":"AA B1 04 03E8 0212 0BD6 1C34 3819090818181819090818181909081819090908190909090A 55"}}
```

b1r63r, 2020-11-26:
I am actually restarting my home automation project so i have not gotten
around to testing. Other things intervened. Any day now...