Portisch/RF-Bridge-EFM8BB1 #83: Need Help defining Protocols for Atag Kitchen Extractor (afzuigkap) and Generic Doorbell

MRvH88, 2019-01-22:
If been successfully using a Sonoff RF Bridge with Tasmota and Portisch for a while now, however I am only able to send RFraw signals and I am not able to receive these signals without sniffing for B1 signals. Would it be possible to receive these signals without B1 sniffing by defining these in RF_Protocols.h?

I have sniffed and converted the following signals for my Atag Kitchen Extractor and Generic Doorbell, these signals work 100% of the time. I have tried the to use the Wiki to define te protocols, but I kinda get lost. Can somebody assist me with creating these protocols?

**Atag Kitchen Extractor (https://www.atag.nl/collectie/producten/000000000000725480_0001)**

**Kitchen Extractor Plus Button**
AA B1 04 014B 010E 028B 32C8 00220022002200220022002203 55
RfRaw AAB0170414014B010E028B32C80022002200220022002200220355 (20x)

**Kitchen Extractor Minus Button**
AA B1 04 0148 010E 028F 32BE 00220022002200220022020023 55
RfRaw AAB01704140148010E028F32BE0022002200220022002202002355 (20x)

**Kitchen Extractor Light Button**
AA B1 04 0299 02A3 010E 3D2C 22002200220022002200202022002202020023 55
RfRaw AAB01D0408029902A3010E3D2C2200220022002200220020202200220202002355 (8x)

**Kitchen Extractor Timer Button**
AA B1 03 0149 02E5 3D2C 00110011001100110011010100110010110102 55
RfRaw AAB01B0314014902E53D2C0011001100110011001101010011001011010255 (20x)

**Generic 433mhz Doorbell**
AA B1 05 01B4 019A 0096 00FA 1B3A 31303030320230020200320204 55
rfraw 176 AAB019051401B4019A009600FA1B3A3130303032023002020032020455

Pre10der, 2019-01-30:
I too have been trying to find an answer to this. I would like to be able to receive these signals without sniffing for B1 signals so that I can allow my sensors in home assistant to keep their state. I’ll stay tuned.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

MRvH88, 2019-01-31:
Thanks for the reply! Used the latest master and the received B1's are indeed much much better and more consisted!

I used the new master to sniff B1 and converted them B0 for the extractor (no time for the doorbell yet unfortunately).

The new B0 are all working correctly, however I would also like to receive the signals from the remote itself without having to turn sniffing on. Am I correct to say that if I use below codes and define  them in RF_Protocols.h, I would be able to see receive them without sniffing?

I have tried to use the Wiki to define the protocols, but I kinda get lost. Can somebody assist me with creating these protocols?

Up (+):
AA B1 03 015E 028A 32C8 28091809180918091809180918 55
AA B0 15 03 14 015E 028A 32C8 18280918091809180918091809 55

Down (-)
AA B1 03 015E 028A 32C8 28091809180918091809181809 55
AA B0 15 03 14 015E 028A 32C8 09280918091809180918091818 55

Light (I use two different repeats because the light can be dimmed and with repeat 20 it will not turn off):
AA B1 03 015E 028A 32C8 28091809180918091809181818091809090918 55
AA B0 1B 03 0A 015E 028A 32C8 18280918091809180918091818180918090909 55
A B0 1B 03 14 015E 028A 32C8 18280918091809180918091818180918090909 55

Timer:
AA B1 03 015E 028A 32C8 28091809180918091809181818091809091818 55
AA B0 1B 03 14 015E 028A 32C8 18280918091809180918091818180918090918 55

Portisch, 2019-02-01:
This is strange. You have 2 differnent bit counts for the same remote? Like Up/Down is the [HT12A](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/393d535603095307ee43271f2d02abdb8a4a972d/inc/RF_Protocols.h#L137-L146) protocol. The Light/Timer is the same as HT12 but with 18 bit insteed of 12!? This is hard to implement because both protocols would be triggered but when 12 bits are reached all decoding is getting stopped. So you never will get the 18 bits decoded.

I modified the HT12 settings: [Atag branch](https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/Atag)
May you try if it is working with only 12bits. Take a look to wiki commands: 0xA6, 0xA8,...

The default sniffing is like the original. If you implement your remote in Protocols.h you have to activate as minimum 0xA6 sniffing. How to handle this you have to ask at Tasmota forum.

MRvH88, 2019-02-01:
@Portisch Thanks for the work on the Atag branch!! I will try to flash it today to see how it works. Concerning the differences between +/- and light/timer, that might me my mistake (still learning), I changed the bucket times to see if I could find common buckets that would fit for the protocol. The buckets for light and timer are a bit different than the ones for +/- but they are working with the ones I added above. Below is a list of sniffs I did for the four options.

Here is the total list of sniffs:

Up (+):
10:39:41 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0280 32FA 28091809180918091809180918 55”}}
10:39:44 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 029E 32D2 28091809180918091809180918 55”}}
10:39:50 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 0294 32C8 28091809180918091809180918 55"}}
10:39:54 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 0294 32D2 28091809180918091809180918 55"}}
10:39:58 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 028A 32C8 28091809180918091809180918 55"}}
10:40:03 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 0294 32D2 28091809180918091809180918 55”}}

Down (-):
10:40:51 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 0294 32D2 28091809180918091809181809 55"}}
10:40:55 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 029E 32D2 28091809180918091809181809 55"}}
10:40:58 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 0294 32D2 28091809180918091809181809 55"}}
10:41:01 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 028A 32C8 28091809180918091809181809 55"}}
10:41:05 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 0280 32D2 28091809180918091809181809 55"}}
10:41:08 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 0294 32C8 28091809180918091809181809 55”}}

Light:
10:41:32 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02C6 3D4A 28091809180918091809181818091809090918 55”}}
10:41:36 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02D0 3D54 28091809180918091809181818091809090918 55"}}
10:41:40 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02D0 3D4A 28091809180918091809181818091809090918 55"}}
10:41:43 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02D0 3D54 28091809180918091809181818091809090918 55"}}
10:41:46 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 017C 02D0 3D5E 28091809180918091809181818091809090918 55”}}

Timer:
10:42:08 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02D0 3D2C 28091809180918091809181818091809091818 55”}}
10:42:08 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 01A4 02F8 3D36 28090809180918091809181818091809091818 55"}}
10:42:14 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 017C 02C6 3D22 28091809180918091809181818091809091818 55"}}
10:42:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02D0 3D2C 28091809180918091809181818091809091818 55"}}
10:42:24 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02D0 3D22 28091809180918091809181818091809091818 55"}}
10:42:29 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02C6 3D2C 28091809180918091809181818091809091818 55”}}
 
Besides the Atag remote I also sniffed my doorbell and found that I get multiple outputs per ring (I think this is because I can set multiple channels so it does not ring if my neighbors use the same one).

I sniffed and received the following list from which only the first one is working after converting it to B0:

AA B0 15 03 14 00C8 017C 1B26 0928181818180909090909180955
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 017C 1B26 28181818180909090909180909 55"}}

17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 008C 01B8 1B30 38080808081A1A0A0A0A080A08 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 0168 0078 1B26 38180808082909090909180909 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0096 01B8 1B1C 38080808081A1A1A1A1A081A1A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0082 01CC 1B26 38080808081A1A1A1A1A081A0A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0082 01C2 1B1C 38080808081A1A1A0A0A080A0A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 008C 01B8 1B1C 38080808081A1A1A1A18081A1A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0082 01CC 1B1C 38080808081A1A1A1A1A081A1A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0082 01CC 1B1C 38080808081A1A1A1A08080A0A 55"}}
17:12:20 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 008C 01B8 1B1C 38080808081A1A1A1A1A081A1A 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 008C 01C2 1B30 38080808081A1A1A1A1A081A18 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 008C 01B8 1B1C 38080808081A1A0A0A0A080A0A 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0096 01B8 1B26 38080808081A1A0A1A1A081A1A 55"}}
[... cut ...]
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0082 01C2 1B1C 38080808081A1A1A0A0A080A0A 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01C2 1B26 38080808081A1A1A1A1A081A08 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01B8 1B26 38080808081A1A0A0A1A081A1A 55"}}
17:12:21 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0078 01CC 1B1C 38080808081A1A1A1A1A081A1A 55"}}

MRvH88, 2019-02-01:
Used the Atag branch with rfraw AAA655 but nothing is received when using the remote, neither +/- nor light/timer.

MRvH88, 2019-02-01:
@Portisch Many thanks to you! I reversed engineered your RF_protocols.h and added the protocols for both the light/timer and the doorbell and compiled them and can receive all three now. The 0x03 version really is a huge improvement!

Only thing now is that my doorbell comes in twice with a single ring, but I guess that is probably because of the repeat?

Portisch, 2019-02-02:
Good, may you make a pull request!?

MRvH88, 2019-02-06:
@Portisch are the defined protocols still subject to a maximum number as in the previous version or can more protocols be defined with version 0x03?

Portisch, 2019-02-06:
Until the efm8 is running out of memory...