Portisch/RF-Bridge-EFM8BB1 #85: Brilliant Lighting Model #17698

Pre10der, 2019-02-04:
I am running the latest firmware (https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)

I can successfully sniff, convert and send the rfraw commands from Home Assistant to Tasmota and they work well. I would however like to make this stateful by allowing Tasmota to see the button presses on the remote and announce them via MQTT (I can do this on a different brand fan that I have). From what I understand, to do this I need to add a new protocol (?) and that's where I get stuck.

The remotes I am working with are 'Brilliant Lighting' model number '17698'. They are a fan/light conbo remote and each remote has 4 dip switches that are set to match the Fan controller.

The sniffed data on the latest firmware is as follows:
Light Off, Fan High:
21:50:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 2684 28090909090918180909090909090909180909181818 55"}}
21:50:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0140 029E 0064 268E 3809090909091A180909090909090909180909181818 55"}}
21:50:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 2684 28090909090918180909090909090909180909181818 55"}}
21:50:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0104 01C2 0334 2666 0320 481919282828381919 55"}}
21:50:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 268E 38090909090918180909090909090909180929181818 55"}}

Light Off, Fan Med:
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 02B2 2620 28090909090918180909090909090918090918090909 55"}}
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 01C2 0348 2684 38191919191928281919282819191928191928191919 55"}}
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 265C 28090909090918180909090909090918090918090909 55"}}
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0276 2666 28090909090918180919091909090918191918091909 55"}}
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 2670 28090909090918180909090909090918090918090909 55"}}
21:50:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2670 38090919190918180909092909090918292918192909 55"}}
21:50:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 2666 38090909090918180909090909092918190918090909 55"}}
21:50:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 2666 28090909090918180909090909090918090918090909 55"}}
21:50:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 265C 38090909090918180929090929090918290918092929 55"}}
21:50:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0280 2670 28090909090918180909090909190918090918090919 55"}}
21:50:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0280 2670 28090919181918180909091909091918190918090909 55"}}

Light Off, Fan Low
21:50:44 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 02BC 2620 28090909090918180909090909090918180918090918 55"}}
21:50:44 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 01CC 0352 2670 38191919191928281919191919191928281928191928 55"}}
21:50:44 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 2684 28090909090918180909090909090918180918090918 55"}}
21:50:44 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2666 38090909090918180909090909090918182918292918 55"}}
21:50:44 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 029E 265C 28090909090918180909090909090918180918090918 55"}}
21:50:46 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 02A8 2666 28090909090918180909090909090918180918090918 55"}}

Light On, Fan High
21:51:33 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0140 029E 00E6 2634 3809090909091818090909090909180909091A091A19 55"}}
21:51:33 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0122 028A 01D6 00BE 268E 480909290909181B1909192919091829091918191829 55"}}
21:51:33 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 265C 28090909090918180909090909191819090918091809 55"}}
21:51:33 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0294 2666 28090909090918180909090909091809090918091809 55"}}
21:51:34 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 029E 01D6 267A 38090909090918180909290929091809092918091809 55"}}
21:51:34 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 26B6 28090909090918180909090909091809090918091809 55"}}

Light On, Fan Med:
21:52:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 06 012C 0294 01E0 00B4 2698 2828 58090909090918180919291919191B1B19191B1B19194809191919191B18291929191929181819191818092948290909092918180929092929291818090918180909 55"}}
21:52:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 267A 28090909090918180909090909091818090918180909 55"}}
21:52:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 029E 01E0 2666 38090909290918180909090909091818090918180909 55"}}
21:52:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 2666 28090909090918180909090909091818090918180909 55"}}
21:52:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 2666 38090909090918180909090909291818090918180909 55"}}
21:52:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 029E 265C 28090909090918180909090909091818090918180909 55"}}
21:52:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 2666 38090909090918180909092909091818090918180909 55"}}
21:52:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 268E 28090909090918180909090909091818090918180909 55"}}

Light On, Fan Low:
21:53:04 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 02BC 262A 28090909090918180909090909091818180918180918 55"}}
21:53:04 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 01B8 02E4 268E 38192819191928281919191919192828281928281928 55"}}
21:53:07 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 2670 28090909090918180909090909091818180918180918 55"}}
21:53:07 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 032A 01C2 032A 38181809181829 55"}}
21:53:07 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 029E 265C 28090909090918180909090909091818180918180918 55"}}
21:53:07 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 032A 01C2 032A 381829 55"}}
21:53:07 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 0334 01C2 032A 38181809181829 55"}}

All Off:
21:51:08 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02D0 27A6 28090909090918180909090909090909090909181809 55"}}
21:51:08 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 01E0 0352 26F2 38191919191928281919191919191919191919282819 55"}}
21:51:08 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2710 38090909090918182909092909192929290929181809 55"}}
21:51:08 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 26CA 38090909090918180909090909090909090929181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 029E 2666 28090909090918180909090909091909190919181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 265C 28090909090918180909090909090909091909181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 028A 2666 28090909090918180909090909090909090909181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 028A 01C2 2652 38092929192918180929292919292929292929181829 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 267A 28090909090918180909090909090909090909181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 2666 28090909180918180909090909090909090909181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 265C 28090909091918180909090909090909090909181809 55"}}
21:51:09 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 2666 38090909090918180909090929090909090909181809 55"}}
21:51:10 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 2684 28090909090918180909090909090909090909181809 55"}}

Can someone please tell me how to get this to become a natively understood protocol?

Portisch, 2019-02-04:
Check the new branch: https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/f317ecb39d5f3ab4e5f19a0e343361c7aad580bd

Just use the [BitConverter](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Decode-0xB1-sniffed-data) tool to create a new protocol.
[image]
On the graph you can see it is inverted (starting with low bucket) and you can see the 21 bits.

When the new protocol got implemented you can use the [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6) command.

Pre10der, 2019-02-06:
Thanks @Portisch. I flashed the firmware, but unfortunatley I cannot receive data using the OxA6 command with Tasmota. Below are a bunch more sniffs to see if this helps.

I have been trying hard to build my own protocol, but I am getting lost.

```
15:04:25 CMD: rfraw 177
15:04:25 MQT: stat/sonoff/RF_Bridge/RESULT = {"RfRaw":"ON"}
15:04:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

Light On, No Fan
```
15:06:50 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 2684 38090909090918180909292919191809090918191829 55"}}
15:06:50 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 267A 38090909090918182909292909091809092918091829 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 029E 01D6 2684 38090909090918180909092929291829090918291829 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 01C2 0352 2698 38191919191928281909191919192819191928192819 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 267A 28090909091918180909090909091819090918091809 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0280 267A 28090909090918180909090909091809190918091809 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2666 38090919090918180929090929091819290918291829 55"}}
15:06:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01E0 2670 38090909090918180929290909091809290918091829 55"}}
15:06:52 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 2666 28090909090918180909090909091819090918091809 55"}}
15:06:52 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 01C2 0334 2666 38191919191928280919191919192819191928192819 55"}}
15:06:52 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 2666 28090909090918180909090909091819090918091809 55"}}
```

Light On, Fan High
```
15:07:34 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 01F4 0366 0320 38192819281928 55"}}
15:07:34 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 01CC 0352 2698 38191919191928281919191919192819281928192828 55"}}
15:07:34 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0294 2670 28090909090918180909090909091809180918091818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 265C 38090909090918180909090909091809180918291818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 265C 38090909090918180909290909091809180918291818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 029E 2666 28090909090918180909090909091809180918191818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 2652 28090909090918180909090909091809180918091818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2666 38090909090918180929090929291829180918091818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 2666 38090909090918180909292909091829180918291818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 028A 01C2 2666 38090929092918182929292929291829182918291818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 2670 38090909092918180909290909191829182918091818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0280 2666 28090909090918180919191919191819180918091818 55"}}
15:07:35 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 028A 01C2 265C 38090929292918182929291929291829181918291818 55"}}
15:07:36 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 028A 2652 28090909090918180909090909091809180918091818 55"}}
15:07:36 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 028A 01CC 2666 38090929292918182929292929291829182918291818 55"}}
15:07:36 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 265C 28090909090918180909090909091809180918091818 55"}}
```

Light On, Fan Medium
```
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0280 265C 28090909090918180909190909191818090918180909 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0294 01D6 265C 38090909090918180909090929091818090918180929 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 265C 28090909090918180909090909091818090918180909 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0276 2670 28090909091918181909090909191818090918180919 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 2666 28090909090918180909090909091818090918181909 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0280 265C 28090909090918180919090919091818090918180909 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 028A 2666 28090909090918180909090909091818090918180909 55"}}
15:07:51 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 01E0 0334 2666 38190909090928280909090909092828090928281919 55"}}
[... cut ...]
15:10:15 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 029E 2666 28090909090918180909090909090918180918090918 55"}}
15:10:15 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 2666 38090909090918180909092909090918180918290918 55"}}
15:10:15 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 029E 265C 28090909090918180909090909091918180918090918 55"}}
15:10:15 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0294 265C 28090909090918180909090909090918180918090918 55"}}

```

Portisch, 2019-02-11:
I have done a new test version: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/62d3cbd5411cb019492e808b5b4c9d31ebe60384/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Please try this again with 0xA6. If no red LED on the bridge is flashing please report!

Pre10der, 2019-02-24:
Sorry for the late reply. I uploaded the lined firmware, but it did not take. I tried a number of times, but the RF Bridge hung and rebooted after about a minute indicating it was stuck in a loop (00:00:10 MQT: tele/sonoff/RF_Bridge/INFO3 = {"RestartReason":"Blocked Loop"}).

I reverted to the previous HEX and it is back working again so it seems like the new firmware is still not right.

Thanks!

Portisch, 2019-02-24:
As this thread was inactive the branch didn't get updated. I will do tomorrow. Use the latest hex file from the master branch and do 0xB1 sniffing again please.

Pre10der, 2019-02-24:
For some reason, I cannot get the master branch firmware to apply from the Tasmota interface. I get the same boot loop issue: (00:00:10 MQT: tele/sonoff/RF_Bridge/INFO3 = {"RestartReason":"Blocked Loop"}). I can apply the previous tasmota firmware no problem....Don't know why that is.

Pre10der, 2019-02-24:
Thanks @Portisch - ignore my last, just had to give it a number of attempts until it finally took.

Light On, No Fan
```
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 0280 00A0 2706 38090909191918180909090919191A19190918091819 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0122 0294 01D6 00D2 26D4 48090909091918180909290909091B19090918191809 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 028A 00D2 2698 38090919091918180909090909091A19090918091809 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 029E 00D2 2666 38090909091918180909090909091A19090918091809 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 029E 01E0 2684 38090909091918180909090909091829090918091829 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0294 267A 28090909091918180909090909091809090918091809 55"}}
23:27:26 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 029E 01D6 267A 38090909291918180929290929191829292918291829 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 028A 268E 28090909090918180909090909191809090918091809 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0122 028A 00C8 01D6 267A 4809090909091A181909090909091839193918091809 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 029E 00C8 268E 3809090909091A180909090909091809190918091809 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 00D2 2684 3809090909091A180909090909091819190918091809 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 012C 0294 00D2 01E0 268E 4809090909091A180909090909091839190918091809 55"}}
23:27:27 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 029E 00D2 2684 3809090909091A180909090909091819190918091809 55"}}
```

Light On, Fan High
```
23:27:46 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0280 01B8 0104 0A8C 38181818180A18 55"}}
23:27:46 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0186 00FA 0794 028A 0370 C092B0 55"}}
23:27:46 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 02B2 06CC 2666 3809090909091828090909090918092918091818 55"}}
23:27:46 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 0244 07BC 26B6 381919191919181819191919191918191819182818 55"}}
23:27:47 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0294 0690 00A0 2648 48090909090918180909090929180918091809181B 55"}}
23:27:47 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0280 00D2 2648 3809091909091818090919091919180918091809181A 55"}}
23:27:47 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 028A 2652 28090909090918180909090909091819180918091818 55"}}
23:27:47 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0280 2666 28090919191918181909091919091819180918191818 55"}}
```

Light On, Fan Medium
```
23:28:00 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0118 01D6 0316 00C8 2666 4819090909092828191919191919282809192B281919 55"}}
23:28:01 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0104 01D6 0352 00BE 267A 481919191919282819191919191928281919282B1919 55"}}
23:28:01 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 01B8 0316 265C 38190919191928281919191919192828191928281919 55"}}
23:28:01 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 0352 01F4 02D0 38182A2A18 55"}}
23:28:01 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0122 0294 008C 01D6 265C 480909090909181A1919391909091818391918180909 55"}}
23:28:02 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0122 029E 01D6 00B4 265C 480909090929181B1919290909091818291918180909 55"}}
```

Light On, Fan Low
```
23:28:16 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0352 0208 030C 3818182A18182A 55"}}
23:28:17 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 01E0 07D0 030C 381928 55"}}
23:28:17 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00D2 035C 01CC 02F8 3818182A18182A 55"}}
23:28:18 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 01F4 033E 26A2 38191919191928281919191919192828281928281928 55"}}
23:28:18 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0118 01CC 0320 00A0 26A2 4819090909092828091919191919282828192B281928 55"}}
```

Light Off, Fan High
```
23:29:24 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0276 01B8 0104 0A82 3818080A08080A 55"}}
23:29:24 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 029E 06C2 00D2 263E 480909190909181809090908290909180909181B1B 55"}}
23:29:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0154 029E 00F0 2698 380909090909181809090909090909091A090918181A 55"}}
23:29:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 01AE 0352 26A2 38191919191928281919191919191919281919282828 55"}}
23:29:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0294 01E0 267A 38090909090918180909291909292929182909181818 55"}}
23:29:25 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 028A 01D6 26A2 38090909090918180909090909292919180909181818 55"}}
[... cut ...]
23:29:50 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 0280 265C 28090909090918180909090909090918180918090918 55"}}
23:29:54 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 01C2 033E 2666 38190919091928281919191919191928281928191928 55"}}
23:29:54 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 029E 2666 28090909090918180909090909090918180918090918 55"}}
23:29:54 MQT: tele/sonoff/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 028A 267A 28090909090918180919190909090918181918090918 55"}}
```

Portisch, 2019-02-25:
Please try the new binary with 0xA6 sniffing:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/25b90c9ff90ab09c973d1805381f108100b0e143/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Pre10der, 2019-02-25:
Sorry @Portisch - Still not working. Firmware uploaded fine, however I get no response when listening with the rfraw 166 command. I also don't see any orange light flash.