Portisch/RF-Bridge-EFM8BB1 #87: Digoo DG-HOSA Smart 433MHz Wireless Household Carbon Monoxide Sensor

spider7611, 2019-02-06:
Have a new CO sensor and the Rf Bridge not received code when I push the test buttons. 

Only show this if I use the Rfraw 177:

`20:16:47 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 023A 0604 0186 3494 3818181819090909081819090819081908181818181819090A 55"}}
20:16:48 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0258 05E6 01AE 348A 3818181819090909081819090819081908181818181819090A 55"}}
20:16:49 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 03 024E 05DC 348A 28181818190909090818190908190819081818181818190908 55"}}
20:16:49 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0258 05DC 01AE 348A 3818181819090909081819090819081908181818181819090A 55"}}`

Portisch, 2019-02-11:
Please try the new branch https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/DG-HOSA with command [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)!

spider7611, 2019-02-11:
Hi, I upload the new hex. Now not received any signal in normal mode. When I type this: rfraw AAA655 then the old remote control show codes like this: 17:28:34 CMD: Rfraw AAA655
17:28:34 MQT: stat/sonoff_rf/RESULT = {"RfRaw":"ON"}
17:28:34 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:33:35 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA6040034334355"}}
But from the CO sensor not received anything.

Portisch, 2019-02-19:
I pushed the branch again. Please try the new hey file:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/0af84201b55dee1bf9a8ce358208c79ad4abd08d/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Use it with rfraw aaa655!

spider7611, 2019-02-19:
I tryed it. In normal mode now working again. 
But after rfraw aaa655  not receiving any code from the CO sensor.

spider7611, 2019-02-19:
Only RfRaw 177 show this:

`12:01:00 MQT: stat/sonoff_rf/RESULT = {"RfRaw":"ON"}
12:01:00 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:01:09 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 060E 0190 348A 3818181819090909081819090819081908181818181819090A 55"}}
12:01:22 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 0622 3494 28181818190909090818190908190819081818181818190908 55"}}
12:01:23 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 05E6 01A4 3480 3818181819090909081819090819081908181818181819090A 55"}}
12:01:23 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0244 05F0 3480 28181818190909090818190908190819081818181818190908 55"}}
12:01:23 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 05E6 01AE 3480 3818181819090909081819090819081908181818181819090A 55"}}
12:01:24 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 05F0 01A4 3480 3818181819090909081819090819081908181818181819090A 55"}}
12:01:35 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0226 060E 0190 3494 3818181819090909081819090819081908181818181819090A 55"}}
12:01:36 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 05E6 01A4 348A 3818181819090909081819090819081908181818181819090A 55"}}
12:01:37 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 05E6 01A4 348A 3818181819090909081819090819081908181818181819090A 55"}}`

Portisch, 2019-02-19:
I think it was because of the "small" snyc pulse of 430µs. Please try this again:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/7a5b0349c2dc49d813d6c99a6820e53bc9f35fee/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

spider7611, 2019-02-19:
Now what I recognized. In normal mode when I press about 4-5 times the test button it give one code:
`12:15:49 CMD: Rfraw 0
12:15:49 MQT: stat/sonoff_rf/RESULT = {"RfRaw":"OFF"}
12:16:06 MQT: tele/sonoff_rf/RESULT = {"RfReceived":{"Sync":13510,"Low":540,"High":1570,"Data":"0F3503","RfKey":"None"}}
12:16:33 MQT: tele/sonoff_rf/RESULT = {"RfReceived":{"Sync":13490,"Low":530,"High":1580,"Data":"0F3503","RfKey":"None"}}`
so this 2 code receive after about 10 push.

Portisch, 2019-02-19:
This is maybe a tolerance issue.
For 0xA4 it should be this timing:
Sync long: 13440µs
Sync short: 13440µs / 31 == 433µs

Bit 0/1 short pulse: 433µs
Bit 0/1 long pulse: 1299µs

You have:
Sync long: 13440µs
Sync short: 13440µs / 31 == 433µs

Bit 0/1 short pulse: 590µs
Bit 0/1 long pulse: 1510µs

spider7611, 2019-02-19:
Now the new hex with RfRaw aaa655 receive codes:

`12:25:07 CMD: Rfraw AAA655
12:25:07 MQT: stat/sonoff_rf/RESULT = {"RfRaw":"ON"}
12:25:07 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:25:21 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}
12:25:26 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}
12:25:32 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}
12:25:32 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}
12:25:33 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}
12:25:34 MQT: tele/sonoff_rf/RESULT = {"RfRaw":{"Data":"AAA604090F350355"}}`

closed 2019-02-19 by Portisch

spider7611, 2019-02-19:
And what now? In normal mode not receive any code from the CO sensor. How will working for me? :)

Portisch, 2019-02-19:
Wrong place. Go to https://github.com/arendst/Sonoff-Tasmota

spider7611, 2019-02-19:
Sorry but don't understand why must go there? I hoped will work if I upload the new modified hex. Or must do any modification too?

Portisch, 2019-02-19:
You have the sniffed 0xA6 RF data and can do what you want with this data on your host.
This repo is only about the sniffing - not the decoding or something else.

spider7611, 2019-02-19:
Oh ok. Thank Your help! 
I just want a code from the CO sensor in normal mode when the sensor send an alarm. But I don't know what can I do with this sniffed data, how to put in the Sonoff Rf Bridge to receive it and recognized it?!

Portisch, 2019-02-19:
No, this device do not use the "standard" PT226x protocol.