Portisch/RF-Bridge-EFM8BB1 #89: RF SIGNAL NOT ALWAYS RECEVIED

SuoaJ, 2019-02-10:
Hi guys,

I'm sorry if this issue has been raised before, but I search the list of issues and did not see any answer. I have Tasmota 6.4.1 installed on my Sonoff RFBridge. I am using a Kerio F52 door button to communicate with it. I have install "RF-Bridge-EFM8BB1-20181127.hex" found in the Tasmota repository on the RF chip. Everything works. But the problem is the door button signal is not always recognized by the RFBridge. Tried with two separate buttons and the result is the same. They work sometimes but not all the time. Any suggestions?

Pre10der, 2019-02-10:
Could it be the network side and not RF side? I had Wifi issue with Tasmota 6.4.1 - it was dropping in and out of WiFi every 5mins or so. I rectified this by downgrading to 6.3. In my experience, connecting to WiFi on the ESP8266 devices is resource intensive and if it's doing it repeatedly, it could explain the behaviour you are describing.

SuoaJ, 2019-02-10:
I don't think it is network. I don't see any packet losses using ping over an extended period. I am also monitoring the RF signals received through the console feature of the RFBridge. Another observation is that the red light on the bridge(RF) blinks intermittently even if I am not using any 433Mhz RF sensor. These "blinks" are  not registered in the console either.

SuoaJ, 2019-02-10:
Hi @Pre10der, a little off topic, but I'm trying to test your theory by downgrading to 6.3. How did you do it? OTA(through the web gui) or over the serial connection? I tried OTA over the web interface but it doesn't seem to work. Getting this error, "Upload Failed. Upload buffer miscompare". Thanks.

Pre10der, 2019-02-10:
Serial is always the best bet so that it zero's out the chip, however, I have done it a number of times by first 'upgrading' (really downgrading) to the 6.3 **minimal** Tasmota firmware, then upgrading again to the standard 6.3 firmware. You need to go to the minimal version first as there is not enough memory to hold both full firmware's during the up/downgrade.

SuoaJ, 2019-02-11:
Thanks. The ESP8266 downgrade process worked. However, the connection issue between the door button and the RF Bridge persist. There is something weird about the red RF light blinking even if there is no 433Mhz radio signal. I don't recall observing this with the factory firmware on the RF chip. . My next step will be to downgrade the firmware on the RF chip(EFM8BB1) back to the factory default. If that doesn't work, I will try the version available under Releases in the Portisch repo.

Portisch, 2019-02-11:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

### Information
What device you try to sniff data from? Description? Brand and type?

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

closed 2019-02-11 by Portisch

SuoaJ, 2019-02-12:
Thanks @Portisch. The problem is solved. I have not noticed this issue since I upgraded the RF chip to Version 0x03. Someone should tell the guys at Tasmota about this issue with the build on their repo.