Portisch/RF-Bridge-EFM8BB1 #90: New 0x03 firmware not receive any data in normal mode.

spider7611, 2019-02-13:
### Latest firmware used?
Version 0x03

### Information
Not receiving any data after upgrade to newest firmware. Must downgrade it. What is the problem?

### Sniffed data
No data.

Portisch, 2019-02-13:
Yes, maybe possible... I have done a "fix" on "normal" mode. I will check it!

Portisch, 2019-02-14:
Please try this version until I receive my new Arduino:
[RF-Bridge-EFM8BB1.zip fix 0x03, fix standard sniffing](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2863738/RF-Bridge-EFM8BB1.zip)

spider7611, 2019-02-14:
Tryed it just now and not working.

Portisch, 2019-02-14:
Can you please try this version:
[RF-Bridge-EFM8BB1.zip](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2863798/RF-Bridge-EFM8BB1.zip)

spider7611, 2019-02-14:
Not working!

zerog2k, 2019-02-15:
I also notice A6/A7 dont work with digoo door/window switch from a few feet away. (B0 does show raw data sniffing), using 0x03 version (2/11 date). I think earlier 0x03 version from releases page which was Jan 30? date worked, but I think I overwrote this file with newer version :(

Portisch, 2019-02-15:
I rebased 0x03 and it looks like it break 0xA4. But 0xA6 should work.
Just browse the source for each commit as there is always a hex file included.

This is the first 0x03 version: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/393d535603095307ee43271f2d02abdb8a4a972d/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

zerog2k, 2019-02-17:
Thanks Portisch. You are right, it is A4 decoding which is broken for me. Rolling back to that version it's working.

Portisch, 2019-02-19:
Solved by https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/a6da9e46497a5e4e9fa9ea3c3207466c3510c75d

New Release:
https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x03

closed 2019-02-19 by Portisch