Portisch/RF-Bridge-EFM8BB1 #91: Sonoff Door/Window Sensor and PIR Sensor

MRvH88, 2019-02-13:
@Portisch Just merged and uploaded the new fixes you committed, however my sonoff door/window and PIR sensor are not being received through default. 

Before the new commits I could receive them in `rfraw 0` (sonoff default/ `0xA4`) or through `rfraw 177`, (`0xB1`), `rfraw 166` (`0xA6`) was not getting any results. But with the new commit I can only get an output though `rfraw 177`, both `rfraw 0` and `rfraw 166` are not giving any results.

If I understand correctly, I should be able to receive standard Sonoff RF devices though all three options ?

Below are some `0xB1` results

`13:07:16 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 04E2 0186 30AC 28181819081819090909090818190908190908190818181909 55"}}
13:07:16 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0500 0172 30A2 28181819081819090909090818190908190908190818181909 55"}}
13:07:24 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 04F6 017C 30A2 28181819081819090909090818190908190908190818181909 55"}}
13:07:30 MQT: Sonoff_RF_Bridge/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 04EC 0190 30F2 28181819081819090909090818190908190908190818181909 55"}}`

Portisch, 2019-02-13:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/90

Portisch, 2019-02-14:
Please try this version until I receive my new Arduino:
[RF-Bridge-EFM8BB1.zip fix 0x03, fix standard sniffing](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2863738/RF-Bridge-EFM8BB1.zip)

miit86, 2019-02-16:
Same Problem here.
Cannot receive RF data in rfraw 0 or rfraw 166
rf 177 is working great.
Sending rfraw with MQTT functions great.

what i tried:
- flashed tasmota again
- flashed efm again
- sonoff PIR, Sonoff door/windows, KAKU

Portisch, 2019-02-19:
Solved by https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/a6da9e46497a5e4e9fa9ea3c3207466c3510c75d

New Release:
https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x03

closed 2019-02-19 by Portisch