Portisch/RF-Bridge-EFM8BB1 #92: B1 to b0 problem

plbartek89, 2019-02-13:
Hello , im tryin to decode my 433mhz door bell 
Can someone help me to decode  & understand that ?
`AA B1 03 00A0 01D6 137E 28190819081818181818181819090909081908181819081818 55`

Portisch, 2019-02-13:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

### Information
What device you try to sniff data from? Description? Brand and type?

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

plbartek89, 2019-02-13:
Hello , i useg latest .hex file 
Its a 433mhz old evology door bell i also have some 433mhz Smoke detectors (smartwares fa22rf)

I use tasmota , with RfRaw AAA655 command no data , with RfRaw AAB155 i recive data , 

`23:58:14 CMD: RfRaw 177
23:58:14 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
23:58:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
23:58:18 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0064 0172 01E0 280918 55"}}

23:58:22 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00D2 01B8 006E 1428 38181818181818181818181819292929281928181819281818 55"}}

23:58:22 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 01CC 1392 28181818181818181818181819090909081908181819081818 55"}}

23:58:22 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 0078 136A 38181818181818181818181819292929281928181819281818 55"}}

23:58:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01AE 006E 1360 38181818181818181818181819292929281928181819281818 55"}}

23:58:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00B4 01C2 0082 136A 38181818181818181818181819290909081908181819081818 55"}}

23:58:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 01D6 139C 28181818181818181818181819090909081908181819081818 55"}}

23:58:32 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01AE 006E 1360 38181818181818181818181819292929281928181819281818 55"}}

23:58:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00BE 01AE 006E 1356 38181818181818181818181819292929281928181819281818 55"}}

23:58:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 0078 1360 38181818181818181818181819292929281928181819281818 55"}}

23:58:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00BE 01C2 0082 136A 38181818181818181818181819290909081908181819081818 55"}}
23:58:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 01D6 1392 28181818181818181818181819090909081908181819081818 55"}}

23:58:39 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 0078 1356 38181818181818181818181819292929281928181819281818 55"}}

23:58:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 006E 1356 38181818181818181818181819292929281928181819281818 55"}}

23:58:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 01CC 139C 28181818181818181818181819090909081908181819081818 55"}}

23:58:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01AE 006E 1360 38181818181818181818181819292929281928181819281818 55"}}

23:58:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 01C2 1392 28181818181818181818181819090909081908181819081818 55"}}

23:58:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 0078 136A 38181818181818181818181819292929281928181819281818 55"}}

23:58:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00C8 01B8 006E 1356 38181818181818181818181819292929281928181819281818 55"}}

23:58:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00B4 01C2 0082 1392 38181818181818181818181819290909081908181819081818 55"}}`

Portisch, 2019-02-14:
Why you just do not use the [BitBucketConverter](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Decode-0xB1-sniffed-data)?
[image]

It has 24bit, decoded value is: 000000000000111101000100b == 0x000F44

closed 2019-02-18 by Portisch