Portisch/RF-Bridge-EFM8BB1 #96: Request: Include Prologue Sensor

Forceu, 2019-03-08:
It would be great if the Prologue Sensor was included. It is a protocol used by some Chinese weather stations.

From https://github.com/merbanan/rtl_433/blob/master/src/devices/prologue.c : 

> The sensor sends 36 bits 7 times, before the first packet there is a sync pulse.
> The packets are ppm modulated (distance coding) with a pulse of ~500 us
> followed by a short gap of ~2000 us for a 0 bit or a long ~4000 us gap for a
> 1 bit, the sync gap is ~9000 us.
> The data is grouped in 9 nibbles
>     [type] [id0] [id1] [flags] [temp0] [temp1] [temp2] [humi0] [humi1]
> - type: 4 bit fixed 1001 (9) or 0110 (5)
> - id: 8 bit a random id that is generated when the sensor starts, could include battery status
>   the same batteries often generate the same id
> - flags(3): is 0 the battery status, 1 ok, 0 low, first reading always say low
> - flags(2): is 1 when the sensor sends a reading when pressing the button on the sensor
> - flags(1,0): the channel number that can be set by the sensor (1, 2, 3, X)
> - temp: 12 bit signed scaled by 10
> - humi: 8 bit always 11001100 (0xCC) if no humidity sensor is available

```
r_device prologue = {
    .name           = "Prologue Temperature Sensor",
    .modulation     = OOK_PULSE_PPM,
    .short_width    = 2000,
    .long_width     = 4000,
    .gap_limit      = 7000,
    .reset_limit    = 10000 }
```

I don't think it would be necessary to actually decode the values, but it would be great if the encoded data was output. This would increase the code size only very slightly and would therefore still fit on the chip.

Portisch, 2019-03-08:
Make a 0xB1 sniffing!

Forceu, 2019-03-08:
First one is from the automatic broadcast, the second one when pressing a button. All on internal channel 1

```
17:38:37 MQT: sonoff-rf/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 02A8 0F78 0794 22CE 0102020102010102020202020202020202020202010201020202020202020101020101010203 55"}}
model     : Prologue sensor                        id        : 9
rid       : 96           Channel   : 1             Battery   : LOW           Button    : 0
Temperature: 16.00 C     Humidity  : 55 %
```

```
17:41:28 MQT: sonoff-rf/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0294 0F96 079E 22E2 0102020102010102020202020201020202020202010201020102010202020101010202020203 55"}}
17:41:28 MQT: sonoff-rf/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0294 07A8 0FA0 0102 55"}}

model     : Prologue sensor                        id        : 9
rid       : 96           Channel   : 1             Battery   : LOW           Button    : 1
Temperature: 17.00 C     Humidity  : 56 %
```

Portisch, 2019-03-11:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/Prologue_Sensor/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex
When you use the BitBucketConverter you can see 37 bits insteed of 36!?
I added the protocol, please test it. You should now receive data in 0xA6 sniffing mode.

Forceu, 2019-03-13:
I just tested it and it works great in 0xA6 mode!

> When you use the BitBucketConverter you can see 37 bits insteed of 36!?
I didn't use BitBucketConverter, this was the output from the Tasmota Console with 0xB1 sniffing.

How would I modify the firmware code to actually display the data without being in 0xA6 mode? I am still reading through the code, but do not understand it 100%.

closed 2019-03-18 by Portisch