Portisch/RF-Bridge-EFM8BB1 #98: Support for Manchester Coding.

FatBeard, 2019-03-12:
Would it be possible to get support for Manchester coding. Having it would help with other types of protocols such as Oregon etc.

Portisch, 2019-03-12:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

### Information
What device you try to sniff data from? Description? Brand and type?

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

Why is nobody using the template?
And no, manchester coding will not be handled as the EFM8BB1 is running out of storage.

closed 2019-03-12 by Portisch

FatBeard, 2019-03-13:
It’s for an owl electricity monitor which I already have working on arduino. I was going to port it over to Portisch. Hence the template was irrelevant. I was going to do the protocol work. It uses Manchester coding though. Hence the question. I’ve already given you the 1byone protocol pull request which shows I am able to do the protocol work myself.

JoostvdB94, 2019-09-01:
Because the protocols can now be configured in `RfConfig.h`, i was hoping storage wouldn't be a problem anymore: We could choose to opt-in Manchester support as well?