Portisch/RF-Bridge-EFM8BB1 #102: Byron BY302 Doorbell

Idcsl, 2019-03-19:
Hi,
I finally got round to looking into this once again following my last attempt in December. Today 19th March I have installed the latest master EFM8 firmware and here follows the results of 0xB1 sniffing. I'm hoping you may be able to now see if this can be added to the protocol list.
Thanks

16:45:35 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01B8 041A 03F2 28181818 55"}}
16:45:35 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 0BFE 281809091809180918090918090909181818181818 55"}}
16:45:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03E8 28181818 55"}}
16:45:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 03F2 28181818 55"}}
16:45:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03D4 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03F2 28181818 55"}}
16:45:37 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:37 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03E8 03F2 28181818 55"}}
16:45:40 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 0424 0406 28181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03F2 0BF4 281809091809180918090918090909181818181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03FC 03FC 28181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 0BC2 281809091809180918090918090909181818181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 28181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 28181818 55"}}
16:45:41 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:42 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03FC 28181818 55"}}
16:45:42 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:42 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 28181818 55"}}
16:45:42 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 019A 0438 0424 28181818 55"}}
16:45:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01C2 03FC 0BEA 281809091809180918090918090909181818181818 55"}}
16:45:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 0406 28181818 55"}}
16:45:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:46 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03FC 03FC 28181818 55"}}
16:45:46 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:46 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03F2 28181818 55"}}
16:45:46 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:46 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 28181818 55"}}
16:45:47 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01A4 042E 0406 28181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01C2 03FC 0BF4 281809091809180918090918090909181818181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 0406 03FC 28181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 0BC2 281809091809180918090918090909181818181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03F2 03F2 28181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 0BB8 281809091809180918090918090909181818181818 55"}}
16:45:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03F2 28181818 55"}}
16:45:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03E8 0BC2 281809091809180918090918090909181818181818 55"}}
16:45:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03F2 28181818 55"}}
16:45:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BC2 281809091809180918090918090909181818181818 55"}}
16:45:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03F2 03FC 28181818 55"}}
16:45:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BC2 281809091809180918090918090909181818181818 55"}}

Portisch, 2019-03-20:
Please try this binary: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/BY302/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Idcsl, 2019-03-20:
Ok thanks. I have just installed that binary. Its not recognising the code by default so here is the result of 0xB1 sniffing:

14:48:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03DE 28181818 55"}}
14:48:55 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01C2 0406 0BFE 281809091809180918090918090909181818181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 0406 0406 28181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 0BCC 281809091809180918090918090909181818181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 0406 0406 28181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 0BCC 281809091809180918090918090909181818181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03FC 28181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
14:48:56 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03F2 28181818 55"}}
14:48:57 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
14:48:57 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03F2 28181818 55"}}
14:48:57 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
14:48:57 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03F2 03F2 28181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 0424 0BB8 281809091809180918090918090909181818181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 041A 0424 28181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 0BCC 281809091809180918090918090909181818181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 03FC 03FC 28181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03F2 0BC2 281809091809180918090918090909181818181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 28181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03E8 28181818 55"}}
14:49:07 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03DE 0BB8 281809091809180918090918090909181818181818 55"}}
14:49:08 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03E8 03E8 28181818 55"}}
14:49:08 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BC2 281809091809180918090918090909181818181818 55"}}
14:49:08 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03F2 03F2 28181818 55"}}

Portisch, 2019-03-20:
You used mode [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)?

Idcsl, 2019-03-20:
Ah yes sorry no I didn't it was was 0XB1. This is with 0xA6:

15:15:51 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:15:52 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:16:14 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA60409AA2D0455"}}
15:16:18 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA60409AA2D0455"}}
15:16:31 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:16:31 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:16:43 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:16:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}
15:16:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AAA6040F6ADC0055"}}

Portisch, 2019-03-21:
https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/a364d24df3e3b3dd1ea497a8fd672d3d3d9b3d9a

closed 2019-03-21 by Portisch

Idcsl, 2019-03-21:
Hi @Portisch, I see you closed the issue adding the doorbell to the firmware but having installed this latest version it doesn't give a result in the Tasmota console other than when you issue the 0xA6 command and the result then is as before i.e {"RfRaw":{"Data":"AAA6040F6ADC0055"}}.

Portisch, 2019-03-21:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/94#issuecomment-467835586