Portisch/RF-Bridge-EFM8BB1 #103: Fan remote - trying to understand the protocol used FRM87

danps1, 2019-03-21:
### Latest firmware used?
Using 0.4 Portisch Firmware with Tasmota 6.4.1

### Information
The device is a ceiling fan remote - https://www.mercator.com.au/product/rf-remote-control/

### Sniffed data
I only see results with RFRAW 177 and then every button press results in 2 lines appearing, not 1. I have tried to use BitBucketConverter to replay, but because it goes line-by-line, I don't think the whole code is being sent so no getting any success with replaying.
I don't know how to understand these 2 lines of output for a single button press. If anyone could help that would be amazing!

Here is some capture...
```
10:52:28 CMD: rfraw 177
10:52:28 MQT: sonoff/stat/RESULT = {"RfRaw":"ON"}
10:52:28 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
10:52:42 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0302 04A6 0118 04BA 38181A 55"}}
10:52:42 CMD: ### Press Off
10:52:42 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:52:50 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A62 28090909091818181818180918 55"}}
10:52:51 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 026C 2A44 28090909091818181818181818 55"}}
10:53:04 CMD: ### Press Off
10:53:04 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:53:06 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A58 28090909091818181818180918 55"}}
10:53:13 CMD: ### Press Off
10:53:13 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:53:16 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A58 28090909091818181818180918 55"}}
10:53:16 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A62 28090909091818181818181818 55"}}
10:53:22 CMD: ### Press Off
10:53:22 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:53:25 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 015E 028A 03D4 00FA 2A3A 48090909091818181818180A38 55"}}
10:53:25 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0280 2A4E 28090909091818181818180918 55"}}
10:53:25 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 026C 2A58 28090909091818181818181818 55"}}
10:53:40 CMD: ### Press Off
10:53:40 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:53:43 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 014A 032A 006E 0262 2A58 48092909093838381838383838 55"}}
10:53:51 MQT: sonoff/tele/STATE = {"Time":"2019-03-21T10:53:51","Uptime":"0T00:15:16","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PSClover-IOT","BSSId":"72:3A:CB:84:7A:C8","Channel":6,"RSSI":100}}
10:53:52 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 0078 012C 281819 55"}}
10:54:00 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 038E 047E 281818 55"}}
10:54:00 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0302 010E 047E 0564 380929 55"}}
10:54:06 CMD: ### Press Low
10:54:06 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:54:08 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0140 0280 2A58 2A44 380909090918181818091818182809090909181818181818181828 55"}}
10:54:22 CMD: ### Press Low
10:54:22 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:54:25 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 0276 2A62 28090909091818181818181818 55"}}
10:54:34 CMD: ### Press Low
10:54:34 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:54:39 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0168 00FA 028A 2A30 381A1A1A0A282828280A282828 55"}}
10:54:39 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0258 2A6C 28090909091818181818181818 55"}}
10:54:40 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 01F4 00F0 0438 01AE 381A29 55"}}
10:54:43 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 0794 07C6 281818 55"}}
10:54:50 CMD: ### Press Low
10:54:50 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:54:53 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0168 0104 0294 2A4E 381A1A1A1A2828282828282828 55"}}
10:55:07 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0078 0B54 017C 0366 38182A 55"}}
10:55:07 CMD: ### Press Med
10:55:07 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:55:10 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A3A 28090909091818181818181818 55"}}
10:55:18 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 028A 0136 00D2 04B0 38182A 55"}}
10:55:23 CMD: ### Press Med
10:55:23 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:55:28 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0276 2A62 1F04 2A4E 480909090918181818181818182809090909181818181818181883 55"}}
10:55:32 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 026C 074E 0190 03FC 0F6E 4818181818283818 55"}}
10:55:33 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0186 0848 07EE 281818 55"}}
10:55:33 CMD: ### Press Med
10:55:33 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:55:35 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 0280 2A58 28090909091818180918181818 55"}}
10:55:35 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 0280 2A58 28090909091818181818181818 55"}}
10:55:43 CMD: ### Press Med
10:55:43 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:55:46 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 0280 2A58 28090909091818180918181818 55"}}
10:55:46 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0276 2A62 28090909091818181818181818 55"}}
10:55:57 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 006E 0456 280919 55"}}
10:55:59 CMD: ### Press Hi
10:55:59 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:56:01 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0078 00FA 0384 08E8 B81928 55"}}
10:56:02 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 0276 2A58 28090909091818091818181818 55"}}
10:56:02 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 026C 2A6C 28090909091818181818181818 55"}}
10:56:10 CMD: ### Press Hi
10:56:10 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:56:11 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 015E 0280 00A0 2A3A 38092909091818091818181818 55"}}
10:56:11 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 014A 0276 2A4E 28090909091818091818181818 55"}}
10:56:15 CMD: ### Press Hi
10:56:15 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:56:17 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0276 2A58 28090909091818091818181818 55"}}
10:56:17 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0276 2A6C 28090909091818181818181818 55"}}
10:56:27 CMD: ### Press Hi
10:56:27 MQT: sonoff/stat/RESULT = {"Command":"Unknown"}
10:56:27 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 008C 0A6E 026C 2AD0 03C0 1C52 581A3C 55"}}
10:56:28 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 028A 2A44 28090909091818091818181818 55"}}
10:56:29 MQT: sonoff/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 0276 00E6 2A6C 38090929091818181818181818 55"}}
```

Portisch, 2019-03-21:
This protocol is already implemented: HT12. Just the timing is different. Please try this binary:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/FRM87/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

danps1, 2019-03-22:
Thanks @Portisch - it's really awesome of you to help me! 
I'm struggling to flash the hex file linked above to my RF chip.
Should I be able to use the "upload firmware" mechanism via Tasmota? (6.4.1)
This had previously worked for me, but now I am trying every permutation (switch position, usb vs pin power, wires between C2ck, etc etc) and every time I'm getting the Magic Byte error.
Any idea what I could be doing wrong?
Thanks again.

danps1, 2019-03-22:
One thing to amend - I have just upgraded Tasmota to 6.5, and also issued the FF command, and discovered that I have the v3 Portisch rather than v4 as previously stated. Still no joy putting the firmware on there though :(
Also tried the hex in the Keil folder - same Magic Byte error

Portisch, 2019-03-22:
Download the complete zip and not the file. Take a look to the file by a text editor to see if it is a valid hex file.

danps1, 2019-03-22:
OMG - feeling pretty humble / stupid right now. Opening up in text editor made me realise that when I'd been right clicking and choosing "Save Link As..." I'd then subsequently been trying to upload a bunch of HTML to my RF-chip and pulling my hair out wondering why it wasn't working! 
I've now successfully uploaded latest firmware (as a hex file!!) and been able to also successfully replay some B0 captures for the first time. Not seeing it as decoded protocol though, but will cross that bridge next.
Thanks again for the help, and for this fantastic firmware.

Portisch, 2019-03-22:
Use 0xA6 command! Take a look to the wiki.

danps1, 2019-03-22:
That was actually the very first thing I tried once I'd upgraded the firmware, but it didn't decode the signal. I think it should be protocol 6, but i get no output when pressing and even holding the remote button...
```
CMD: rfraw AA A6 55
MQT: stat/rf/RESULT = {"RfRaw":"ON"}
MQT: tele/rf/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

danps1, 2019-03-22:
If it helps, here is some of the B0 data that I captured and then successfully replayed...
```
Bedroom Off - rfraw AA B0 15 03 04 0140 0276 2A80 18280909090909181818181809 55
Bedroom Low - rfraw AA B0 15 03 04 014A 028A 2A6C 18280909090909181818091818 55
Bedroom Med - rfraw AA B0 15 03 04 014A 0280 2A76 18280909090909181809181818 55
Bedroom Hi  - rfraw AA B0 15 03 04 014A 0280 2A80 18280909090909180918181818 55

Guest Off - rfraw AA B0 15 03 04 0154 028A 2AA8 18280909180918181818181809 55
Guest Low - rfraw AA B0 15 03 04 015E 028A 2A94 18280909180918181818091818 55
Guest Med - rfraw AA B0 15 03 04 015E 0294 2A94 18280909180918181809181818 55
Guest Hi  - rfraw AA B0 15 03 04 0154 028A 2AB2 18280909180918180918181818 55

Office Off - rfraw AA B0 15 03 04 0140 0276 2A4E 18280909090918181818181809 55
Office Low - rfraw AA B0 15 03 04 0154 0280 2A3A 18280909090918181818091818 55
Office Med - rfraw AA B0 15 03 04 0140 0280 2A44 18280909090918181809181818 55
Office Hi  - rfraw AA B0 15 03 04 0136 026C 2A4E 18280909090918180918181818 55
```

Portisch, 2019-03-26:
Try the last hex file from the master branch: https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/af1bddb3d81c79d67063184219ec21f8249dffd0
I adjusted the values for the HT12 protocol and disabled the other once. Please have a try...

danps1, 2019-03-28:
Excellent - thanks again!
It is now decoding the button presses, with 0xA6 sniffing....
```
Guest Off - rfraw AA B0 15 03 04 0154 028A 2AA8 18280909180918181818181809 55   ----> AAA603032FD055
Guest Low - rfraw AA B0 15 03 04 015E 028A 2A94 18280909180918181818091818 55   ----> AAA603032F7055
Guest Med - rfraw AA B0 15 03 04 015E 0294 2A94 18280909180918181809181818 55   ----> AAA603032EF055
Guest Hi  - rfraw AA B0 15 03 04 0154 028A 2AB2 18280909180918180918181818 55   ----> AAA603032DF055 / AAA603032FF055
```
However, if I'm understanding the protocols.h file correctly, these commands are being decoded as protocol 0x03 which is a weather station, rather than 0x06 which i was expecting for HT12? Am I misunderstanding what is going on? (again) :)