Portisch/RF-Bridge-EFM8BB1 #104: Define protocol for 5514 SILENT Dual Tech

tt1pjm, 2019-03-22:
### Latest firmware
Tested with Version 0x04

### Information
I'm trying to define a new protocol for 2 motion sensors. I've sniffed the following data for both:
Sensor A
{"RfRaw":{"Data":"AA B1 03 0190 02D0 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02DA 1342 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 017C 02E4 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02DA 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02D0 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02DA 132E A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02D0 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02DA 132E A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 1342 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 1342 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02DA 1342 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 1342 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02D0 1338 A0909180909091809180918180918090918180909090909090918090909091818090918090918180 55"}}
Sensor B
{"RfRaw":{"Data":"AA B1 03 0186 02E4 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 1338 A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 1338 A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02E4 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02D0 1338 A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 017C 02E4 1338 A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02DA 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0186 02E4 1324 A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}
{"RfRaw":{"Data":"AA B1 03 0190 02D0 132E A0909180909180918180918180918090918180909090909090918090909091818090918090909181 55"}}

For both sensors I get the buckets:
Bucket 0: 400
Bucket 1: 720
Bucket 2: 4920
SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(SILENT_5514)[], static uint16_t, SI_SEG_CODE) = { 400, 720, 4920 };

I've then used the following to run tests using 166.

Sensor A
Sync Bucket: 0A
First sync bucket number: [0] & 0x07 == 0
First sync bucket high/low marking: [0] & 0x08 > 0 == low
Second sync bucket number: [A] & 0x07 == 2
Second sync bucket high/low marking: [A] & 0x08 > 0 == high
SI_SEGMENT_VARIABLE(PROTOCOL_START(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(2) };
Bit0
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(0) };
Bit1
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(1) };

Sensor B
Sync Bucket: 1A
First sync bucket number: [1] & 0x07 == 1
First sync bucket high/low marking: [1] & 0x08 > 0 == low
Second sync bucket number: [A] & 0x07 == 2
Second sync bucket high/low marking: [A] & 0x08 > 0 == high
SI_SEGMENT_VARIABLE(PROTOCOL_START(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(2) };
Bit0
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(0) };
Bit1
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(SILENT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(1) };

I've run tests using a length of 39 and 40 bits but I'm not getting anything back on 166 so I've obviously gone wrong in defining the protocol. Can anyone see where I need to make changes?

Portisch, 2019-03-22:
Try this patch:

```
diff --git a/inc/RF_Config.h b/inc/RF_Config.h
index 5a47d73..6576afd 100644
--- a/inc/RF_Config.h
+++ b/inc/RF_Config.h
@@ -33,3 +33,4 @@
 #define EFM8BB1_SUPPORT_Prologue_PROTOCOL              1               // Prologue Sensor, Issue #96
 #define EFM8BB1_SUPPORT_DOG_COLLAR_PROTOCOL            1               // Generic dog training collar - board label T-187-n (TX)-1, PR #100
 #define EFM8BB1_SUPPORT_BY302_PROTOCOL                 1               // Byron BY302 Doorbell, Issue #102
+#define EFM8BB1_SUPPORT_DT_5514_PROTOCOL               1               // 5514 SILENT Dual Tech, Issue #104
diff --git a/inc/RF_Protocols.h b/inc/RF_Protocols.h
index 2c4dd4d..bd7100b 100644
--- a/inc/RF_Protocols.h
+++ b/inc/RF_Protocols.h
@@ -295,6 +295,17 @@ SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(BY302)[], static uint8_t, SI_SEG_CODE) = { LOW
 SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(BY302)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(1) };
 #endif

+/*
+ * 5514 SILENT Dual Tech
+ */
+#if EFM8BB1_SUPPORT_DT_5514_PROTOCOL == 1
+#define DT_5514
+SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(DT_5514)[], static uint16_t, SI_SEG_CODE) = { 400, 720, 4910 };
+SI_SEGMENT_VARIABLE(PROTOCOL_START(DT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(2) };
+SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(DT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(1) };
+SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(DT_5514)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(0) };
+#endif
+
 SI_SEGMENT_VARIABLE(PROTOCOL_DATA[], static struct BUCKET_PROTOCOL_DATA, SI_SEG_CODE) =
 {
 #if EFM8BB1_SUPPORT_PT226X_PROTOCOL == 1
@@ -544,6 +555,19 @@ SI_SEGMENT_VARIABLE(PROTOCOL_DATA[], static struct BUCKET_PROTOCOL_DATA, SI_SEG_
                        20
                },
 #endif
+#if EFM8BB1_SUPPORT_DT_5514_PROTOCOL == 1
+               /*
+                * 5514 SILENT Dual Tech
+                */
+               {
+                       { &PROTOCOL_BUCKETS(DT_5514), ARRAY_LENGTH(PROTOCOL_BUCKETS(DT_5514)) },
+                       { &PROTOCOL_START(DT_5514), ARRAY_LENGTH(PROTOCOL_START(DT_5514)) },
+                       { &PROTOCOL_BIT0(DT_5514), ARRAY_LENGTH(PROTOCOL_BIT0(DT_5514)) },
+                       { &PROTOCOL_BIT1(DT_5514), ARRAY_LENGTH(PROTOCOL_BIT1(DT_5514)) },
+                       { NULL, 0 },
+                       39
+               },
+#endif
 };

 #define PROTOCOLCOUNT (sizeof(PROTOCOL_DATA) / sizeof(PROTOCOL_DATA[0]))
```

tt1pjm, 2019-03-22:
Thanks for the help. Made the changes and ran tests and still nothing coming through on 166. Can still see data on 177. Any suggestions?

Portisch, 2019-03-22:
Show your source can help... Just make a fork, new branch and when it's working a pull request.

tt1pjm, 2019-03-23:
Hi Portisch,
Have a look here: https://github.com/tt1pjm/RF-Bridge-EFM8BB1/commit/c88db603c5c09375bbde5d6305ff32da3f7f9dd4