Portisch/RF-Bridge-EFM8BB1 #105: Not an issue per se; what the status of using Tasmota to flash?

Trippersgarage, 2019-03-22:
Just wondering; there was a time there where uploading this using Tasmota seemed to brick the RF Bridge Rf chip. Has that issue been resolved? I lost 2 bridges doing that and not looking to lose a third! :)

Thank you!

ascillato, 2019-03-22:
Hi,

There is no way of bricking the chip by flashing. You can reflash it without problems. Just follow the steps of Tasmota wiki and you can flash all your 3 devices.

Just tested and works fine. :)

Portisch, 2019-03-22:
Sometimes it can happen that the method by Tasmota brick the flash in the EFM8B1. I had it twice. I switched back to the external flashing by using this programmer: https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Hardware

closed 2019-03-22 by Portisch

Trippersgarage, 2019-03-22:
@portisch Just wanted to confirm before I tried it again and thank you for the reply. I’ll avoid using tasmota to flash your brilliant ones and zeros. 

@ascillato Yes it has and I and portisch can confirm it did. Perhaps you assumed I meant flashing the tasmota firmware? In any event, trying to flash portisch’s firmware using tasmota has caused issues.