Portisch/RF-Bridge-EFM8BB1 #106: Help with Auriol H13726 Weather Station

keteflips, 2019-03-25:
### Latest firmware used?
Yes, last firmware.

### Information
Auriol H13726 Weather Station

### Sniffed data

I installed Tasmota in a Sonoff RF Bridge with the last EFM8BB1 firmware.
I get this raw info.
But in realy lost, I dont know what to do with it...

```
17:48:28 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0226 0F3C 0780 0F46 38182828 55"}}
17:48:33 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F32 0776 2274 38182828182818281828282818182828182828282818282828182828281818282828181828 55"}}
17:48:33 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F46 0780 0F3C 38182828 55"}}
17:48:35 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 021C 0780 0F46 2292 5B36 0F32 581828384828 55"}}
17:48:35 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F32 076C 2274 38182828182818281828282818182828182828282818282828182828281818282828181828 55"}}
17:48:36 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F3C 0780 0F3C 38182828 55"}}
17:49:00 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 00BE 01A4 006E 0302 B090A2 55"}}
17:49:27 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F32 2288 0F3C 38181828282828 55"}}
17:49:28 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F3C 0776 0F28 3818182818 55"}}
17:50:29 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F32 227E 0F3C 38181828282828 55"}}
17:50:49 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 01D6 0122 00B4 0082 015E C092B1 55"}}
17:51:00 MQT: sonoff_B2E2C9/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 0230 0F32 0776 2288 38182828182818281828282828282818182828282818282828282818281818282828282818 55"}}
```

Portisch, 2019-03-26:
https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/9ddd25114381cb9326c55fc410a6b4e897a97529

https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/9ddd25114381cb9326c55fc410a6b4e897a97529/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Use this new hex firmware. And then sniff data with command [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)
Here you will find a description how to decode the data:
https://github.com/yu55/auriol_reader/blob/master/doc/auriol_protocol_v20.pdf

closed 2019-03-26 by Portisch

keteflips, 2019-03-26:
> [9ddd251](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/9ddd25114381cb9326c55fc410a6b4e897a97529)
> 
> https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/9ddd25114381cb9326c55fc410a6b4e897a97529/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex
> 
> Use this new hex firmware. And then sniff data with command [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)
> Here you will find a description how to decode the data:
> https://github.com/yu55/auriol_reader/blob/master/doc/auriol_protocol_v20.pdf

Amazing!!! THANKS!!!

keteflips, 2019-03-31:
Ok, I think Im too idiot to understand how this works...

Im reading all the information but I dont understand how to convert the raw data to usable data.

Portisch, 2019-04-01:
Compile the source for yourself. Disable all protocols except yours: EFM8BB1_SUPPORT_H13726_PROTOCOL
Maybe some protocols are similar and the firmware is using them for decoding.

jpduhen, 2019-07-22:
@keteflips Have you found a way to read the Auriol sensors with the Sonoff Bridge?