Portisch/RF-Bridge-EFM8BB1 #108: difference for A6 and A9 sniffing

haegar33, 2019-03-27:
### Latest firmware used?
latest pull from master branch

### Information
Powerplug REV Typ 008341C

I got a problem to understand the difference between the two sniffing/learing command A6 and A9. When I receive output from these command how can I used this "decoded" values for transmitting a command or assigning a rfkey?
Example A6:
I get an answer after pressing a button on my REV power plug

`AAA605071CB7336155`

This should be protocol index 7 with a data length 5 and data 1CB73361. **Looks good but whats next??**

Example A9:
I get an answer:

`AAAB0400FB48CC55`

which indicated with a (wrong?) code AB that I have "learned" some data with protocol index 0.  **Looks good but whats next??**

I got the feeling that the achknowledge commands are swapped: Success is AB and failure is AA??

Portisch, 2019-03-27:
Looks like the protocol tolerance of the default protocol 0 is too big. If you don't need it disabled it in the config: EFM8BB1_SUPPORT_PT226X_PROTOCOL

haegar33, 2019-03-27:
Ok, I realised the **A8 command** is for sending Rf codes from the predefined protocols after sniffing or learning. Actually the REV plugs are all consistent with protocol 7 and I could decode all needed data ;) Thus for these plugs no further patching needed.

I got another set from Brennestuhl which do not even react to any A6 sniffing. Here I probably would need B1 and B0 raw sniffing which I might try later....

However the response codes for A9 learning is still the other way round as described in the wiki. I get a success code AB with info about protocol and data. I get AAAA55 for a failure. A correction and some more guidance  in the wiki  would be appreciated.

Thanks, Haegar.

Portisch, 2019-03-27:
Corrected wiki, thx!

closed 2019-03-27 by Portisch