Portisch/RF-Bridge-EFM8BB1 #109: Decode and mimic Rfraw 166 and Rfraw 177 codes from Dooya DC306 remote

mioke77, 2019-03-30:
### Latest firmware used?
RF Bridge running 6.5.0 with the latest Portisch firmware upgrade, all upgrades successful.

### Information
I have a DC306 Dooya Remote, 5-channel.

### Sniffed data
in one of the remote channel:
------------------------------------------------------------------------------------------------
CMD: rfraw 166
MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}

- down command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9523355"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9523355"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9523C55"}}

- up command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9521155"}}

- stop command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9525555"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9525555"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606056531E9525555"}}

------------------------------------------------------------------------------------------------
CMD: rfraw 177
MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}

- down command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 014A 02DA 26E8 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3B2B2A3A3B2B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05DC 014A 02DA 2710 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3B2B2A3A3B2B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05E6 0140 02DA 26F2 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3B2B2A3A3B2B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05E6 014A 02E4 2742 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3B2B2A3A3B2B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05DC 0154 02DA 26F2 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3B2B2A3A3B2B 55"}}

- up command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1270 05FA 0136 02F8 26E8 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3A3B2A3A3A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05DC 0154 02DA 2738 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3A3B2A3A3A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 0154 02D0 26F2 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3A3B2A3A3A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05E6 0154 02D0 2710 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3A3B2A3A3A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05DC 0154 02D0 26F2 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3A3A3B2A3A3A3B 55"}}

- stop command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05DC 0154 02D0 26E8 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3B2A3B2A3B2A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 014A 02D0 2710 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3B2A3B2A3B2A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05DC 014A 02DA 26E8 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3B2A3B2A3B2A3B 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05DC 0154 02DA 2742 481A3B2B2A3A3B2A3B2A3A3B2B2A3A3A3B2B2B2B2A3B2A3A3B2A3B2A3B2A3A3B2A3A3B2A3B2A3B2A3B 55"}}
------------------------------------------------------------------------------------------------

Any help is highly appreciated!

Thanks and Kind Regards!

mioke77, 2019-04-06:
Latest firmware used?
RF Bridge running 6.5.0 with the latest Portisch firmware comes with Sonoff-Tasmota-6.5.0 package (RF-Bridge-EFM8BB1-20181127.hex) upgrade, all upgrades successful.

Information
I have a DC306 Dooya Remote, 5-channel.

Sniffed data
in one of the remote channel:

09:39:00 CMD: rfraw 166
09:39:00 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
09:39:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}

down command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9523355"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9523C55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9523355"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9523C55"}}

up command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521E55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521155"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9521E55"}}

stop command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9525555"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9525555"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9525555"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA606016531E9525555"}}

-------------------------
CMD: rfraw 177
MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}

down command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05F0 0154 02DA 26F2 0123323223233223322323323223232332323232233223233223322332232332232323323223233234 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05F0 0154 02DA 26F2 0123323223233223322323323223232332323232233223233223322332232332232323323223233234 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05F0 0154 02DA 26F2 0123323223233223322323323223232332323232233223233223322332232332232323323223233234 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 0154 02E4 26F2 0123323223233223322323323223232332323232233223233223322332232332232323323223233234 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05F0 0154 02E4 26F2 0123323223233223322323323223232332323232233223233223322332232332232323323223233234 55"}}

up command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05F0 0154 02DA 26F2 0123323223233223322323323223232332323232233223233223322332232332232323233223232334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 015E 02D0 26F2 0123323223233223322323323223232332323232233223233223322332232332232323233223232334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1284 05E6 015E 02DA 26E8 0123323223233223322323323223232332323232233223233223322332232332232323233223232334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 0154 02E4 26E8 0123323223233223322323323223232332323232233223233223322332232332232323233223232334 55"}}

stop command:
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1252 062C 010E 032A 26F2 0123323223233223322323323223232332323232233223233223322332232332232332233223322334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05E6 0154 02EE 26F2 0123323223233223322323323223232332323232233223233223322332232332232332233223322334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 1248 0636 010E 032A 26FC 0123323223233223322323323223232332323232233223233223322332232332232332233223322334 55"}}
MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 128E 05F0 0154 02DA 26FC 0123323223233223322323323223232332323232233223233223322332232332232332233223322334 55"}}

mioke77, 2019-04-06:
@Portisch why there are differences for rfraw 177 when I used different firmware, is normal? what firmware should I use? is this new protocol?
Many Thanks!

mioke77, 2019-04-06:
Resolved after I checked this:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23#
It is the same remote control vendor, same protocol but different remote control model.

Many thanks @Portisch ! the only question remain why rfraw sniffed data with rfraw 177, it seems to be wrong with the latest firmware 20190326. Maybe @Portisch will clarify this?

closed 2019-04-06 by mioke77

BaronorOmnia, 2023-12-02:
Beste,

I new with this product and have the next thing running:
00:00:00.002 HDW: ESP8285N08
00:00:00.046 CFG: Loaded from flash at FB, Count 29
00:00:00.199 QPC: Reset
00:00:00.217 SER: Set to 8N1 19200 bit/s
00:00:00.221 Project tasmota - Tasmota Version 13.2.0(tasmota)-2_7_4_9(2023-10-19T09:02:07)
00:00:00.811 WIF: Connecting to AP1 Online Channel 1 BSSId F0:9F:C2:37:C1:03 in mode 11n as tasmota-B47183-4483...
00:00:01.754 WIF: Connected
00:00:02.010 HTP: Web server active on tasmota-B47183-4483 with IP address 192.168.2.18
17:05:19.026 RSL: INFO1 = {"Info1":{"Module":"Sonoff Bridge","Version":"13.2.0(tasmota)","FallbackTopic":"cmnd/DVES_B47183_fb/","GroupTopic":"cmnd/tasmotas/"}}
17:05:19.028 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-B47183-4483","IPAddress":"192.168.2.18"}}
17:05:19.030 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":13}}
17:05:23.438 RSL: STATE = {"Time":"2023-12-02T17:05:23","Uptime":"0T00:00:08","UptimeSec":8,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Online","BSSId":"F0:9F:C2:37:C1:03","Channel":1,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:05:42.715 CMD: rfraw
17:05:42.719 RSL: RESULT = {"RfRaw":"OFF"}
17:05:47.032 CMD: rffaw 1
17:05:47.038 RSL: RESULT = {"Command":"Unknown"}
17:05:54.464 CMD: rfraw 177\
17:05:54.469 RSL: RESULT = {"RfRaw":"ON"}
17:05:58.526 CMD: rfraw 1
17:05:58.531 RSL: RESULT = {"RfRaw":"ON"}
17:06:04.044 CMD: rfraw 166
17:06:04.049 RSL: RESULT = {"RfRaw":"ON"}

But I don't see any command.  Can some help my on the right way ?

BaronorOmnia, 2023-12-19:
@Portisch  can you have a look and help me ?