Portisch/RF-Bridge-EFM8BB1 #112: Home Decorators Collection Fan from Home Depot Remote Decoding

charcharwon, 2019-04-11:
I'm just a hobbyist and cannot figure out this remote, is very fancy and sends out a lot of lines of code with every button press.
when I press the light button once
19:39:37 CMD: rfraw 177
19:39:37 MQT: stat/sonoff-rf/RESULT = {"RfRaw":"ON"}
19:39:37 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AAA055"}}
19:39:42 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0276 26D4 280918091818091809090918091809090909090918091809090918 55"}}
19:39:42 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 010E 02B2 26E8 381A280A28280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A28 55"}}
19:39:42 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02B2 26D4 280918091818091809090918091809090909090918091809090918 55"}}
19:39:42 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 010E 02B2 26D4 381A280A28280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A28 55"}}

when I press it again

19:39:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 026C 26D4 280918091818091809090918091809090909090918091809090918 55"}}
19:39:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 010E 02B2 26E8 381A280A28280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A28 55"}}
19:39:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02C6 26D4 381A080A28080A080A1A0A081A280A0A0A0A0A0A280A280A0A0A28 55"}}
19:39:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02A8 26DE 381A280A08280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A28 55"}}
19:39:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02A8 26CA 280918091818091809090918091809090909090918091809090918 55"}}

I disabled the dimming function because it would even contain more lines when pressed once.
Any guidance on what directions to go with this data?

The fan power i get from one press
19:43:11 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0226 00E6 032A A08181 55"}}
19:43:11 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 029E 272E 280918091818091809090918091809090909090909091809090909 55"}}
19:43:11 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0190 00FA 02DA 26E8 381A081A08081A081A1A1A081A081A1A1A1A0A0A0A0A080A0A1A0A 55"}}
19:43:12 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 01B8 00DC 02DA 26DE 381A0818080818081818180818081818181818181818081A181818 55"}}
19:43:12 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02BC 26D4 381A280A28281A280A0A0A280A280A0A0A0A0A0A0A0A280A0A0A0A 55"}}
19:43:12 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02B2 26FC 280918091818091809090918091809090909090909091809090909 55"}}
19:43:12 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02C6 26DE 381A080A08081A280A0A0A280A081A0A0A0A0A0A0A0A280A0A0A0A 55"}}
19:43:12 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 02A8 26E8 280918091818091809090918091809090909090909091809090909 55"}}
19:43:13 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02A8 26E8 381A280A28081A281A0A0A281A081A0A0A0A0A0A0A0A280A0A0A0A 55"}}
19:43:13 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0168 029E 0104 26D4 380918091818291809090918091809090909090909091809090909 55"}}

and one press again

19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 010E 02C6 26DE 381A280A28280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A0A 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02C6 26D4 381A280A08081A281A1A0A280A281A0A1A0A0A0A280A280A0A0A0A 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 0104 02C6 26DE 381A081A08081A081A1A1A281A280A0A0A0A0A0A280A280A0A0A0A 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 02B2 010E 26D4 380918291818291809090918091809090909090918091809090909 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 0104 02BC 26F2 381A080A28280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A0A 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02BC 26F2 280918091818091809090918091809090909090918091809090909 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 02B2 010E 26D4 380918291818091809090918091809090909090918091809090909 55"}}
19:43:52 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 0294 26DE 280918091818091809090918091809090909090918091809090909 55"}}
19:43:53 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 015E 02A8 00FA 26D4 380918291818091809090918091809090909090918091809090909 55"}}
19:43:53 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02B2 26F2 280918091818091809090918091809090909090918091809090909 55"}}
19:43:53 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02BC 26E8 381A281A28280A280A1A0A281A280A0A0A0A0A0A280A280A0A0A0A 55"}}
19:43:53 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 02A8 26DE 280918091818091809090918091809090909090918091809090909 55"}}
19:43:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 010E 02B2 26DE 381A281A08280A280A0A0A280A280A0A0A0A0A0A280A280A0A0A0A 55"}}
19:43:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02BC 26DE 280918091818091809090918091809090909090918091809090909 55"}}
19:43:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0190 0104 02BC 26E8 381A081A08081A080A1A0A080A280A0A0A1A1A0A280A280A0A0A0A 55"}}
19:43:54 MQT: tele/sonoff-rf/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 0104 02C6 26E8 381AAAB104019000E602BC26DE381A081A08081A081A1A0A080A280A0A0A0A1A1A280A281A0A0A0A 55"}}

https://www.homedepot.com/p/Home-Decorators-Collection-Breezemore-56-in-LED-Indoor-Mediterranean-Bronze-Ceiling-Fan-with-Light-Kit-and-Remote-Control-51556/301161776

Thank you for reading my post

Portisch, 2019-04-12:
Try with this new branch:
https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/04c281384c78066b647ca6291784fb8d05257dc0/Keil%208051%20v9.53%20-%20Release
Now you should receive the commands with [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)

charcharwon, 2019-04-12:
Thank you Portisch, I let you know the results.

charcharwon, 2019-04-12:
Got the codes, but I think I messed up my Sonoff soldering sending the codes didn't get reactions from the fan, I thought it was firmware related. I flashed back the main one and try using it with my DC90 device that I know works and now the same problem. Looks like I need another Sonoff before I can test fan codes :(.

Power on cool AAA605015A28140055

Power off cool AAA605015A28040055

Speed cool
AAA605015A28140055
AAA605015A28240055
AAA605015A28340055
AAA605015A28440055
AAA605015A28540055
AAA605015A28640055
AAA605015A28740055
AAA605015A28840055
AAA605015A28940055

Breeze mode
b1 AAA605015A28B40055
b2 AAA605015A28C40055
b3 AAA605015A28D40055
b0 AAA605015A28900055

Timer mode
2H-b3 AAA605015A28D40055
4H-b3 AAA605015A28D6C055
8H-b3 AAA605015A28D6C055
Timer Off b3 AAA605015A28D68055

2H-b0 AAA605015A28928055
4H-b0 AAA605015A2892C055
8H-b0 AAA605015A28930055
Timer Off b0 AAA605015A28924055

2H-b1 AAA605015A28B68055
4H-b1 AAA605015A28B6C055
8H-b1 AAA605015A28B70055
Timer Off b1 AAA605015A28B64055

2H-b2 AAA605015A28C68055
4H-b2 AAA605015A28C6C055
8H-b2 AAA605015A28C70055
Timer Off b2 AAA605015A28C64055

breeze mode cool AAA605015A28940055
warm mode	 AAA605015A28980055

light on/off AAA605015A28990055

brightness
decrease AAA605015A2898C055

increase AAA605015A28988055

charcharwon, 2019-04-12:
Okay, so the bridge is not dead. It seems the newest master transmitting my DC90 remote code also doesn't do anything, flashing the one I had been using since February restored the DC90 controlling my awning.

Portisch, 2019-04-12:
On the new firmware I disabled DC90. So download Simplicity studio and make your own firmware.

charcharwon, 2019-04-12:
> On the new firmware I disabled DC90. So download Simplicity studio and make your own firmware.

Ah, ok thank you for the quick reply. Will do that.
What about transmitting the fan codes and not getting any reaction from the fan?

Portisch, 2019-04-12:
When you send it you hopefully used command 0xA8? Otherwise please study the Wiki.

charcharwon, 2019-04-12:
> When you send it you hopefully used command 0xA8? Otherwise please study the Wiki.

Thank you for the hint for the proper formatting, I remember reading about when I was trying to get the DC90 to work, but it worked without then so was spoiled by that experience.

Keep up the good work, been wasting so much time since February on this fan and finally, it is working.

closed 2019-04-14 by charcharwon