Portisch/RF-Bridge-EFM8BB1 #115: Simple doorbell knob.

bakman2, 2019-04-24:
### Latest firmware used?
- Yes. 
- Sonoff-Tasmota 6.5.0

### Information
Doorbell knob

### Sniffed data
```
17:41:55 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 169E 280909181809091809181809180918181818 55"}}
17:41:56 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 168A 280909181809091809181809180918181818 55"}}
17:41:57 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 0172 1694 280909181809091809181809180918181818 55"}}
17:41:58 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0168 1680 280909181809091809181809180918181818 55"}}
17:41:58 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 1694 280909181809091809181809180918181818 55"}}
17:41:59 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 168A 280909181809091809181809180918181818 55"}}
17:42:00 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 168A 280909181809091809181809180918181818 55"}}

```

This is a simple doorbell, I was hoping this was supported. In RF_Config.h I see a doorbell that is disabled, is this the one by any chance ? If so how can this be enabled ?

Portisch, 2019-04-24:
try this: https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/Doorbell_Simple/Keil%208051%20v9.53%20-%20Release

https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Compile-this-project

bakman2, 2019-04-24:
Thanks for the superquick response. 

No success unfortunately:

I see my wall switch and motion PIR:
```
18:20:21 CMD: rfraw 0
18:20:21 MQT: stat/sonoff-rf-bridge/RESULT = {"RfRaw":"OFF"}
18:20:24 MQT: tele/sonoff-rf-bridge/RESULT = {"RfReceived":{"Sync":10400,"Low":350,"High":1060,"Data":"EE8B88","RfKey":"None"}}
18:20:28 MQT: tele/sonoff-rf-bridge/RESULT = {"RfReceived":{"Sync":10410,"Low":340,"High":1060,"Data":"EE8B88","RfKey":"None"}}
18:20:35 MQT: tele/sonoff-rf-bridge/RESULT = {"RfReceived":{"Sync":12530,"Low":380,"High":1190,"Data":"ED59CE","RfKey":1}}
18:22:01 CMD: rfraw 1
18:22:01 MQT: stat/sonoff-rf-bridge/RESULT = {"RfRaw":"ON"}
18:22:04 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA428B4014A0410EE8B8855"}}
18:22:29 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA430E8017C04A6ED59CE55"}}
18:22:29 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA430AC019A04C4ED59CE55"}}
```
Here comes the doorbell - `280909181809091809181809180918181818` (with other devices broadcasting):
```
18:22:42 CMD: rfraw 177
18:22:42 MQT: stat/sonoff-rf-bridge/RESULT = {"RfRaw":"ON"}
18:22:42 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
18:22:50 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 015E 16BC 280909181809091809181809180918181818 55"}}
18:22:51 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 046A 0172 1694 280909181809091809181809180918181818 55"}}
18:22:51 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 0442 0442 281809 55"}}
18:22:51 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16A8 280909181809091809181809180918181818 55"}}
18:22:52 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 017C 042E 0442 281809 55"}}
18:22:52 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 169E 280909181809091809181809180918181818 55"}}
18:22:53 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 015E 169E 280909181809091809181809180918181818 55"}}
18:22:54 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 017C 168A 280909181809091809181809180918181818 55"}}
18:22:55 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 017C 1694 280909181809091809181809180918181818 55"}}
18:23:01 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00DC 0226 041A 0096 0398 48182818082B18 55"}}
18:23:04 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00D2 030C 021C 021C 381828 55"}}
18:23:08 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 0244 042E 281818 55"}}
18:23:08 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 02F8 021C 01EA 381828 55"}}
18:23:10 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 02 00DC 024E 180808 55"}}
18:23:11 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 0406 03F2 280818 55"}}
18:23:11 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 385E 0410 0438 381A28 55"}}
18:23:13 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00A0 0140 029E 0334 38181828 55"}}
```

bakman2, 2019-04-24:
btw can this be compiled with arduino in stead of simplicity studio ? The latter requires registration and their privacy policy does not conform to gdpr and to myself.

Portisch, 2019-04-24:
You tried 0xA6? Look at Wiki!

bakman2, 2019-04-24:
When using rfraw 166

```
19:40:03 CMD: rfraw 166
19:40:03 MQT: stat/sonoff-rf-bridge/RESULT = {"RfRaw":"ON"}
19:40:03 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
19:40:06 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60400EE8B8855"}}
19:40:16 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
19:40:17 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
19:40:18 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
19:40:19 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
19:40:20 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
19:40:21 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AAA60401CD280055"}}
```
First one is a switch, others are doorbell. Motion PIR does not show up. (is a sonoff PIR)

bakman2, 2019-04-27:
Something else I can do ?
I will try to produce some output with the converter.

bakman2, 2019-05-01:
Output from the converter:

```
###### Processing line 17:44:26 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16C6 280909181809091809181809180918181818 55
###### Processing line 17:44:27 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16C6 280909181809091809181809180918181818 55
###### Processing line 17:44:27 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16BC 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16BC28090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16BC 280909181809091809181809180918181818 55
###### Processing line 17:44:29 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 0172 16E4 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030456017216E428090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0456 0172 16E4 280909181809091809181809180918181818 55
###### Processing line 17:44:30 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16BC 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16BC28090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16BC 280909181809091809181809180918181818 55
###### Processing line 17:44:31 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16BC 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16BC28090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16BC 280909181809091809181809180918181818 55
```

Oscilloscope

[image]

Portisch, 2019-05-02:
So you have 3 devices?

> First one is a switch, others are doorbell. Motion PIR does not show up. (is a sonoff PIR)
Make from each device a rfraw 177 sniffing. Right now there is only PT226x & your doorbell included.

bakman2, 2019-05-02:
My test setup are 3 devices (might have noise of neighbor stuff):

```
###### Processing line 10:36:22 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 06 00BE 0140 023A 3886 03FC 044C 581818283C 55"}}
 ######
Filtered 0xB1 data: AAB10600BE0140023A388603FC044C581818283C55
Resulting 0xB0 data: AA B0 13 06 08 00BE 0140 023A 3886 03FC 044C 581818283C 55
###### Processing line 10:36:26 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0212 02D0 280818 55"}}
 ######
Filtered 0xB1 data: AAB10300D2021202D028081855
Resulting 0xB0 data: AA B0 0B 03 08 00D2 0212 02D0 280818 55
###### Processing line 10:36:26 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0154 28FA 28181909081908181819090819081818181909090909090819 55"}}
 ######
Filtered 0xB1 data: AAB1030438015428FA2818190908190818181909081908181818190909090909081955
Resulting 0xB0 data: AA B0 21 03 08 0438 0154 28FA 28181909081908181819090819081818181909090909090819 55
###### Processing line 10:36:27 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0136 2968 28181909081908181819090819081818181909090909081909 55"}}
 ######
Filtered 0xB1 data: AAB1030438013629682818190908190818181909081908181818190909090908190955
Resulting 0xB0 data: AA B0 21 03 08 0438 0136 2968 28181909081908181819090819081818181909090909081909 55
###### Processing line 10:36:31 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 0136 28B4 28181819081818190819090908190818181909090819090909 55"}}
 ######
Filtered 0xB1 data: AAB1030456013628B42818181908181819081909090819081818190909081909090955
Resulting 0xB0 data: AA B0 21 03 08 0456 0136 28B4 28181819081818190819090908190818181909090819090909 55
###### Processing line 10:36:37 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 04E2 0190 30E8 28181819081819081908190818190908181819090818181909 55"}}
 ######
Filtered 0xB1 data: AAB10304E2019030E82818181908181908190819081819090818181909081818190955
Resulting 0xB0 data: AA B0 21 03 08 04E2 0190 30E8 28181819081819081908190818190908181819090818181909 55
###### Processing line 10:36:38 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 04EC 0186 30C0 28181819081819081908190818190908181819090818181909 55"}}
 ######
Filtered 0xB1 data: AAB10304EC018630C02818181908181908190819081819090818181909081818190955
Resulting 0xB0 data: AA B0 21 03 08 04EC 0186 30C0 28181819081819081908190818190908181819090818181909 55
###### Processing line 10:36:46 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0154 1716 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB10304420154171628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0442 0154 1716 280909181809091809181809180918181818 55
###### Processing line 10:36:47 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 0442 044C 281809 55"}}
 ######
Filtered 0xB1 data: AAB10301720442044C28180955
Resulting 0xB0 data: AA B0 0B 03 08 0172 0442 044C 281809 55
###### Processing line 10:36:47 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017216C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 0172 16C6 280909181809091809181809180918181818 55
###### Processing line 10:36:47 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 044C 044C 281809 55"}}
 ######
Filtered 0xB1 data: AAB1030172044C044C28180955
Resulting 0xB0 data: AA B0 0B 03 08 0172 044C 044C 281809 55
###### Processing line 10:36:47 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017C16C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 017C 16C6 280909181809091809181809180918181818 55
###### Processing line 10:36:48 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017216C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0460 0172 16C6 280909181809091809181809180918181818 55
###### Processing line 10:36:48 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 044C 0456 281809 55"}}
 ######
Filtered 0xB1 data: AAB1030168044C045628180955
Resulting 0xB0 data: AA B0 0B 03 08 0168 044C 0456 281809 55
###### Processing line 10:36:49 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 017C 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030456017C16C628090918180909180918180918091818181855
Resulting 0xB0 data: AA B0 1A 03 08 0456 017C 16C6 280909181809091809181809180918181818 55
###### Processing line 10:36:49 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 044C 044C 281809 55"}}
 ######
Filtered 0xB1 data: AAB1030172044C044C28180955
Resulting 0xB0 data: AA B0 0B 03 08 0172 044C 044C 281809 55
###### Processing line 10:36:49 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16C6 280909181809091809181809180918181818 55"}}
 ######
Filtered 0xB1 data: AAB1030460017216C628090918180909180918180918091818181855
[... 5 lines cut ...]
###### Processing line 10:36:50 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 0456 0456 281809 55"}}
 ######
Filtered 0xB1 data: AAB10301680456045628180955
[... cut ...]
###### Processing line 10:36:54 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 0334 021C 280818 55"}}
 ######
Filtered 0xB1 data: AAB10300BE0334021C28081855
Resulting 0xB0 data: AA B0 0B 03 08 00BE 0334 021C 280818 55
###### Processing line 10:36:57 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00BE 0136 0226 042E 3818281818 55"}}
 ######
Filtered 0xB1 data: AAB10400BE01360226042E381828181855
Resulting 0xB0 data: AA B0 0F 04 08 00BE 0136 0226 042E 3818281818 55
```

output (much shorter somehow)

[image]

Portisch, 2019-05-02:
I will need a scan for each device - seperate them, I can't find what device is which code?

bakman2, 2019-05-02:
My apologies:

Doorbell:
```
10:55:52 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16C6 280909181809091809181809180918181818 55"}}
10:55:53 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16B2 280909181809091809181809180918181818 55"}}
10:55:54 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16BC 280909181809091809181809180918181818 55"}}
10:55:55 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 015E 1702 280909181809091809181809180918181818 55"}}
10:55:55 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0172 16BC 280909181809091809181809180918181818 55"}}
10:55:56 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 0172 16C6 280909181809091809181809180918181818 55"}}
10:55:57 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 046A 0172 16DA 280909181809091809181809180918181818 55"}}
10:55:57 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0456 0172 16C6 280909181809091809181809180918181818 55"}}
10:55:58 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16C6 280909181809091809181809180918181818 55"}}
10:55:59 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 017C 16BC 280909181809091809181809180918181818 55"}}
```

Wall switch with 2 switches:

Left button:
```
10:52:57 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 014A 28F0 28181909081908181819090819081818181909090909090819 55"}}
10:52:59 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0140 290E 28181909081908181819090819081818181909090909090819 55"}}
10:53:00 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0140 2918 28181909081908181819090819081818181909090909090819 55"}}
10:53:02 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0140 2918 28181909081908181819090819081818181909090909090819 55"}}
10:53:04 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0140 2918 28181909081908181819090819081818181909090909090819 55"}}
```
Right button:
```
10:53:26 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0140 2968 28181909081908181819090819081818181909090909081909 55"}}
10:53:27 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0140 2972 28181909081908181819090819081818181909090909081909 55"}}
10:53:29 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0140 2968 28181909081908181819090819081818181909090909081909 55"}}
10:53:30 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0140 2972 28181909081908181819090819081818181909090909081909 55"}}
10:53:32 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0438 0140 2968 28181909081908181819090819081818181909090909081909 55"}}
```

Wall switch (same brand) with 1 switch:
```
10:54:14 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 046A 012C 2878 28181819081818190819090908190818181909090819090909 55"}}
10:54:16 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0140 28AA 28181819081818190819090908190818181909090819090909 55"}}
10:54:17 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 014A 28AA 28181819081818190819090908190818181909090819090909 55"}}
10:54:18 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0140 28B4 28181819081818190819090908190818181909090819090909 55"}}
10:54:19 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0460 0140 28B4 28181819081818190819090908190818181909090819090909 55"}}
```

Sonoff motion PIR (emits 2 codes, on/off?):
```
10:49:32 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 04E2 0190 30F2 28181819081819081908190818190908181819090818181909 55"}}
10:49:33 MQT: tele/sonoff-rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 04EC 0186 30B6 28181819081819081908190818190908181819090818181909 55"}} 
```

Portisch, 2019-05-02:
ok, Doorbell is Doorbell protocol
Wall switch & PIR is PT226x protocol.
You should be sniffing Doorbell & Wall switch in mode 0xA6.
You should be sniffing Wall switch & PIR in default mode 0xA4 (just reboot the rfbridge)

This is because the PIr does have another timing.
Wall switch: 350, 1050, 10850
PIR: 400, 1200, 12400

There is right now no solution to generic decode PT226x protocol with 0xA6.

Try this new firmware with 0xA6 sniffing:
https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/Doorbell_Simple/Keil%208051%20v9.53%20-%20Release

bakman2, 2019-05-02:
Thanks again for the quick support, with 0xA6 and this firmware I can receive all devices. 

I will monitor stability of the bridge (in other modes it just stops processing rf data after a couple of hours)

closed 2019-05-02 by bakman2