Portisch/RF-Bridge-EFM8BB1 #116: PT2262 Remote with Dip Switches

macavelly, 2019-04-28:
### Latest firmware used?
I'm currently using the latest firmware located here.
https://github.com/Portisch/RF-Bridge-EFM8BB1/releases

### Information
What device you try to sniff data from? Description? Brand and type?
I have the Sonoff RF bridge with the latest tasmota (Sonoff-Tasmota 6.5.0 )

The remote I'm trying to learn has a PT2262 chip in it as well as dip switches.
[image]
[image]

I've tried RfRaw 166 and there is no response posted, the red led on the bridge does not light up
I;ve tried RfRaw 169 and again no response

But if I do RfRaw 177, the bridge responds to the remote and gives me the B1 data. I've taken all the B1 data and using the B1 to B0 converted tool and got the B0 data. I then manually input all the B0 data into the console one at a time to see if I get the gate to close but none of the B0 commands work

### Sniffed data B1 data

```
01:35:40 MQT: stat/rfbridge/RESULT = {"RfRaw":"ON"}
01:35:40 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
01:35:43 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00BE 014A 1FAE B0909090928282818282809090909280928282828090918282 55"}}
01:35:43 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00E6 1FB8 A0909090918181818181809090909180918181818090918181 55"}}
01:35:43 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 037A 1FB8 0398 381809090A 55"}}
01:35:44 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00DC 1FB8 A0909090918181818181809090909180918181818090918181 55"}}
01:35:44 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00BE 0136 1FE0 B0909090918282828281809090909180918181818090918281 55"}}
01:35:45 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 0398 0154 1FAE 0398 481829292B 55"}}
01:35:45 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00C8 0140 1FC2 B0909090918281828181809090909180918181818090918181 55"}}
01:35:45 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 037A 00E6 0186 1FCC B09090909282A2818181809090909180918181818090918181 55"}}
01:35:45 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 0154 1FCC B0909090918181818281809090909180918181818090918181 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00DC 1FAE A0909090918181818181809090909180918181818090918181 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0366 1FCC 03C0 381809090A 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00E6 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 0370 1FE0 03A2 381809090A 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0366 0104 1FE0 A0909090918181818181809090909180918181818090918181 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 037A 1FD6 0398 381809090A 55"}}
01:35:46 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00E6 1FC2 A0909090918181818181809090909180918181818090918181 55"}}
01:35:48 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 03AC 015E 1FAE 0398 481829292B 55"}}
01:35:48 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00D2 017C 1FB8 B0909090918181818181809090909180918181818090918182 55"}}
01:35:48 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00C8 014A 1FAE B0909090928282828282809090909280918282818090918282 55"}}
01:35:48 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 03D4 0096 015E 00FA 1F9A C09090909282828282828090909092809282828280B0B28282 55"}}
01:35:48 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00D2 1FEA A0909090918181818181809090909180918181818090918181 55"}}
01:35:49 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00D2 015E 1FE0 B0909090918181828181809090909180918181818090918181 55"}}
01:35:50 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0398 00D2 1FEA A0909090918181818181809090909180918181818090918181 55"}}
01:35:50 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 0154 1FAE B0909090918181818182809090909180918181818090918181 55"}}
01:35:50 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00DC 1FEA A0909090918181818181809090909180918181818090918181 55"}}
01:35:50 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 038E 00D2 0154 0294 1FCC C09090909181818181818090909091809282B2828090928281 55"}}
01:35:51 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 00B4 0140 1FE0 B0909090928282828282809090909280928282828090918282 55"}}
01:35:51 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 037A 00F0 0078 1FCC B09090909181818181818090A0A0A180918181818090918181 55"}}
01:35:51 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00C8 0154 1FAE B0909090928282828182809090909180918282828090928182 55"}}
01:35:51 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1FCC A0909090918181818181809090909180918181818090918181 55"}}
01:35:51 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0384 00E6 0172 0096 1FCC C0909090918181818181809090909180918182828090B18182 55"}}
01:35:52 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00F0 1FE0 A0909090918181818181809090909180918181818090918181 55"}}
01:35:52 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 00B4 0140 1FE0 B0909090928282828282809090909280928282828090928182 55"}}
01:35:52 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 037A 1FF4 03AC 381809090A 55"}}
01:35:53 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00E6 1FCC A0909090918181818181809090909180918181818090918181 55"}}
01:35:54 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 00BE 0136 1FB8 B0909090918281828282809090909280928281818090918282 55"}}
01:35:54 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00E6 006E 1FCC B090909091818181818180A0A0909180918181818090918181 55"}}
01:35:54 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00E6 1FE0 A0909090918181818181809090909180918181818090918181 55"}}
01:35:55 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00F0 1FF4 A0909090918181818181809090909180918181818090918181 55"}}
01:35:55 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00DC 0384 015E 1FB8 03AC 481829290B 55"}}
01:35:55 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0370 00FA 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
01:35:56 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00B4 0136 1FEA B0909090928282828282809090909280928282828090928282 55"}}
01:35:56 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00F0 1FEA A0909090918181818181809090909180918181818090918181 55"}}
01:35:56 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 07 0096 0410 0064 017C 02B2 1FA4 03A2 681A3C3C3D 55"}}
01:35:57 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00DC 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
01:35:57 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 037A 1FE0 03AC 381809090A 55"}}
01:35:57 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00DC 1FCC A0909090918181818181809090909180918181818090918181 55"}}
01:35:57 MQT: tele/rfbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 0384 1FE0 0398 381809090A 55"}}
[... 83 lines cut ...]
Resulting 0xB0 data: AA B0 11 05 08 00D2 0398 0154 1FAE 03A2 481809290B 55
Resulting 0xB0 data: AA B0 21 03 08 0398 00D2 1FC2 A0909090918181818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 23 04 08 0398 00BE 0140 1FE0 B0909090928282828282809090909180918181828090928282 55
Resulting 0xB0 data: AA B0 23 04 08 0398 00C8 014A 1FCC B0909090918181828181809090909180918181818090918182 55
Resulting 0xB0 data: AA B0 21 03 08 0384 00D2 1FD6 A0909090918181818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 23 04 08 038E 00D2 0154 1FA4 B0909090918282818181809090909180918181828090918181 55
Resulting 0xB0 data: AA B0 21 03 08 0384 00DC 1FB8 A0909090918181818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 0F 05 08 03DE 0082 00D2 017C 02BC C090A3 55
Resulting 0xB0 data: AA B0 21 03 08 0384 00E6 1FCC A0909090918181818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 25 05 08 037A 00DC 006E 01AE 1FEA C09090A0A183B383838380A090909180918181818090918181 55
Resulting 0xB0 data: AA B0 21 03 08 038E 00DC 1FEA A0909090918181818181809090909180918181818090918181 55
[... cut ...]
Resulting 0xB0 data: AA B0 21 03 08 037A 00DC 1FEA A0909090918181818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 11 05 08 00D2 03A2 0168 1FAE 0398 481829292B 55
Resulting 0xB0 data: AA B0 23 04 08 038E 00C8 014A 1FCC B0909090918281818181809090909180918181818090918181 55
Resulting 0xB0 data: AA B0 21 03 08 0366 00F0 1FCC A0909090918181818181809090909180918181818090918181 55
```

I've tried everything I could think of, I initially posted an issue on the tasmota page but was referred here.

Thank you for any help.

macavelly, 2019-05-01:
I decided to give this hex file a try "https://github.com/Portisch/RF-Bridge-EFM8BB1/tree/FRM87/Keil%208051%20v9.53%20-%20Release" but it was the same result.

rfraw 166 gave no result only rfraw 177

I did notice something. If I press the button about 10 times (press, not hold) I get this result:

```
20:12:46 CMD: rfraw 177
20:12:46 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
20:12:46 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
20:12:59 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00D2 0154 1F9A B0909090918181828181809090909180918181818090918181 55"}}
20:13:04 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 0398 014A 1F86 03AC 481829292B 55"}}
20:13:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 015E 1FAE B0909090918281818282809090909180918181818090918181 55"}}
20:13:11 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00DC 015E 1FA4 B0909090928181818181809090909180918181818090918181 55"}}
20:13:13 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1F9A A0909090918181818181809090909180918181818090918181 55"}}
20:13:15 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03B6 00A0 014A 1FC2 B0909090928282828282809090909280928282828090928282 55"}}
20:13:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00D2 0398 0154 1F9A 0398 481829290B 55"}}
20:13:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00B4 0136 1F9A B0909090928282828282809090909280928282828090918282 55"}}
20:13:23 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00C8 038E 0172 1FA4 03A2 481829292B 55"}}
20:13:25 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00D2 0154 1FA4 B0909090928181818181809090909180918181818090918181 55"}}
20:13:28 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00D2 1F9A A0909090918181818181809090909180918181818090918181 55"}}
20:13:29 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 037A 1FA4 038E 381809090A 55"}}
20:13:38 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00D2 1FCC A0909090918181818181809090909180918181818090918181 55"}}
20:13:40 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00DC 038E 0172 1F90 0398 481809092B 55"}}
20:13:45 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00A0 03A2 0168 1F90 03A2 481829292B 55"}}
20:13:46 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00C8 014A 1FAE B0909090928282828282809090909180918181828090918181 55"}}
20:13:51 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 038E 0140 1FA4 03A2 481829292B 55"}}
20:13:51 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00BE 0136 1FA4 B0909090928282828181809090909280928281828090918181 55"}}
20:13:53 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0366 00E6 1FB8 A0909090918181818181809090909180918181818090918181 55"}}
20:13:54 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 0398 0140 1F90 03AC 481809292B 55"}}
20:13:56 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 038E 017C 1F90 03AC 481829292B 55"}}
20:13:56 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00C8 014A 1FA4 B0909090928282828181809090909180918182828090918182 55"}}
20:14:01 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 0398 0168 1F9A 0384 481829292B 55"}}
20:14:01 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00D2 1FAE A0909090918181818181809090909180918181818090918181 55"}}
20:14:05 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00E6 1F90 A0909090918181818181809090909180918181818090918181 55"}}
20:14:06 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 03AC 015E 1F90 0398 481829292B 55"}}
20:14:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00D2 0154 1FC2 B0909090918181818182809090909180918181818090918181 55"}}
20:14:11 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00D2 038E 0154 1F90 03AC 481809092B 55"}}
20:14:13 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 0398 0168 1F86 0398 481829292B 55"}}
20:14:13 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1FAE A0909090918181818181809090909180918181818090918181 55"}}
20:14:16 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00BE 0140 1F86 B0909090928282828282809090909280918181828090918282 55"}}
20:14:18 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 0398 0172 1F9A 0398 481829292B 55"}}
20:14:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 038E 0154 1F90 03A2 481829292B 55"}}
20:14:22 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00C8 0398 0154 1FA4 03A2 481829292B 55"}}
20:14:25 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 0370 1FC2 03C0 381809090A 55"}}
20:14:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00D2 0154 1F68 B0909090928282828181809090909180918181818090918181 55"}}
20:14:32 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1FAE A0909090918181818181809090909180918181818090918181 55"}}
20:14:34 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 03B6 0172 1F90 0398 481829292B 55"}}
20:14:34 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00C8 0168 1F9A B0909090928281818181809090909180928282818090928181 55"}}
20:14:35 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00BE 03AC 0154 1F9A 0398 481829292B 55"}}
20:14:38 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00C8 0154 1F9A B0909090928181818182809090909280928181828090918182 55"}}
[... cut ...]
20:19:43 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 00B4 0154 1FCC B0909090928282828282809090909280928282828090918281 55"}}
20:19:43 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03C0 00AA 0140 1FD6 B0909090928282828282809090909280928282828090928282 55"}}
20:19:43 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0366 00F0 1FC2 A0909090918181818181809090909180918181818090918181 55"}}
```

The main thing I noticed is that just quick pressing the button I get a lot more  5 bucket data, Not sure if that makes any sense. Any help in trying to figure this out would be appreciated. Thanks!

Portisch, 2019-05-02:
Your remote is using the PT226x protocol, yes - but it is inverted!!
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/PT2262/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Your original receiver for the remote is working?

macavelly, 2019-05-02:
Hi, thanks for your reply. Yes the original receiver (a garage rolling gate) is working with the remote.
I loaded the new hex file and did a "rfraw 166"

This was the response for the down button
```
09:59:18 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60401F03D0C55"}}
09:59:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60401F03D0C55"}}
09:59:48 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60401F03D0C55"}}
15:28:52 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60401F03D0C55"}}
```

However if I send the command via the console as: 
`rfraw AAA80401F03D0C55`
The gate does not go down or respond at all.

I tried "rfraw 169" and I get a AB answer.
```
15:50:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAAB0400F03D0C55"}}
```

If it helps I also redid the rfraw 177 and pressed the button about 3 times, holding it down for about 3 seconds each.

```
15:51:12 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
15:51:12 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
15:51:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00C8 0154 1FA4 B0909090928282828282809090909180918182828090928181 55"}}
15:51:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 00AA 0136 1FB8 B0909090928282828282809090909280928282828090928282 55"}}
15:51:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00E6 1FCC A0909090918181818181809090909180918181818090918181 55"}}
15:51:18 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00C8 0140 1FF4 B0909090928281828182809090909280918281818090918181 55"}}
15:51:18 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00F0 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
15:51:18 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00C8 0140 1FE0 B0909090918182818182809090909180918181818090918181 55"}}
15:51:19 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00E6 1FF4 A0909090918181818181809090909180918181818090918181 55"}}
15:51:19 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00DC 0172 1FD6 B0909090918181818181809090909180918282828090918181 55"}}
15:51:19 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 0168 1FE0 B0909090928181828181809090909180918181818090918181 55"}}
15:51:19 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0370 00F0 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
15:51:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00E6 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
15:51:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 0154 1FD6 B0909090918181818181809090909180918181818090928281 55"}}
15:51:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00D2 1FC2 A0909090918181818181809090909180918181818090918181 55"}}
15:51:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00C8 014A 1FCC B0909090918282828181809090909180918181818090928181 55"}}
15:51:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03B6 00AA 014A 1FEA B0909090928282828282809090909280928282828090928282 55"}}
15:51:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1FEA A0909090918181818181809090909180918181818090918181 55"}}
15:51:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00D2 0154 1FD6 B0909090918182818181809090909180918181818090918181 55"}}
15:51:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03A2 00BE 0140 1FF4 B0909090928281828281809090909280918181818090918181 55"}}
15:51:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 038E 00DC 1FE0 A0909090918181818181809090909180918181818090918181 55"}}
15:51:21 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00DC 015E 1FCC B0909090918281828181809090909180918181818090918181 55"}}
15:51:22 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 035C 00FA 1FF4 A0909090918181818181809090909180918181818090918181 55"}}
15:51:23 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00C8 0140 1FC2 B0909090918182818181809090909180918181818090918181 55"}}
15:51:24 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00DC 1FD6 A0909090918181818181809090909180918181818090918181 55"}}
15:51:24 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00C8 0140 1FD6 B0909090928181828182809090909180918181818090918181 55"}}
15:51:24 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00DC 1FE0 A0909090918181818181809090909180918181818090918181 55"}}
15:51:24 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 00B4 0154 1FEA B0909090928282828282809090909280928282828090928282 55"}}
15:51:24 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0384 00DC 1FEA A0909090918181818181809090909180918181818090918181 55"}}
15:51:25 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0384 00E6 017C 1FD6 B0909090918181818181809090909280918181818090918181 55"}}
15:51:25 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00E6 1FCC A0909090918181818181809090909180918181818090918181 55"}}
15:51:25 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00DC 03A2 1FD6 0398 381809090A 55"}}
15:51:26 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03B6 00A0 0154 1FFE B09090909282828282828090909092809282828280A0928282 55"}}
15:51:26 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00FA 1FEA A0909090918181818181809090909180918181818090918181 55"}}
15:51:26 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 037A 00E6 0096 0172 0276 1FEA D0909090918181818181809090909180918181818090A3C383 55"}}
[... cut ...]
15:51:49 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00D2 038E 015E 1FCC 03B6 481809092B 55"}}
15:51:50 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 00C8 0154 1FFE B0909090928281818282809090909180928281828090918181 55"}}
15:51:50 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 037A 00F0 1FF4 A0909090918181818181809090909180918181818090918181 55"}}
```

Please let me know if there is any additional information I can provide or any additional sets you would recommend me try.

Thank you for your help thus far. I appreciate your time and effort.

macavelly, 2019-05-04:
So this is strange, I bought this remote from amazon
https://www.amazon.com/gp/product/B01KJRGMHI/

And I was able to copy the up and down keys from the original remote, however, even though they were copied onto the new remote, the new remote does not work either.

Using the remote I bought on amazon this is the output I get for the DOWN key
```
03:50:18 DMP: A4 20 58 01 18 03 48 F0 3D 0C 55
03:50:18 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8280,"Low":280,"High":840,"Data":"F03D0C","RfKey":"None"}}
03:50:19 DMP: A4 20 58 01 0E 03 3E F0 3D 0C 55
03:50:22 DMP: A4 20 58 01 18 03 48 F0 3D 0C 55
03:50:22 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8280,"Low":280,"High":840,"Data":"F03D0C","RfKey":"None"}}
03:50:24 DMP: A4 20 58 01 0E 03 3E F0 3D 0C 55
03:50:24 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8280,"Low":270,"High":830,"Data":"F03D0C","RfKey":"None"}}
03:50:25 DMP: A4 20 58 01 18 03 48 F0 3D 0C 55
03:50:28 DMP: A4 20 62 01 18 03 48 F0 3D 0C 55
03:50:28 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8290,"Low":280,"High":840,"Data":"F03D0C","RfKey":"None"}}
03:50:29 DMP: A4 20 62 01 18 03 48 F0 3D 0C 55
03:50:31 DMP: A4 20 58 01 18 03 48 F0 3D 0C 55
03:55:52 DMP: A4 20 9E 00 E6 03 20 F0 3D 0C 55
03:55:52 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8350,"Low":230,"High":800,"Data":"F03D0C","RfKey":"None"}}
03:55:55 DMP: A4 20 9E 01 0E 03 16 F0 3D 0C 55
03:55:55 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8350,"Low":270,"High":790,"Data":"F03D0C","RfKey":"None"}}
03:56:32 DMP: A4 20 A8 00 DC 03 2A F0 3D 0C 55
03:56:32 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8360,"Low":220,"High":810,"Data":"F03D0C","RfKey":"None"}}
03:58:11 DMP: A4 20 3A 01 0E 03 48 F0 3D 0C 55
03:58:11 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8250,"Low":270,"High":840,"Data":"F03D0C","RfKey":"None"}}
03:58:12 DMP: A4 20 76 00 C8 02 F8 F0 3D 0C 55
03:58:13 DMP: A4 20 4E 01 18 03 3E F0 3D 0C 55
03:58:14 DMP: A4 20 3A 01 18 03 48 F0 3D 0C 55
03:58:14 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8250,"Low":280,"High":840,"Data":"F03D0C","RfKey":"None"}}
03:58:53 CMD: rfraw 166
03:58:53 SRC: WebConsole from 192.168.1.196
03:58:53 RSL: Group 0, Index 1, Command RFRAW, Data 166
03:58:53 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
03:58:53 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
03:59:05 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60401F03D0C55"}}
```

This is the output for the UP key

```
4:09:19 DMP: A0 55
04:09:24 DMP: A4 1F D6 01 22 03 3E F0 3D C0 55
04:09:24 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8150,"Low":290,"High":830,"Data":"F03DC0","RfKey":"None"}}
04:09:26 DMP: A4 1F D6 01 2C 03 3E F0 3D C0 55
04:09:26 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8150,"Low":300,"High":830,"Data":"F03DC0","RfKey":"None"}}
04:09:28 DMP: A4 1F D6 01 22 03 3E F0 3D C0 55
04:09:28 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8150,"Low":290,"High":830,"Data":"F03DC0","RfKey":"None"}}
04:09:30 DMP: A4 1F D6 01 2C 03 3E F0 3D C0 55
04:09:31 DMP: A4 1F D6 01 22 03 3E F0 3D C0 55
04:09:31 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8150,"Low":290,"High":830,"Data":"F03DC0","RfKey":"None"}}
04:09:32 DMP: A4 1F E0 01 2C 03 3E F0 3D C0 55
04:09:34 DMP: A4 1F D6 01 2C 03 3E F0 3D C0 55
04:09:34 RSL: tele/sonoff/RESULT = {"RfReceived":{"Sync":8150,"Low":300,"High":830,"Data":"F03DC0","RfKey":"None"}}
```

I'm not sure what I could be missing here.