Portisch/RF-Bridge-EFM8BB1 #118: Help with Friedland Libra Doorbell Protocol

StudioXbiz, 2019-05-21:
I'm trying to receive a reliable signal from a 433Mhz doorbell protocol that is not currently catered for out-of-the-box. (Friedland Libra protocol)

I do not need to transmit the signal, I just need to convert to an MQTT message when the signal is broadcast by the doorbell push button.

I have flashed the latest Portisch firmware and am able to receive inconsistent results using RfRaw 177 (RfRaw 166 produces no results).

My question is; is there a method to make these inconsistent results more consistent so that they can be acted on reliably?

Here is a sample capture:

```
23:04:34 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 02D0 02DA 281818 55"}}
23:04:41 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0370 0154 024E 0492 B092929281928192818192929292929292928192929281929292929292929292929292 55"}}
23:04:49 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 014A 0258 0456 B0929292A192A192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:05:11 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 02EE 02EE 281818 55"}}
23:05:27 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 0140 0262 047E B092929281928192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:05:57 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 02E4 02EE 281818 55"}}
23:06:06 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 0302 02D0 281818 55"}}
23:06:13 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 02EE 02DA 281818 55"}}
23:06:20 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 0140 0262 0488 B0929292A192A192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:06:38 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 014A 0262 04A6 B09292928192A192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:06:59 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 038E 014A 0262 044C B0929292A192A192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:07:26 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 02EE 02DA 281818 55"}}
23:08:07 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 008C 0064 01FE 281808 55"}}
23:09:06 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0398 014A 0262 0460 B0929292A192A192A1A19292929292929292A1929292A1929292929292929292929292 55"}}
23:09:19 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 02BC 02C6 281818 55"}}
23:09:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00FA 02A8 02B2 281818 55"}}
```

From what I already know about the Libra protocol I think the lines with 4 buckets is most inline with the protocol.

Please advise how to produce reliable console results and avoid the false readings.

Thanks

Portisch, 2019-05-22:
Try this new hex file: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/Friedland_Libra/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex
And use sniff mode 0xA6. I guess the decoding is better than 0xB1 bucket sniffing.

StudioXbiz, 2019-05-22:
Works a treat in rfraw 166 (0xA6) mode! :)

I'm receiving 2 listings on each button press, but that's fine.

Thanks for all the great work Portisch!

closed 2019-05-22 by Portisch