Portisch/RF-Bridge-EFM8BB1 #121: Adding support for new modules

bkenobi, 2019-06-11:
I am using a Sonoff Bridge with the latest firmware and the latest Tasmota.  I have a couple Acurite devices (00592TX temperature sensors, 5in1 weather station) as well as a Blue Line Innovations Power Meter Sensor.  I have the decoded protocol and how to convert the binary into useful data.  Is there an easy way to submit the protocol so it could be added to the main firmware in the next release?  I have not been very successful in using the BitBucketConverter yet, though I admittedly have not spent sufficient time to understand it.  Since I already know the protocol that step seems like a waste of time though.

I have already published an Arduino project that contains all 3 protocols but I'm hoping to switch to a Sonoff Bridge for simplicity.

bkenobi, 2019-06-11:
Here's a link to the project if it helps answer any questions:

https://github.com/bkenobi/ookDecoder

Portisch, 2019-06-11:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

### Information
What device you try to sniff data from? Description? Brand and type?

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

closed 2019-06-11 by Portisch

bkenobi, 2019-06-11:
### Latest firmware used?
Firmware version 0x04 (current release)
Sonoff Bridge with Tasmota 6.5.0

### Information
Acurite 00592TX temperature sensor

### Sniffed data
I recorded B1 strings but BitBucketConvert did not seem to generate the data I was expecting.  The timing info that is known.  There are a total of 64 bits as follows:

Sync (8 bits):
Sync high: 600ms
Sync low: 600ms

Data (56 bits):
1 = long then short
0 = short then long
short = 200ms
long = 400ms

I believe I can define this in the RF_Protocols.h file with use of the following page:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h

I was not successful in compiling the firmware on my system as I could not get the compiler configured.  I have no problem defining the protocol if I understand the link above correctly, but I don't have a way to test it.  I can provide the details on the 2 other modules (or more info on the Acurite 592TX if there is more needed).  I simply can't confirm that a new protocol I submit is valid through local testing.

Portisch, 2019-06-11:
This is not the latest version. Use the hex file from the master branch. Tasmota is out of date.

bkenobi, 2019-06-11:
I installed the Portisch firmware last week so I assumed it would be the latest version listed.  I can update again if that makes a difference to my question.  I believe this is a protocol definition issue not a version related question.  The Tasmota firmware was updated late last week as well so 6.5.0 is listed as the latest stable as of that time I believe.

bkenobi, 2019-06-11:
Tasmota shows Version 6.5.0 20190319 in the change log so yes there are newer versions as of 6/11.
The Portisch version is Keil 8051 v9.53 dated 3/26/2019 which appears to be the latest version.

I can update the Tasmota firmware if that will make a difference.

bkenobi, 2019-06-11:
I updated Tasmota to the current repository level (6.5.0.16 20190611).  I will try to capture some strings from the device but they will be redundant since I already know the timings.  Will update when valid capture is available.

bkenobi, 2019-06-11:
Based on an earlier issue:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/111

I believe that the Acurite protocol should be:

H (header bits), consisting of pulses of 600us followed by delays of 600us;
0 (zeros), pulses of 200 us followed by delays of 400us;
1 (ones), pulses of 400 us followed by delays of 200us.

I found that a tolerance of up to 100us might be needed to pick up the pulses.  So a 400us signal would be 300-500us.

bkenobi, 2019-06-14:
In case this isn't being ignored, I'll continue to update it.  As I posted previously, the timing of the unit I am currently considering is known.  The B1 data has been captured while the module was sitting ~1" away from the Sonoff bridge so it's as good as it will get.  I captured over 4000 points via writing the MQTT traffic to a file.  I can also capture the console text, but it's not as clean and is the same data.

When I attempt to use BitBucketConverter.py it appears that the signal is either rejected or the converter isn't happy with it for some reason.  As an example, I get the following:

root@kali-pi:~# python BitBucketConverter.py
Enter B1 line: AAB10300BE019A019028190855
Filtered 0xB1 data: AAB10300BE019A019028190855
Resulting 0xB0 data: AA B0 0B 03 08 00BE 019A 0190 281908 55
Usage: BitBucketConverter.py [options]

Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  -e DEVICE, --dev=DEVICE
                        device (ip or hostname) to send RfRaw B0 command
  -r REPEAT, --repeat=REPEAT
                        number of times to repeat
  -d, --debug           show debug info
  -v, --verbose         show more detailed info
  -f FILE, --file=FILE  go through a file and try each line interactively
None
Enter B1 line:

I am attaching the full file for your consideration.  I believe there are 2 modules being picked up (Acurite 592TX and Acurite 5in1 weather station).

bkenobi, 2019-06-14:
I could not attach a file so I have copied a section of the 4000+ lines to the comment instead.

tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 012C 006E 01F4 023A 38080929 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00BE 019A 0190 281908 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 00AA 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 017C 00F0 00AA 0212 38181A0A0A081A0A0A0A0A090A0A0A0A0A0A 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0168 0104 00A0 01CC 38181A 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0172 00F0 00AA 01A4 381A0819 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0190 00D2 01FE 28181818190909090909090909090819 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0122 0064 01FE 024E 38080929 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 0082 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 03B6 16DA 0B90 6202 38080808080808080818282808080828080828082808080808080808080808080808080808080808082808282808082828080828282808280828 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0082 01EA 01EA 28181818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 008C 010E 01F4 A0919191 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0154 0096 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 019A 00DC 00A0 0208 3818190909081909090909090A0A0A0A0A0A 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 015E 008C 01EA 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 0096 01E0 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 015E 0096 01E0 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 05 017C 00E6 00A0 006E 0208 48181819090A081A090A0A0A0A0A0A0A0B 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 015E 0082 01B8 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0136 0078 01EA 28080909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0136 008C 01FE 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 00C8 01A4 0078 0212 3818181818181818181819081818181909090A181A18 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0186 00D2 0212 281908181819090908190909090909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 012C 0078 01E0 01F4 38080929 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0154 00AA 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0078 01D6 01B8 2818181818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 012C 0078 01E0 28080909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 008C 01D6 01B8 2818181818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0096 01B8 0104 01B8 38181928 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 0082 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0082 01E0 01EA 281818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0190 00D2 0208 2818181909090819090909090909090909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0118 006E 01EA 01FE 380929 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0140 006E 01B8 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0140 006E 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 019A 00D2 008C 01FE 38181818190909090909090A0A0A090819 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 012C 0082 01F4 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0186 00D2 019A 28181909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0096 01C2 0104 01D6 38181928 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00C8 0190 019A 281818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00C8 0190 01A4 28190908 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 008C 0064 01D6 015E 381828 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 008C 01CC 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0140 006E 01EA 28080909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 0096 01EA 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0082 01E0 01CC 2818181818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 015E 0096 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 014A 0082 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 008C 010E 01FE A0919191 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 00D2 0190 0096 0208 3819081818181818181818181818181819090909081A 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0154 008C 01EA 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0136 0078 01E0 28080909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0154 00A0 01D6 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 017C 00E6 01AE 28190909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 010E 0082 01FE 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 01AE 00AA 01EA 2818181819090909090909090909090819 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 017C 00DC 01A4 28181909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00AA 01AE 01C2 2818181818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0190 00DC 01A4 28181909 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0082 0122 01EA A0919191 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 006E 01E0 014A 01EA 38181A28 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0096 0104 01EA A09191 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0172 00FA 00A0 01AE 381A0A0A 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0140 0082 023A 280809 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0096 010E 01FE A09191 55"}}
[... cut ...]
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 006E 01E0 0136 01F4 381A281A28 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00AA 01A4 01C2 281818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0064 01F4 01FE 281818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0078 01E0 0136 01FE 381A281A28 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 04 0096 0122 01D6 01E0 B091A0 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 00A0 01B8 01B8 281818 55"}}
tele/sonoff/RESULT {"RfRaw":{"Data":"AA B1 03 0096 0104 0208 A09191 55"}}

bkenobi, 2019-06-14:
In case you prefer to see the Tasmota conole text, here is what was printed this evening with the extraneous lines removed.

16:52:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 008C 0104 0208 A09191 55"}}
16:52:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 0064 0226 280809 55"}}
16:52:50 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 0104 01FE A09191 55"}}
16:52:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 03B6 16E4 0B86 620C 38080808080808080818282808080828080828082808080828280808080808080808080808080808082808282808280808280828282808280828 55"}}
16:53:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0082 01D6 014A 01FE 38181A28 55"}}
16:53:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 0190 0190 281818 55"}}
16:53:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 006E 01D6 01EA 2818181818 55"}}
16:53:38 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 01C2 01B8 281818 55"}}
16:54:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 0168 01E0 A08181 55"}}
16:54:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 017C 00A0 01F4 38181818190819081819090819090A1A1A1A1A181818081A1808080A 55"}}
16:54:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 01AE 01C2 281818 55"}}
16:54:32 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 00F0 0186 0096 006E 01FE 480808180818190908181818190808081808080A1A1808080A180A1A1B 55"}}
16:54:32 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0172 00F0 019A 28190909 55"}}
16:54:50 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 008C 01CC 014A 01E0 38181A28 55"}}
16:54:50 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0190 019A 281818181818 55"}}
16:55:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 017C 00E6 0190 28181819 55"}}
16:55:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 008C 0104 01FE A09191 55"}}
16:55:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 006E 01D6 01EA 281818 55"}}
16:55:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0082 01CC 01EA 2818181818 55"}}
16:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0064 01F4 0212 281818 55"}}
16:56:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 014A 00F0 0078 0186 B81928 55"}}
16:56:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 008C 0104 01FE A09191 55"}}
16:56:38 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00F0 01A4 01F4 28090808080808080808181809081818190819080818 55"}}
16:56:38 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 01A4 01B8 281818 55"}}
16:56:56 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0064 0212 280809 55"}}
16:57:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0096 01D6 0154 01D6 381A2A28 55"}}
16:57:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0190 00DC 019A 28190909 55"}}

thermseekr, 2019-07-04:
Hello,

I'm the poster of issue #111, I'm resuming work on trying to read data from my Lacrosse sensors. Thank you for sharing your findings so far, I'll let you know if I have any advances.

jpduhen, 2019-07-24:
Hi guys @bkenobi @thermseekr , 
I like to be in the loop also. I'm trying to read the data from an [Auriol H23726a](https://ns323666.ip-37-187-156.eu/manuals/Auriol%20Weather%20Station/Auriol%20H13726%20Weather%20Station.pdf), which has a console with a LCD screen and 2 separate exterior sensors: 
- rain-meter,
- windspeed/-direction & temp/humidity meter.

I like to get the sensor-data into HomeAssistant via Sonoff Bridge - >MQTT
As it seems the latest @Portisch firmware (26-03-2019) for my Sonoff Bridge (Tasmota 6.6.0.0) should be albe to decode the RF signals. So I flashed the Sonoff and all sees to be OK.

When I enter the command
`RfRaw 166`
I get the following responses:
`
16:57:19 CMD: rfraw 166

16:57:19 MQT: stat/BRIDGE/RESULT = {"RfRaw":"ON"}
16:57:19 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA055"}}
16:57:23 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060EA46FB910A055"}}
16:57:52 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA16C0C000055"}}
16:58:25 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4411814C055"}}
16:58:25 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4411814C055"}}
16:58:25 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4411814C055"}}
16:58:25 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4411814C055"}}
16:58:25 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4411814C055"}}
16:58:29 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA16C0C000055"}}
16:59:06 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA16C0C000055"}}
16:59:27 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA4680020A055"}}
16:59:43 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA16C0C000055"}}
17:00:20 MQT: tele/BRIDGE/RESULT = {"RfRaw":{"Data":"AAA6060AA16C0C000055"}}
`
Looking into the described [rf-decode](https://github.com/yu55/auriol_reader/blob/master/doc/auriol_protocol_v20.pdf) document, I get stuck ;-( just like @keteflips
Anyone ideas? Much appreciated!