Portisch/RF-Bridge-EFM8BB1 #122: Combine standard (A4) & predefined protocols (A6) sniffing?

hongtat, 2019-06-11:
Hi,

Would it be possible to listen to both the standard 0xA4 (Itead default) and advanced sniffing by predefined protocols (0xA6) concurrently? If possible, B1 bucket as well?

Use-case:

I have devices that are on different "protocols" (A4, A6 & B1), and I'd like to capture their "state" (they send a rf data upon on/off/open/close/etc, but can be different protocol).

Currently, if I set to A4, it is not able to capture A6, and if I set to A6, it cannot capture A4.
If I set to B1, it captures everything, which is good, however, I read it somewhere (correct me, if I'm wrong), it resets and doesn't listen to B1 after some hours.

Thanks for you help.

Portisch, 2019-06-11:
No, add all protocols to one sniffing method will solve your problem. Take a look to the Wiki how to define new protocols.

closed 2019-06-11 by Portisch

hongtat, 2019-06-11:
Thanks for your response.

Is it possible to include the standard (0xA4) PT2260, PT2262, PT2264, EV1527, etc protocols into the predefined protocols (0xA6) sniffing?

Portisch, 2019-06-11:
They are already included: 0xA6 is 0xA4 + all others defined