Portisch/RF-Bridge-EFM8BB1 #123: Nothing detected but Ive done research, can you tell me if this can be supported?

erew123, 2019-06-17:
I dont know very much about rf and Im sure youre busy, so if you get time to look at this at all, its much appreciated!!!

I have a Drayton Digistat 2 RF (also sold as Worcester thermostat OR British Gas RC2). Ive just updated my RF bridge today, using Tasmota 6.50 and your latest firmware for the RF side (Keil 8051 v9.53 - Release).

When I use the thermostat to send a On/Off to my rf receiver, I pick nothing up in the tasmota console (I do pick up other devices), so Im assuming the signal could be out of the range/capability of the Sonoff bridge? (but you will know better than I).

Looking around the internet, here is all the relevant/useful information I can find.

**Pulses listed here (captured in RFlink, which supposedly now supports this thermostat):**

On
405	915	1410
0000000000001210000001001001100100000011000000001000000110

Off
405	915	1410
0000000000001210000001001001100100000000000000001110000110

[(https://www.letscontrolit.com/forum/viewtopic.php?f=10&t=3586&p=19345&hilit=digistat#p19345)]

**Another packet breakdown here, and someone suggests its using "Manchester Encoding":**

; Boiler Off packet   AA AC 59 99 59 55 59 56 AA
; Boiler On  packet   AA AC 59 99 59 55 95 65 AA

[https://groups.google.com/forum/#!topic/homecamp/kEiAfm4yhdI]

**Someone giving details of capturing the packets via arduino/audacity:**

[http://www.stevenhale.co.uk/main/2013/08/home-automation-reverse-engineering-a-worcester-bosch-dt10rf-wireless-thermostat/]

Suggestion here that each signal is sent 3 times:

[https://forum.allaboutcircuits.com/threads/thermostat-hacking-help-needed.102739/]

**Another Arduino capture, which says** "I then multiplied these values by 5.2083333 to get the number of microseconds (μs). I ended up with 60 values for “heating on” and 62 values for “heating off”"

[https://damn.technology/controlling-british-gas-wr1-receiver-arduino]

If you have any thoughts/suggestions on this, it would be appreciated, even if its a "no, give up"

Thanks!!

Portisch, 2019-06-17:
Old firmware

closed 2019-06-17 by Portisch

speedyrazor, 2019-10-08:
I am also interested in this topic for boiler control, and have the same issue, nothing detected.
I am on firmware vision from this thread, which works well with my doorbell integration - https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/131

Your last reply to this thread said "Old firmware", could you elaborate on that please?
Kind regards.