Portisch/RF-Bridge-EFM8BB1 #125: receive normal mode code and 0xA6 code at same time?

StudioXbiz, 2019-06-20:
Portisch kindly added the [Libra protocol definitions](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/118) to the RF_Protocols.h so I can use 0xA6 sniffing to pickup the signal however I need to also sniff a different protocol signal that is only picked up in normal mode.

Is there a way to receive protocols from both modes at the same time without moving between modes?

Portisch, 2019-06-20:
The other one from normal sniffing should be supported also in advanced sniffing.

closed 2019-06-20 by Portisch

StudioXbiz, 2019-06-20:
The normal mode (non-Libra) received code is the following and does not appear in the advanced (166) mode
{"RfReceived":{"Sync":6680,"Low":230,"High":660,"Data":"978A61","RfKey":"None"}}

I notice in the Libra commit the RF_Config.h turns off most protocols by setting the from 1 to 0 to... is this why the above is not detected in advanced mode? Why is it received in normal mode? Does normal mode not refer to this file for active protocols?

StudioXbiz, 2019-06-24:
When you say "The other one from normal sniffing should be supported also in advanced sniffing."

Do you mean that I have to somehow add the normal mode protocol to advanced sniffing to work or that it should already work in advanced sniffing? I thought that when you activate advanced sniffing it de-activates all the protocols that work in normal mode?

Portisch, 2019-06-24:
No, standard sniffing is only PT62... devices. Advanced sniffing is this device and all other enabled ones in the config.

StudioXbiz, 2019-06-25:
I've just run another random test from a different remote control for a different product that works in standard mode and found that also doesn't work in Advanced mode:

`{"RfReceived":{"Sync":10030,"Low":270,"High":920,"Data":"155558","RfKey":"None"}}`

The one that works in standard mode that I am trying to receive in advanced mode (alongside the libra protocol in advanced mode ) is:

`{"RfReceived":{"Sync":6680,"Low":230,"High":660,"Data":"978A61","RfKey":"None"}}`

How does one debug a protocol that appears in standard sniffing (rfraw 0) that doesn't in advanced sniffing (rfraw 166)?

Portisch, 2019-06-25:
ahh,
this is not supported! The standard sniffing is using a variable sync time range.
Advanced sniffing is using a defined sync time: `10850` +- tolerance
So you need to change https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/af1bddb3d81c79d67063184219ec21f8249dffd0/inc/RF_Protocols.h#L81
`350, 1050, 10850` to `270, 920, 10030`
Then add a copy of PT226X protocol and rename it like PT226X_2 and use these settings:
`230, 660, 6680`

Then both should be decoded in advanced sniffing mode.

StudioXbiz, 2019-06-25:
That's great, thanks Portisch!

OK, on to try and figure out how to setup the build environment to give that a spin!

StudioXbiz, 2019-06-30:
I've tried downloading and running Simplicity Studio 4 in an attempt figure out how to build this project and to try the above code changes however I'm finding that it's not an inuitive build environment for the uninitiated and is going to require quite some time to get acquainted. I have the IDE running on linux now but I'm scratching my head with regards to importing the project files.

Would you be able to provide to quick outline of setting Simplicity Studio 4 up for the purpose of the build. The 'Compile-this-project' wiki page may benefit from a little more detail to aid people that have never used the build environment get started.

StudioXbiz, 2019-07-02:
I finally got Simplicity Studio 4 working in a Virtual Machine and have built the firmware with the above changes.

Working great! Thanks Portisch

turin400, 2019-08-09:
You could upload the compiled code, I am not able to run the Simplicity Studio.
Thank you.