Portisch/RF-Bridge-EFM8BB1 #126: KAKU doorbell

phoxy84, 2019-06-27:
### Latest firmware used?
I run the latest Tasmota (6.5.0) and Portisch (0x04) . 

### Information
I use it on a Sonoff RF Bridge, the signal is sent from a KAKU (Klik aan klik uit) doorbell (ACDB-7000B), which is connected to my doorbell.

I can sniff the 0xB1 data but nothing else. See below.

### Sniffed data
This is my sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) (rfraw 177) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6) (rfraw 166). When pressing the doorbell after 0xA6 (rfraw 166) nothing appears. Pressing it with 0xB1, 8 to 12 rows appear, that is when pressing the doorbell only **one** time. The lines are very different as you can see. Also nothing appears after rfraw 1 of rfraw 169 (0xA9).

What am I missing?

```
11:13:33 CMD: Rfraw 1
11:13:33 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:14:23 CMD: Rfraw 166
11:14:23 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:14:23 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
11:14:58 CMD: Rfraw 169
11:14:58 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:15:29 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAAA55"}}
11:17:46 CMD: Rfraw 177
11:17:46 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:17:46 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
11:17:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A50 00DC 04E2 281E 481828383808380828383808083828382838380838083808380828380838380838080838380838080838380808380838380838080838083838083808283838083808 55"}}
11:17:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0A5A 04EC 280A 381808282808280808282808082808280828280828082808280808280828280828080828280828080828280808280828280828080828082828082808082828082808 55"}}
11:17:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00DC 04E2 2814 481828383828382828383828283828382838382838283808380808380838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
11:17:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383828283828382838382838083828380828382838380838080838380838280838380808380838380838080838083838083808283838083828 55"}}
11:17:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A46 00D2 04D8 2814 481828383808380828383828083828380838380838083808380808380838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
11:17:53 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00C8 04D8 2814 481828383828382828383828283828382838382838083828380828382838380838080838380838082838380828380838380838080838083838083808083838083828 55"}}
11:17:53 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A3C 00D2 04E2 2814 481828383828382828383828283828382838380838083808380828382838380838082838380838082838380828380838380838082838083838083808083838083828 55"}}
11:17:55 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 280A 481828383828382828383828283828382838380838283828380828380838380838080838380838082838380808380838380838080838083838083808083838083828 55"}}
11:17:57 MQT: tele/sonoffrf/STATE = {"Time":"2019-06-27T11:17:57","Uptime":"0T00:05:16","Vcc":3.179,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"Wifi":{"AP":1,"SSId":"VFNL-C9D1E0","BSSId":"00:1D:AA:C9:D1:E0","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:18:12 CMD: Rfraw 1
11:18:12 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:18:27 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A46 00DC 04D8 280A 4818283838283808283838082838283808383808380838083808083808383808380808AAB103012C04F604D8280818 55"}}
11:18:27 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A50 00DC 04E2 281E 481828383828380828383808083828380838380838083808380808382838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
11:18:27 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0A5A 04F6 2814 381808282808280808282808082808280828280828082808280808280828280828080828280828080828280808280828280828080828082828082808082828082808 55"}}
11:18:28 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383808283828382838382838083828380808382838380838082838380838082838380828380838380838082838283838083808283838083808 55"}}
11:18:28 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 280A 481828383828382828383828283828382838380838083828380828382838380838082838380838080838380808382838380838080838283838083808083838083808 55"}}
11:18:28 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A3C 00D2 04D8 2814 481828383828382828383828283828382838382838083808380828382838380838082838380838280838380808380838380838080838083838083808083838083828 55"}}
11:18:28 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383828283828382838380838083808380828380838380838082838380838080838380808382838380838082838083838083808083838083808 55"}}
11:18:29 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 280A 481828383828382828383828283828382838382838083808380828380838380838080838380838082838380808380838380838080838083838083808083838083808 55"}}
11:18:29 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383828283828382838382838083808380808380838380838080838380838082838380808380838380838080838083838083808083838083828 55"}}
11:18:29 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383828283828382838382838083808380808382838380838082838380838080838380828380838380838080838083838083808083838083828 55"}}
11:18:29 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 280A 481828383828380828383828283828382838380838283808382828382838380838082838380838080838380808382838380838080838083838083808283838083828 55"}}
11:18:30 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A3C 00D2 04E2 280A 481828383828382828383828283828380838382838083828380828380838380838080838380838080838380828380838380838080838083838083808083838083828 55"}}
11:18:44 CMD: Rfraw 166
11:18:44 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:18:44 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
11:19:02 CMD: Rfraw 167
11:19:02 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:19:02 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
11:19:07 CMD: Rfraw 169
11:19:07 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:19:38 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAAA55"}}
```

```
11:23:11 CMD: Rfraw 177
11:23:11 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
11:23:11 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
11:23:13 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 04E2 04CE 281808 55"}}
11:23:14 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A5A 00DC 04E2 2828 481828383808380808383808283808380838380838083808380828380838380838080838380838080838380808380838380838080838083838083808083838083828 55"}}
11:23:14 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 04 012C 0A5A 04EC 2814 381808282808280808282808082808280828280828082808280808280828280828080828280828080828280808280828280828080828082828082808082828082808 55"}}
11:23:14 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A50 00D2 04D8 281E 481828383828382828383828283828382838380838083808380828382838380838080838380838080838380828380838380838080838083838083808083838083828 55"}}
11:23:14 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 2814 481828383828382828383828283828382838382838083808380828382838380838082838380838080838380808380838380838080838083838083808283838083808 55"}}
11:23:15 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 280A 481828383828382828383808283828382838380838083828380828380838382838082838380838080838380808380838380838082838083838083808083838083808 55"}}
11:23:15 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00C8 04D8 2814 481828383828382828383828283828382838382838083808382828382838380838082838382838080838380808382838380838080838083838083808083838083828 55"}}
11:23:15 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00DC 04E2 281E 481828383828382828383828283828382838380838083828380828382838382838080838380838082838380808382838380838082838083838083808083838083828 55"}}
[... cut ...]
11:23:16 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A3C 00C8 04D8 281E 481828383828382828383828283828382838382838083808380808382838380838082838380838082838380828380838380838080838283838083808283838083828 55"}}
11:23:16 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00DC 04D8 2814 481828383828382828383828283828382838380838283808380828382838380838080838380838080838380808382838380838080838083838083808283838083828 55"}}
11:23:16 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 2814 481828383828382828383828283828382838382838083808380808382838380838082838380838080838380808382838380838080838283838083808283838083808 55"}}
```

Portisch, 2019-06-28:
You tried advanced sniffing [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6)
This protocol got already implemented. Use the master branch, there it should be included.
If it is not working try a little bit different timing from your scan:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/af1bddb3d81c79d67063184219ec21f8249dffd0/inc/RF_Protocols.h#L233
Change `{ 300, 2560, 140, 1150, 10230 }` to `{ 320, 2630, 210, 1240, 10260 }`

phoxy84, 2019-06-28:
Thanks for the answer, I will try tonight or this weekend. The master branch is [this ](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) one, right?

I will look into changing the timing as well. Maybe a dumb question, how do I change that data?

Portisch, 2019-06-28:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki

phoxy84, 2019-06-28:
I don't get it but will first try the master branch. Thanks

phoxy84, 2019-06-28:
It’s good now, 1 preses gives me:

```
15:55:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050878F3699B55"}}
15:55:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087FF3699B55"}}
15:55:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050878F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087FF3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087CF3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050878F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087FF3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087EF3699B55"}}
15:55:51 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087CF3699B55"}}
15:55:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050878F3699B55"}}
15:55:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050868F3699B55"}}
15:55:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA605087CF3699B55"}}
15:55:52 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA6050878F3699B55"}}

```

closed 2019-06-28 by phoxy84