Portisch/RF-Bridge-EFM8BB1 #127: Keep getting "Magic byte is not 0xE9" errors

tlc76, 2019-06-28:
I flashed Tasmota 6.5.0 on two RF Bridge devices, version R2.

When I get to flashing the Portisch firmware hex file, I always get the "Magic byte is not 0xE9" error.

Please let me know which version of the Portisch firmware is working with Tasmota 6.5.0.
Also, please advise if it is necessary to use any mandatory "define" statements in user_config_override.h (or any other compiler-level setting), to allow flashing of Portisch hex file firmware.

I have attached all the files that I have tried so far without success.
[portisch_hex_files.zip](https://drive.google.com/file/d/10kvbd157Bhwa1w4AoPu6Ql3W9zveJG6q/view?usp=sharing)

Notes:
- tried with CR/LF (Windows) and LF only (Unix) line separators
- tried with and without an empty new line at the end of the file
...in all cases I get the bloody "Magic byte is not 0xE9" error.

Thank you!
Cristian

Portisch, 2019-06-28:
You download it wrong. Download the whole repo as zip and extract the hex from the zip.

closed 2019-06-28 by Portisch

tlc76, 2019-06-28:
Thanks for the advice.
I downloaded the whole repository as .zip archive (RF-Bridge-EFM8BB1-master.zip) and extracted the hex file (path within archive: RF-Bridge-EFM8BB1-master.zip\RF-Bridge-EFM8BB1-master\Keil 8051 v9.53 - Release\RF-Bridge-EFM8BB1.hex), but I ran into the same issue...
(in fact I already had the same "exact" file in my archive - called latest.hex - just that I had previously converted LFs to CR/LF to try it out, but initially the file was exactly as the one from the downloaded repository).
Here are the files:
- the one embedded in the repository .zip archive: [RF-Bridge-EFM8BB1.hex](https://drive.google.com/file/d/1Ba8fRQKMQGTJSdPvCJSNxNZoA_jwTsE0/view?usp=sharing)
- the one that I had already, reconverted to Unix line separators: [latest.hex](https://drive.google.com/file/d/1F8Fme_HpZdnUAYkEEbVNfmQNw9M_gFCW/view?usp=sharing)

I had cut the GPIO4 and GPIO5 lines as advised by @arendst, so I tried powering up the device both via USB and 3.3V/GND pins from the USB/TTL converter --> same error in both attempts.

What can I be doing wrong?
I'm thinking maybe the contacts GPIO4-C2D and GPIO5-C2Ck are not good, but I believe the magic byte error is at file level, so the firmware upload routine does not get even get to the point where it sends out the hex file to the RF chip. Any advice?
Could the issue be caused by any incompatibility between Tasmota 6.5.0 and the Portisch firmware?

Thanks a lot for your support!

ErikAndren, 2019-06-28:
When building Tasmota ensure you have this line defined:
https://github.com/arendst/Sonoff-Tasmota/blob/6ac1904e2822859f9e8913d21e697156c8f34c41/sonoff/my_user_config.h#L445
Also make sure you are not building the full sonoff and not the basic.
Finally ensure after flashing and logging into tasmota using a browser ensure you have configued tasmota to be a sonoff rf bridge.

tlc76, 2019-06-28:
I finally did it...
It seems I was selecting the "env_default = sonoff-basic" build in platformio.ini.
I switched to "env_default = sonoff" by uncommenting the appropriate line and everything went fine!

I would add a comment in the wiki, but can someone confirm which are the builds that support uploading the Portisch firmware?

Thank you,
Cristian

tlc76, 2019-06-28:
> When building Tasmota ensure you have this line defined:
> https://github.com/arendst/Sonoff-Tasmota/blob/6ac1904e2822859f9e8913d21e697156c8f34c41/sonoff/my_user_config.h#L445
> Also make sure you are [...]

Thank you for the advice, I realized that quite late, after spending around 3 hours trying different things :)

tlc76, 2019-06-28:
I added a Warning in the [Sonoff RF Bridge 433](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#rf-firmware-upgrade), step 6:

> Hit Start Upgrade. The flashing should complete within 60 seconds. Device will restart after upgrade completes. _***WARNING: make sure you have previously installed the full Tasmota build on your Sonoff RF Bridge, otherwise the "Magic byte is not 0xE9" error will occur._