Portisch/RF-Bridge-EFM8BB1 #128: New Protocol - Intelbras XAS 4010 Door/Window sensor

marciogranzotto, 2019-07-01:
### Information
I'm trying to add a new device protocol for a Window/Door sensor made by Intelbras - a brazillian company.

### Sniffed data
```
01:51:02 CMD: rfraw 177
01:51:02 MQT: stat/bridge/RESULT = {"RfRaw":"ON"}
01:51:03 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
01:51:09 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0226 0406 3066 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:09 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03CA 03FC 2809180909 55"}}
01:51:10 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03FC 03F2 2819081818 55"}}
01:51:11 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03FC 3098 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:12 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03CA 30CA A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:12 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01CC 0410 03DE 2819081818 55"}}
01:51:14 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 0406 3070 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:14 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01FE 03D4 03F2 2809180909 55"}}
01:51:14 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 0410 30B6 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:16 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03F2 30A2 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:18 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03F2 30AC 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:19 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0208 03E8 30C0 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:20 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 03FC 3098 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:21 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0208 03FC 30AC A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:23 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 03FC 30A2 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:23 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01E0 03DE 03E8 2809180909 55"}}
01:51:23 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 0442 006E 047E B092A2 55"}}
01:51:25 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 03FC 30A2 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:25 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03F2 03F2 2809180909 55"}}
01:51:26 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03E8 30F2 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:26 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 0406 0406 2819081818 55"}}
01:51:27 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 044C 0262 0474 381829 55"}}
01:51:27 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0226 0406 3066 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:28 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01D6 03D4 30E8 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:28 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01C2 03DE 0406 2819081818 55"}}
```

### What I did so far

Using `AA B1 03 01FE 03E8 3066 2809090918090918181818090918090909091809180909181809180918 55` I ran `BitBucketConverter.py` and got:
```
Enter B1 line: AA B1 03 01FE 03E8 3066 2809090918090918181818090918090909091809180909181809180918 55
Filtered 0xB1 data: AAB10301FE03E83066280909091809091818181809091809090909180918090918180918091855
AAB10301FE03E83066280909091809091818181809091809090909180918090918180918091855
DrawImage, try to draw a image of: 2809090918090918181818090918090909091809180909181809180918
DrawImage, bucket high/low marking is included
Resulting 0xB0 data: AA B0 25 03 08 01FE 03E8 3066 2809090918090918181818090918090909091809180909181809180918 55
```

With that I tried to create a new protocol:
```c
/*
 * Intelbras XAS 4010 SMART door/windows sensor
 */
#if EFM8BB1_SUPPORT_XAS4010_PROTOCOL == 1
#define XAS4010
SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(XAS4010)[], static uint16_t, SI_SEG_CODE) = { 520, 1000, 12480 };
SI_SEGMENT_VARIABLE(PROTOCOL_START(XAS4010)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(2) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(XAS4010)[], static uint8_t, SI_SEG_CODE) = { LOW(0), HIGH(1) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(XAS4010)[], static uint8_t, SI_SEG_CODE) = { LOW(1), HIGH(0) };
#endif

#if EFM8BB1_SUPPORT_XAS4010_PROTOCOL == 1
		/*
		 * Intelbras XAS 4010 SMART door/windows sensor
		 */
		{
			{ &PROTOCOL_BUCKETS(XAS4010), ARRAY_LENGTH(PROTOCOL_BUCKETS(XAS4010)) },
			{ &PROTOCOL_START(XAS4010), ARRAY_LENGTH(PROTOCOL_START(XAS4010)) },
			{ &PROTOCOL_BIT0(XAS4010), ARRAY_LENGTH(PROTOCOL_BIT0(XAS4010)) },
			{ &PROTOCOL_BIT1(XAS4010), ARRAY_LENGTH(PROTOCOL_BIT1(XAS4010)) },
			{ NULL, 0 },
			29
		},
#endif
```

But no matter what, after flashing that to the RF Bridge, it still does not detect the signal unless I use the `rfraw 177` command.

Any ideas in how I can parse that signal and create a new protocol?

Portisch, 2019-07-01:
Check the oscilloscope! You have:
```
SI_SEGMENT_VARIABLE(PROTOCOL_START(XAS4010)[], static uint8_t, SI_SEG_CODE) = { HIGH(2), LOW(0) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(XAS4010)[], static uint8_t, SI_SEG_CODE) = { HIGH(0), LOW(1) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(XAS4010)[], static uint8_t, SI_SEG_CODE) = { HIGH(1), LOW(0) };
```

marciogranzotto, 2019-07-01:
Ohh thanks! I was confused about that.
I'll try it later tonight and post the results

thermseekr, 2019-07-01:
Marcio, another Brazilian here :) Listening carefully and looking forward to see your results. If positive I’ll surely go for those sensors too.

Regards,
Tales Maschio

marciogranzotto, 2019-07-21:
Ok, that didn't work...

If you look at the raw scan, I always get two results
```
01:51:14 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 0410 30B6 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:16 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03F2 30A2 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:18 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 03F2 30AC 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:19 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0208 03E8 30C0 A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:20 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 03FC 3098 2809090918090918181818090918090909091809180909181809180918 55"}}
01:51:21 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0208 03FC 30AC A081818190818190909090818190818181819081908181909081908190 55"}}
01:51:23 MQT: tele/bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 03FC 30A2 2809090918090918181818090918090909091809180909181809180918 55"}}
```

So using by using the one that starts with `28` I get:
```
Enter B1 line: AA B1 03 0212 03F2 30AC 2809090918090918181818090918090909091809180909181809180918 55
Filtered 0xB1 data: AAB103021203F230AC280909091809091818181809091809090909180918090918180918091855
AAB103021203F230AC280909091809091818181809091809090909180918090918180918091855
DrawImage, try to draw a image of: 2809090918090918181818090918090909091809180909181809180918
DrawImage, bucket high/low marking is included
Resulting 0xB0 data: AA B0 25 03 08 0212 03F2 30AC 2809090918090918181818090918090909091809180909181809180918 55
```
[image]

But using the one that starts with `A0` I get:
```
Enter B1 line: AA B1 03 0208 03FC 30AC A081818190818190909090818190818181819081908181909081908190 55
Filtered 0xB1 data: AAB103020803FC30ACA08181819081819090909081819081818181908190818190908190819055
AAB103020803FC30ACA08181819081819090909081819081818181908190818190908190819055
DrawImage, try to draw a image of: A081818190818190909090818190818181819081908181909081908190
DrawImage, bucket high/low marking is included
Resulting 0xB0 data: AA B0 25 03 08 0208 03FC 30AC A081818190818190909090818190818181819081908181909081908190 55
```
[image]

I don't really know which one to use or how to interpret the oscilloscope image.

Also, the byte count is 29 or 28? This sync byte stuff is confusing

marciogranzotto, 2019-08-06:
@Portisch any ideas?