Portisch/RF-Bridge-EFM8BB1 #131: DIO Chacon and KaKu are the same protocol

miguelangel-nubla, 2019-07-15:
### Latest firmware used?
Latest built from master

### Information
Wireless motion detector DIO Chacon 54503

### Sniffed data
https://docs.google.com/spreadsheets/d/1akmaFqjuS9YmJ1XGOrc9-wFbcI69av02PvmEZjVeBS4/edit?usp=sharing

I have been having trouble with a DIO Chacon device.
I could sniff and receive the data but when sending it my wall sockets did not react to it.
Upon further debugging, I found the received data was not decoded correctly, so I was sending the wrong data.

After a bit of searching, I found a [manual for pilight](https://manual.pilight.org/protocols/433.92/switch/kaku.html) that seems to indicate the protocol is the same for various brands, but the timings are not exactly the same as my device. [Other places](https://github.com/1technophile/NewRemoteSwitch/blob/master/docs/new%20kaku%20protocol.txt) seem to report the same differences in timings.

I have searched and compiled the raw data of other GitHub issues for KaKu devices in the spreadsheet linked above. After doing some averages the mean timings seem to be around
321, 2586, 215, 1176, 10228.

I updated the KaKu definition:
~~~
#if EFM8BB1_SUPPORT_Kaku_PROTOCOL == 1
#define KaKu
SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(KaKu)[], static uint16_t, SI_SEG_CODE) = { 321, 2586, 215, 1176, 10228 };
SI_SEGMENT_VARIABLE(PROTOCOL_START(KaKu)[], static uint8_t, SI_SEG_CODE) = {  HIGH(0), LOW(1) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(KaKu)[], static uint8_t, SI_SEG_CODE) = {  HIGH(0), LOW(3),  HIGH(0), LOW(2) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(KaKu)[], static uint8_t, SI_SEG_CODE) = {  HIGH(0), LOW(2),  HIGH(0), LOW(3) };
SI_SEGMENT_VARIABLE(PROTOCOL_END(KaKu)[], static uint8_t, SI_SEG_CODE) = {  HIGH(0), LOW(4) };
#endif
~~~
and disabled DIO protocol (details for why bellow)
~~~
#define EFM8BB1_SUPPORT_DIO_PROTOCOL			0		// DIO Chacon RF 433Mhz, Issue #95
~~~
and seems to be working perfectly for me so far.
~~~
02:14:13 CMD: rfraw 166
02:14:13 MQT: sonoff_F81779/stat/RESULT = {"RfRaw":"ON"}
02:14:13 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
02:14:15 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA60507D781476655"}} // on
02:14:16 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA60507D781476655"}} // on
02:14:20 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA60507D781477655"}} // off
02:14:21 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA60507D781477655"}} // off
~~~

So I would like to ask @degsmfan, @benjaminlundgreen, @Sk4zz, @phoxy84 and @mgenrique to check with these values and report back if it is working for everyone.
Maybe we can have a definition that allows the tolerances to work for all these similar protocols and devices instead of manually defining them for each one.

___

If I don't disable the other protocol I receive A6 messages with a length of 07 and protocol number 7F, so it seems it is getting confused since the signal matches two definitions.
Maybe @Portisch can dig deeper on that to prevent similar protocol definitions to crash in the future.
~~~
02:22:50 CMD: rfraw 166
02:22:50 MQT: sonoff_F81779/stat/RESULT = {"RfRaw":"ON"}
02:22:50 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
02:22:52 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA6077FFFFFFE55AAA605"}}
02:22:52 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA6077FFFFFFE55AAA605"}}
02:23:01 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA6077FFFFFFE55AAA605"}}
02:23:01 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA6077FFFFFFE55AAA605"}}
~~~

phoxy84, 2019-07-15:
Hi, i would like to help of you can guide me. What do i have to change, how and where?

miguelangel-nubla, 2019-07-15:
Update the KaKu protocol definition in [RF_Protocols.h](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/inc/RF_Protocols.h) and build a new firmware to try it out [or download the firmware I built with the changes](https://mega.nz/#!Rd9TVIib!c7h-xG7I3Yv5L20CsiB_Vb7SCkj0TsWIHxnGCWu8EcY), then [flash your device](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Hardware) with it and test it.

If you are using Tasmota:
~~~
# Activate sniffing with "rfraw 166"
21:06:59 CMD: rfraw 166
21:06:59 MQT: sonoff_F81779/stat/RESULT = {"RfRaw":"ON"}
21:06:59 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
# Something like this should pop up when you activate the sensor/button. It is the data sent by your sensor to your receiver.
21:07:10 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA60507D781476655"}}
# Replace the begining with AA[A8](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA8) and re-send it using "rfraw [data]". Your receiver should react as if you had activated your sensor/button manually.
21:09:27 CMD: rfraw AAA80507D781476655
21:09:27 MQT: sonoff_F81779/stat/RESULT = {"RfRaw":"ON"}
21:09:27 MQT: sonoff_F81779/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
# In this case it is a motion sensor and this is the code for on. Keep in mind it will do nothing if it is already on.
# Repeat for off or other buttons....
~~~

Then please report back whether you are able to fully control your devices this way, with the brand and model information. A capture with "rfraw 177" will also be appreciated.

phoxy84, 2019-07-16:
Ok, I will try quickly.

phoxy84, 2019-07-18:
I flashed the new firmware and al is still working fine. I use the sonoff with the Klik aan Klik uit doorbell and receive a code when pressing the doorbell. I have an other device that i did not test yet, I will try this weekend. It is a Kaku built in wall receiver that can be in learning mode and has to connect with the sonoff. I hope I can test this weekend.

jonfin, 2019-07-19:
With this version my Intertechno ITT-1500 is working. The mean values are: `330, 2739, 219, 1298, 10691`

The attached file contains the `rfraw 177` log with all buttons pressed https://gist.github.com/jonfin/454f09a03d9031d6282a3f9e7b395dc2

Surprisingly the code with only 4 buckets did also work. It is missing the `219` value and if I change this bucket to 330 it will still work. (Did not test all)

phoxy84, 2019-07-20:
My data when pressing the doorbeel in rfraw 166 mode
```
08:36:50 CMD: rfraw 166
08:36:50 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
08:36:50 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
08:37:17 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA60507970C966455"}}
08:37:18 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA60507970C966455"}}
08:37:18 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA60507970C966455"}}
08:37:19 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA60507970C966455"}}
```

Nothing happens when I fill rfraw AAA80507970C966455 in the console.

This is rfraw 177

```
08:40:27 CMD: rfraw 177
08:40:27 MQT: stat/sonoffrf/RESULT = {"RfRaw":"ON"}
08:40:27 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AAA055"}}
08:40:41 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0500 04BA 280818 55"}}
08:40:41 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 012C 0A50 04EC 00D2 2814 481808282808280838282808082808280828280828082808280808280828280828080828280828080828280808280828280828080828082828082808082828082808 55"}}
08:40:41 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 281E 481828383828382828383828283828382838380838083808380828380838380838082838380838080838380828380838380838080838283838083808283838083828 55"}}
08:40:42 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A46 00D2 04E2 2814 481828383828382828383808283828380838380838083808380808380838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
08:40:42 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00C8 04D8 2814 481828383828382828383828283828382838380838083808380828380838380838080838380838080838380808382838380838080838083838083808083838083828 55"}}
08:40:42 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A50 00D2 04EC 2814 481828383828382828383808283828382838380838083808380808382838380838080838380838080838380808382838380838082838083838083808283838083828 55"}}
08:40:42 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04E2 2814 481828383828382828383828283828380838382838083808380808382838380838082838380838082838380808380838380838080838083838083808083838083808 55"}}
08:40:43 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A46 00D2 04E2 280A 481828383808380808383808283828380838380838083808380808380838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
08:40:43 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A46 00D2 04E2 280A 481828383828382828383808283828380838380838083808380808380838380838080838380838080838380808380838380838080838083838083808083838083808 55"}}
08:40:43 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A50 00D2 04D8 280A 481828383828382828383828283828382838380838283808382828382838380838082838380838082838380808382838380838080838283838083808283838083808 55"}}
08:40:43 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A3C 00D2 04D8 2814 481828383828382828383828283828382838380838083808380808382838380838080838380838080838380828380838380838082838283838083808083838083808 55"}}
08:40:44 MQT: tele/sonoffrf/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 0A46 00D2 04D8 2814 481828383828382828383808283828382838382838083808380808382838380838082838380838082838380808382838380838080838083838083808083838083828 55"}}
```

lauris-nl, 2020-10-22:
Hi I'm new, Thanks Miguelangel-Nubla,

Ive tried a couple of different (portisch).HEX files on my Sonoff R2...

admin@tv:~/Downloads/tasmota/sonoff_RFBRIDGE$ md5sum *.hex
39179072a8e7e4453c92d71c499582aa  [RF-Bridge-EFM8BB1-20190220.hex](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/download/0x04/RF-Bridge-EFM8BB1.zip)   <--no response rfraw 166
405fae8ccd8dd032fb58426ac2c5da7a  RF-Bridge-EFM8BB1_20FEB2019.hex   <--no response rfraw 166
35c6806a5bccecc8b74ab7750b8bd19b  [RF-Bridge-EFM8BB1_inc_commits_20201022.hex](https://github.com/Portisch/RF-Bridge-EFM8BB1/compare/0x04...master)   <-supports rfraw but invalid nothing happens if transmitted.?
33fe72505414a55e58ece8e80d22b663  [RF-Bridge-EFM8BB1_kaku.hex](https://mega.nz/#!Rd9TVIib!c7h-xG7I3Yv5L20CsiB_Vb7SCkj0TsWIHxnGCWu8EcY)     <-supports rfraw 166 KAKU
And when I press the third on button on my KlikAanKlikUit (KAKU) MODEL AYCT-102 remote.
```
07:37:50 MQT: tele/tasmota_C3ADAB/RESULT = {"Time":"2020-10-22T07:37:50","RfRaw":{"Data":"AAA605079F9E547D55"}}
07:37:51 MQT: tele/tasmota_C3ADAB/RESULT = {"Time":"2020-10-22T07:37:51","RfRaw":{"Data":"AAA605079F9E547D55"}}
07:37:53 MQT: tele/tasmota_C3ADAB/RESULT = {"Time":"2020-10-22T07:37:53","RfRaw":{"Data":"AAA605079F9E547D55"}}
07:37:53 MQT: tele/tasmota_C3ADAB/RESULT = {"Time":"2020-10-22T07:37:53","RfRaw":{"Data":"AAA605079F9E546D55"}}
07:37:54 MQT: tele/tasmota_C3ADAB/RESULT = {"Time":"2020-10-22T07:37:54","RfRaw":{"Data":"AAA605079F9E546D55"}}
```
I can send them back after I changed the first 6 into a 8.
```
08:27:11 CMD: Backlog RfRaw AAA805079F9E546D55; RfRaw 0
```
And I hear Klik and my fish-tank leds are on.
I'm hoping this can be fixed and put in the release folder:
https://github.com/arendst/Tasmota/tree/development/tools/fw_efm8bb1

atv2016, 2021-12-11:
can this new firmware still be downloaded somewhere?