Portisch/RF-Bridge-EFM8BB1 #135: Brennenstuhl Remotes not working

paranerd, 2019-08-15:
Hi all!

I just tried to get my Brennenstuhl RF Sockets working with the Sonoff RF Bridge 433, without any luck.

I was able to sniff via "RfRaw 177" - this was the result:
15:04:27 RSL: tele/sonoffrfbridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 03AC 01A4 0BA4 1C70 01010110011010010101011001011010010101101001010123 55"}}

Using BitBucketConverter.py I got the following:
AA B0 23 04 08 03B6 01AE 0B9A 1C8E 01010110101001010110101010100101100110101001010123 55

I tried executing the following in the Sonoff Bridge Console:
"RfRaw AAB023040803B601AE0B9A1C8E0101011010100101011010101010010110011010100101012355" (without the quotes)

Nothing happened^^

This is kind of a general question, I guess... Is executing the raw B0 command actually the right way to do it? I've seen conversions to "shorter" codes using "SYNC_HIGH", "BIT_LOW_TIME" and the like, which got me confused quite a bit because I couldn't find clear instructions on how to convert properly =/

Those are the sockets I'm using: https://www.amazon.de/Brennenstuhl-Funkschalt-Set-Funksteckdosen-Handsender-Kindersicherung/dp/B074DVCXWM/

Any help here would be highly appreciated!

mc-0815, 2019-12-20:
Hi,

did you solved your isse in the meantime? I've a similar one with some simple radios socket controlled by a 4 button remote control. 

Thanks

paranerd, 2019-12-25:
Sorry, no, I actually replaced them with WiFi sockets in the meantime. But if you get it to work, please let me know!
Merry Christmas!