Portisch/RF-Bridge-EFM8BB1 #136: Remote RE1011

yortiz1979, 2019-08-26:
Hi  @Portisch wonderful work, I need some help on this remote for my owning.

the remote has three four bottom

this is the control remote im using 
[image] like in this link: https://turkish.alibaba.com/product-detail/6-0m-3-0m-large-size-retractable-full-cassette-awning-with-led-60646476303.html

1. OPEN
2.CLOSE
3. STOP
4 . LIGHT ON/OFF

I have the rf bridge v2 succesfully installed latest tamota and upgraded the latest firmware ( RF-Bridge-EFM8BB1.hex)

while using the RfRaw AAB155 and pressing the buttom few time 3-5 secs i get the folowing codes:

`17:13:51 CMD: RfRaw AAB155
17:13:51 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
17:13:51 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:13:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 02B2 0384 180808 55"}}
17:14:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01D6 012C 0ECE 02F8 5E42 4818181818181818181818180818181818080808081818281808083B3B080808083B3B380B083B3B08083B0808083B083B0808083B083B3B083808080B080808381B08083B3B08 55"}}
17:14:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5CE4 481818181818081818080808080808080808080808080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}}
17:14:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0136 0ECE 02F8 5CC6 481818181818181818181808080818180818181818080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}}
17:14:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02EE 5CC6 481818181818081808080808080808080808080808080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}}
17:14:06 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 02F8 5E74 4818181818181818180808080808080808080808080808280808080808380B08080808083808080B080808380B083B3B080808083B3B080808083B0808080838080B18083B3B08080808380B 55"}}
17:14:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5E4C 4818181818181808080808080808080808080808081818280808080808380B08080808083808080B080808380B083B3B080808083B3B080808083B0808080838080B08083B3B08080808380B 55"}}
17:14:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 029E 0122 0398 28090809 55"}}
17:14:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5CEE 481818181818180808080808080808080808080808180828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55"}}
17:14:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 02EE 5CC6 481818181818181818080808080808180808080808080828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55"}}
17:14:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 02F8 5E4C 481818181818181808180808180808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55"}}
17:14:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0EC4 02EE 5E56 481818181818180808080808080808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55"}}
17:14:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 0334 5CEE 481818181818181818080808080808081808080808080828081818180B3808181808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}}
17:14:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 0122 0ECE 032A 5CC6 481818181818080808080808080808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}}
17:14:17 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 0334 5CD0 481818181818181818181808180808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}}
17:14:19 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 028A 03B6 180808080808 55"}}
17:14:24 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0316 0190 0334 280918 55"}}
17:14:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 0334 5E4C 4818181818181818180818080808080808180808080818280808080B083B3808080B08380B38080B083B3B3B083B083B3B083B083B380B3B3B0838080B38080808080B083B 55"}}
17:14:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CEE 48181818181808080808080808080808080808080808082838180B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55"}}
17:14:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CD0 48181818181808180808080808080808080808080808082838080B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55"}}
17:14:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CC6 48181818181818080808080808080808080808080808082838AAB10501CC01220ECE033E5E5648181818181818181818080808080808080808080808082808080B0808080838080B0808083B380B080808083B380B380B3B380B380B3B3B0838080B083808080B3808080B08080808 55"}}
17:14:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 01CC 0136 00DC 0EC4 0334 5E4C 58182818181818181818080808181808080808080808083808080C0808080848080C0808084C480C080808084C480C480C4C480C480C4C4C0848080C084808080C4808080C08080808 55"}}
17:14:37 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 023A 03AC 046A 28180808090808 55"}}
17:17:59 MQT: tele/sonoff/STATE = {"Time":"2019-08-26T17:17:59","Uptime":"0T00:35:23","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Maeva's Palace 2GHZ","BSSId":"C0:05:C2:0E:62:A1","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:19:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0064 0384 0154 0168 381A28 55"}}
17:19:49 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 021C 0082 00FA 01F4 381A0A 55"}}
17:20:45 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 02C6 0104 0078 01AE 0190 481A39 55"}}`

then i have used the sinffed data onBitBucket Converter ( http://bbconv.hrbl.pl/)

getting this result:

`AA B0 56 05 08 01C2 012C 0ECE 02F8 5CEE 481818181818180808080808080808080808080808180828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55
AA B0 56 05 08 01CC 0122 0ECE 02EE 5CC6 481818181818181818080808080808180808080808080828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55
###### Skipping line '"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 02F8 5E4C 481818181818181808180808180808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55"}}' ######
AA B0 58 05 08 01C2 012C 0EC4 02EE 5E56 481818181818180808080808080808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55
AA B0 56 05 08 01CC 012C 0ECE 0334 5CEE 481818181818181818080808080808081808080808080828081818180B3808181808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55
AA B0 56 05 08 01C2 0122 0ECE 032A 5CC6 481818181818080808080808080808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55
AA B0 56 05 08 01CC 0122 0ECE 0334 5CD0 481818181818181818181808180808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55
AA B0 0C 02 08 028A 03B6 180808080808 55
AA B0 0B 03 08 0316 0190 0334 280918 55
AA B0 51 05 08 01CC 012C 0ECE 0334 5E4C 4818181818181818180818080808080808180808080818280808080B083B3808080B08380B38080B083B3B3B083B083B3B083B083B380B3B3B0838080B38080808080B083B 55
AA B0 58 05 08 01C2 012C 0ECE 02D0 5CEE 48181818181808080808080808080808080808080808082838180B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55
AA B0 58 05 08 01C2 012C 0ECE 02D0 5CD0 48181818181808180808080808080808080808080808082838080B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55
[... cut ...]
17:40:53 CMD: cmnd/sonoff/Backlog RfRawAA B0 0B 03 08 0316 0190 0334 280918 55; RfRaw 0]
17:40:54 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
17:40:54 MQT: stat/sonoff/RESULT = {"RfRaw":"OFF"}
17:41:10 CMD: cmnd/sonoff/Backlog RfRaw AA B0 0B 03 08 0316 0190 0334 280918 55; RfRaw 0]
17:41:11 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
17:41:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:41:11 MQT: stat/sonoff/RESULT = {"RfRaw":"OFF"}`

thank you very much in advance for any guidance on this issue!

lbriet, 2019-08-26:
Hi, sorry for spamming/unrelated - trying to make sense of what this firmware will allows for. From what I understand, it adds support to the RF Bridge for a large number of RF vendors, and a comprehensive way of decoding/adding others? Rolling code support remains impossible due to hardware limitations?

yortiz1979, 2019-08-26:
Correct the firmware adds support to more options compared to the native firmware

yortiz1979, 2019-08-28:
Hi a quick update on this post. i managed to get few codes working but they only worked once, after that every time i try to ude any of the code below non of them work: any help, i think im getting closer just need quick help on this. thanks in advance.

`light on:

rfraw AA B0 57 05 08 01C2 0136 0ECE 02D0 5CEE 4818181818181818080808080808080808080808080808283B08083B08380B08083B0808083B083B0838080B3808080808080B3808080808080B0838080B0808080808380B3B08083B0838 55;rfraw off

rfraw AA B0 54 05 08 01C2 0136 0EC4 02EE 5CE4 48181818181818180808080808080808080808080808082808380808080B3B380808080B083B3808080B3B083B3B380B083B3B380B0838080B3B380808080B080808380808080B38 55;rfraw off

close:

rfraw AA B0 55 05 08 01C2 012C 0ECE 02EE 5E60 4818181818080808080808080808080808080808080808280808080808083B08080808080838080B080808083B083B3B3B08083B3B3B08083B08083B380B3B3B08080808080808083B 55;rfraw off
rfraw AA B0 53 05 08 01D6 0118 0EC4 02EE 5CDA 481818181818181818181818181818080808080808180828183B080808380B3B0808083B380B080808380B08083B3B3B08083B3B3B080808083B083B380B08380B3B0808083808 55;rfraw off
rfraw AA B0 51 05 08 01C2 012C 0ECE 02EE 5CBC 481818181818180808080808080808080808080808080828083B08380B3B3B08380B08380B08380B3B0808380B380B0808380B380B0808083B3B3B380B08083B3B08380B38 55;rfraw off`

yortiz1979, 2019-09-04:
anyone able to help????

yortiz1979, 2019-09-07:
hi @Portisch would you be able to give me some hand on this please?

killee, 2021-04-22:
Hi @yortiz1979 I've the same issue with this remote... Have you solved it?