Portisch/RF-Bridge-EFM8BB1 #139: Adding Mebus outdoor temperature sensor

kollera, 2019-10-24:
Hi!

With [rtl_433](https://github.com/merbanan/rtl_433) i was able to analyze/decode the sensor data.
Question: is this information sufficient to add a suitable configuration to [RF_Protocols.h](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h)?
can anybody give me a hint? :)

Thank you in advance!

```
ak@virus75:~/git/rtl_433$ rtl_433 -R 13 -A
rtl_433 version 19.08-24-g14c5b9a branch master at 201910071222 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ak/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 1 out of 140 device decoding protocols [ 13 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Detected OOK package	2019-10-23 20:20:19
Analyzing pulses...
Total count:  403,  width: 799.28 ms		(199821 S)
Pulse width distribution:
 [ 0] count:  402,  width:  496 us [484;520]	( 124 S)
 [ 1] count:    1,  width:  144 us [144;144]	(  36 S)
Gap width distribution:
 [ 0] count:  185,  width: 1944 us [1928;2188]	( 486 S)
 [ 1] count:  207,  width:  964 us [956;984]	( 241 S)
 [ 2] count:   10,  width: 3896 us [3884;3904]	( 974 S)
Pulse period distribution:
 [ 0] count:  185,  width: 2440 us [2428;2688]	( 610 S)
 [ 1] count:  207,  width: 1464 us [1448;1476]	( 366 S)
 [ 2] count:   10,  width: 4392 us [4384;4404]	(1098 S)
Level estimates [high, low]:  15938,     64
RSSI: -0.1 dB SNR: 23.9 dB Noise: -24.0 dB
Frequency offsets [F1, F2]:   17101,      0	(+65.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 144, long_width: 496, reset_limit: 3908, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=144,l=496,r=3908,g=2192,t=140,y=0'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 11 
[00] {33} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 0
[01] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[02] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[03] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[04] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[05] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[06] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[07] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[08] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[09] {37} 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000
[10] {37} 00 00 00 00 08 : 00000000 00000000 00000000 00000000 00001

Detected OOK package	2019-10-23 20:20:30
Analyzing pulses...
Total count:    1,  width: 0.04 ms		(   10 S)
Pulse width distribution:
 [ 0] count:    1,  width:   40 us [40;40]	(  10 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1688,    213
[... 249 lines cut ...]
Analyzing pulses...
Total count:  445,  width: 908.12 ms		(227031 S)
Pulse width distribution:
 [ 0] count:    1,  width:  248 us [248;248]	(  62 S)
 [ 1] count:  443,  width:  504 us [488;520]	( 126 S)
 [ 2] count:    1,  width:  128 us [128;128]	(  32 S)
Gap width distribution:
 [ 0] count:   12,  width: 3888 us [3880;3900]	( 972 S)
 [ 1] count:  204,  width:  956 us [944;976]	( 239 S)
 [ 2] count:  228,  width: 1936 us [1920;1956]	( 484 S)
Pulse period distribution:
 [ 0] count:   12,  width: 4372 us [4128;4400]	(1093 S)
 [ 1] count:  204,  width: 1464 us [1448;1484]	( 366 S)
 [ 2] count:  228,  width: 2440 us [2424;2460]	( 610 S)
Level estimates [high, low]:  15966,    204
RSSI: -0.1 dB SNR: 18.9 dB Noise: -19.0 dB
Frequency offsets [F1, F2]:   17697,      0	(+67.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 248, long_width: 504, reset_limit: 3904, sync_width: 128
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=248,l=504,r=3904,g=0,t=0,y=128'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {444} 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[01] { 0}                                                                                                                                                                         : 
```

kollera, 2019-10-25:
Also tried B1 sniffing & converting.
But which one is the Mebus sensor? 

```
Filtered 0xB1 data: AAB10401B807C603F207C63818182855
Resulting 0xB0 data: AA B0 0E 04 08 01B8 07C6 03F2 07C6 38181828 55
Filtered 0xB1 data: AAB104021C03980F32076C38182855
Resulting 0xB0 data: AA B0 0D 04 08 021C 0398 0F32 076C 381828 55
Filtered 0xB1 data: AAB104021203AC0762078038181818282855
Resulting 0xB0 data: AA B0 10 04 08 0212 03AC 0762 0780 381818182828 55
Filtered 0xB1 data: AAB104021203AC00FA074E38181818092855
Resulting 0xB0 data: AA B0 10 04 08 0212 03AC 00FA 074E 381818180928 55
Filtered 0xB1 data: AAB10401F403B60F1E078038182855
Resulting 0xB0 data: AA B0 0D 04 08 01F4 03B6 0F1E 0780 381828 55
Filtered 0xB1 data: AAB10401FE07760398076C3818182855
Resulting 0xB0 data: AA B0 0E 04 08 01FE 0776 0398 076C 38181828 55
Filtered 0xB1 data: AAB104021C038E0F28076238182855
Resulting 0xB0 data: AA B0 0D 04 08 021C 038E 0F28 0762 381828 55
Filtered 0xB1 data: AAB10402120776036607763818182855
Resulting 0xB0 data: AA B0 0E 04 08 0212 0776 0366 0776 38181828 55
Filtered 0xB1 data: AAB104040602D0006E02BC381A2A55
Resulting 0xB0 data: AA B0 0D 04 08 0406 02D0 006E 02BC 381A2A 55
Filtered 0xB1 data: AAB102024E07301808080855
Resulting 0xB0 data: AA B0 0A 02 08 024E 0730 18080808 55
Filtered 0xB1 data: AAB1050226074E03840EF6076C481818283855
Resulting 0xB0 data: AA B0 11 05 08 0226 074E 0384 0EF6 076C 4818182838 55
Filtered 0xB1 data: AAB104021203980776073A381818281828282855
Resulting 0xB0 data: AA B0 12 04 08 0212 0398 0776 073A 3818182818282828 55
Filtered 0xB1 data: AAB1040230038E07620F0A3818282828282828282818181818081818281818281828282828282828181818182828282855
Resulting 0xB0 data: AA B0 2F 04 08 0230 038E 0762 0F0A 38182828282828282828181818180818182818182818282828282828281818181828282828 55
Filtered 0xB1 data: AAB104023003840762076238182855
Resulting 0xB0 data: AA B0 0D 04 08 0230 0384 0762 0762 381828 55
Filtered 0xB1 data: AAB10400C80262008C0212381A080855
Resulting 0xB0 data: AA B0 0E 04 08 00C8 0262 008C 0212 381A0808 55
Filtered 0xB1 data: AAB1040208076C03980F463818282828182828281828282828282828181828182818181818181818282828281818182855
Resulting 0xB0 data: AA B0 2F 04 08 0208 076C 0398 0F46 38182828281828282818282828282828281818281828181818181818182828282818181828 55
Filtered 0xB1 data: AAB1040208076C03700762381828182818181855
Resulting 0xB0 data: AA B0 12 04 08 0208 076C 0370 0762 3818281828181818 55
Filtered 0xB1 data: AAB103024E0726035C28181855
Resulting 0xB0 data: AA B0 0B 03 08 024E 0726 035C 281818 55
Filtered 0xB1 data: AAB104020803A2076C076C38182855
Resulting 0xB0 data: AA B0 0D 04 08 0208 03A2 076C 076C 381828 55
Filtered 0xB1 data: AAB1040226038E07620F003818282828282828282818181818181818281818281828282828282828181818182828282855
Resulting 0xB0 data: AA B0 2F 04 08 0226 038E 0762 0F00 38182828282828282828181818181818182818182818282828282828281818181828282828 55
Filtered 0xB1 data: AAB104022603980758076238182855
Resulting 0xB0 data: AA B0 0D 04 08 0226 0398 0758 0762 381828 55
Filtered 0xB1 data: AAB10401D603DE07D0079E38181818282855
Resulting 0xB0 data: AA B0 10 04 08 01D6 03DE 07D0 079E 381818182828 55
Filtered 0xB1 data: AAB1040230076C038E076C3818182855
Resulting 0xB0 data: AA B0 0E 04 08 0230 076C 038E 076C 38181828 55
Filtered 0xB1 data: AAB1030226075807582818180855
Resulting 0xB0 data: AA B0 0C 03 08 0226 0758 0758 28181808 55
Filtered 0xB1 data: AAB104021C038E0E06076C38182955
Resulting 0xB0 data: AA B0 0D 04 08 021C 038E 0E06 076C 381829 55
Filtered 0xB1 data: AAB103023A074E07582818180855
Resulting 0xB0 data: AA B0 0C 03 08 023A 074E 0758 28181808 55
Filtered 0xB1 data: AAB1040230076203980F463818282828182828281828282828282828181828182818181818181818282828281818180855
Resulting 0xB0 data: AA B0 2F 04 08 0230 0762 0398 0F46 38182828281828282818282828282828281818281828181818181818182828282818181808 55
Filtered 0xB1 data: AAB105021C0758038E0F000758481818283855
Resulting 0xB0 data: AA B0 11 05 08 021C 0758 038E 0F00 0758 4818182838 55
Filtered 0xB1 data: AAB1040212076C03AC0758381828182818181855
Resulting 0xB0 data: AA B0 12 04 08 0212 076C 03AC 0758 3818281828181818 55
Filtered 0xB1 data: AAB104021203A2076C076238182855
Resulting 0xB0 data: AA B0 0D 04 08 0212 03A2 076C 0762 381828 55
Filtered 0xB1 data: AAB1040230075803840F003808181818181818181808180828282828182828182818181818181818282828281818181855
Resulting 0xB0 data: AA B0 2F 04 08 0230 0758 0384 0F00 38081818181818181818081808282828281828281828181818181818182828282818181818 55
Filtered 0xB1 data: AAB104022603840758075838182855
Resulting 0xB0 data: AA B0 0D 04 08 0226 0384 0758 0758 381828 55
Filtered 0xB1 data: AAB10200E6014018080855
Resulting 0xB0 data: AA B0 09 02 08 00E6 0140 180808 55
[... 165 lines cut ...]
Resulting 0xB0 data: AA B0 10 04 08 0230 0384 008C 0758 380818180A18 55
Filtered 0xB1 data: AAB10402260758038E07623818282855
Resulting 0xB0 data: AA B0 0E 04 08 0226 0758 038E 0762 38182828 55
Filtered 0xB1 data: AAB104023003840F0A078038182855
Resulting 0xB0 data: AA B0 0D 04 08 0230 0384 0F0A 0780 381828 55
Filtered 0xB1 data: AAB1030212076C087028181855
Resulting 0xB0 data: AA B0 0B 03 08 0212 076C 0870 281818 55
Filtered 0xB1 data: AAB10402120776038E07763818282855
Resulting 0xB0 data: AA B0 0E 04 08 0212 0776 038E 0776 38182828 55
Filtered 0xB1 data: AAB1040208038E0F0A077638182855
Resulting 0xB0 data: AA B0 0D 04 08 0208 038E 0F0A 0776 381828 55
Filtered 0xB1 data: AAB10402300758039806EA3818282855
[... cut ...]
Resulting 0xB0 data: AA B0 2F 04 08 0230 074E 038E 0F0A 38182828281828282818082828282828281818281828181818181818182828282818181828 55
Filtered 0xB1 data: AAB1030226038E07622818181855
Resulting 0xB0 data: AA B0 0C 03 08 0226 038E 0762 28181818 55
Filtered 0xB1 data: AAB104008C03F200E6037A3818181A55
Resulting 0xB0 data: AA B0 0E 04 08 008C 03F2 00E6 037A 3818181A 55
Filtered 0xB1 data: AAB104020803B607800F323818282828282828282818181818181818281818281828181828282828181818182828182855
Resulting 0xB0 data: AA B0 2F 04 08 0208 03B6 0780 0F32 38182828282828282828181818181818182818182818281818282828281818181828281828 55

```

kollera, 2019-10-27:
[issue119](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/119) sounds interesting.
so i flashed the mentioned [hex release](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/DG-R8H/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) instead of compiling & flashing master branch.

finally got A6 (advanced sniffing) results! \o/
```
[275716] [RF] received A6060188088D0F0E
[275718] [RF] Action 0xA6
[332717] [RF] received A6060188088D0F0E
[332719] [RF] Action 0xA6
[389858] [RF] received A6060188088D0F0E
[389861] [RF] Action 0xA6
[446719] [RF] received A6060188088D0F0E
[446721] [RF] Action 0xA6
[503719] [RF] received A6060188088D0F0E
[503721] [RF] Action 0xA6
```
another [rtl_433](https://github.com/merbanan/rtl_433/) simultaneously scan
```
rtl_433 version 19.08-24-g14c5b9a branch master at 201910071222 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ak/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 1 out of 140 device decoding protocols [ 13 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-27 14:54:44
model     : Mebus/433    Address   : 31
Channel   : 1            Battery   : OK            Unknown 1 : 0             Unknown 2 : 15
Temperature: 19.10 C     Humidity  : 13 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-27 14:56:38
model     : Mebus/433    Address   : 31
Channel   : 1            Battery   : OK            Unknown 1 : 0             Unknown 2 : 15
Temperature: 19.10 C     Humidity  : 13 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-27 14:57:34
model     : Mebus/433    Address   : 31
Channel   : 1            Battery   : OK            Unknown 1 : 0             Unknown 2 : 15
Temperature: 19.10 C     Humidity  : 13 %

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-27 14:59:29
model     : Mebus/433    Address   : 31
Channel   : 1            Battery   : OK            Unknown 1 : 0             Unknown 2 : 15
Temperature: 19.10 C     Humidity  : 13 %

```
close to  a solution? :thinking:

aleksey2000, 2019-11-10:
> rtl_433

Hi, could you give more details, how you got data from rf bridge and process them? 

> [issue119](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/119) sounds interesting.
> so i flashed the mentioned [hex [...]

Hi, could you please, give more details, how you got data from rf bridge (console/mqtt) and how you  processed them?

kollera, 2019-11-14:
i'm using [espurna](https://github.com/xoseperez/espurna) for all my sonoff devices.
got the data from the debug log after hacking [rfbridge.ino](https://github.com/xoseperez/espurna/blob/dev/code/espurna/rfbridge.ino) to log out received A6 messages.

did not process them, it's the 'raw' data received from the EFM.

bkenobi, 2020-03-09:
@kollera It looks like you have tried compiling your own firmware from a modified master.  What did you have to do to get Simpicity Studio 4 working?  I am attempting to add my own protocol but I can't even load the master for whatever reason so I can't see the code to get started.