Portisch/RF-Bridge-EFM8BB1 #145: RfRAW 177 or RfRAW AAB155 cmd using fixed

supertobi265, 2019-12-27:
Hello everyone.
I am new here. I am using a RF- Bridge with portisch firmware  for decoding a Dooya 1602. That works very well, but my problem is, that the RfRAW mode is not switched on for everytime. Sometimes it works 10 minutes or 10 hours and when I switched off the power from the bridge is the RfRAW mode also switched off. I hope evereryone have good idea.

Greatings from Magdeburg Germany

marceltech, 2020-01-18:
hello I stuck on the same, did you get it solved?

aymeric106, 2020-04-14:
Hi, I think it is expected  on reboot

> 
> RfRaw | This command only works when the firmware has been updated with Portisch firmware. Refer to the Portisch wiki for details.Learning and Decoding RF Codes with Portisch Firmware0 = Set iTead default firmware support and messages (default on restart)
> -- | --

You can add a rule to trigger on boot ( trigger -> System#Boot) 
and perhaps like this trigger RfRaw#Data=OFF

Aymeric

aymeric106, 2020-04-14:
perhaps this also

https://github.com/arendst/Tasmota/issues/7064

vampywiz17, 2020-05-14:
@marceltech @supertobi265 

Old .hex solve this?