Portisch/RF-Bridge-EFM8BB1 #146: Certain commands cause continuous beep

Denao2, 2020-01-14:
I am ruuning firmware V3

I have an issue when sending certain bucket strings. When these commands are issued it causes the internal buzzer to emit a continuous beep until the unit is reset. It only occurs after the unit has been powered on for a while. The commands are generated using this tool: https://github.com/sehaas/yfmos. It genrates a web call to the rfbridge. I have three blinds but one works fine, the other two have the issue.

Example that works:
http://192.168.4.127/cm?cmnd=Backlog%20RfRaw%20AA%20B0%203F%2005%2008%2009E2%2012CA%2004F6%20028A%206AE0%20000000000000001222233333323333332223333233332223333233233233332332332332233233223333333323333223333234%2055%3B%20RfRaw%200

Example that cuases the beep after the bridge has been on for a while:
http://192.168.4.127/cm?cmnd=Backlog%20RfRaw%20AA%20B0%203E%2005%2008%2009E2%2012CA%2004F6%20028A%206AE0%2000000000000000122223333332333333233223323333233223323333333333332332222223323333223333332332233332234%2055%3B%20RfRaw%200

Both work in that the blinds respond and the bridge continues to send and recieve commands.

Any thoughts much appreciated.

bvweerd, 2022-07-09:
Have you ever found a reason why or a solution for this issue? 2.5 years later I encounter the same ;-)

Denao2, 2022-07-09:
Hi, sorry no I did not find a solution.