Portisch/RF-Bridge-EFM8BB1 #147: Sonoff iFan03 compatibility?

mgoeller, 2020-01-23:
Has anyone every tried to flash the Portisch firmware on the RF chip of the Sonoff iFan03?

Goal: I would like to make the iFan03 controllable with a different remote than the stock one.

Picture of the iFan03 pcb layout under the rf shield:
[image]

Picture of test points on the bottom PCB side:
[image]

I guess I would have to connect the C2CK, C2D and any GND pin to an Arduino Mega (waiting for my order currently...) and try to read / flash the firmware with efm8-arduino-programmer.

I'm open for any tips, suggestions and will update this post as soon as I received my Arduino Mega.

hugofran, 2021-09-24:
Hi @mgoeller - did you ever have success with this project?

mgoeller, 2021-09-24:
I unfortunately did not continue on this project.

closed 2021-09-24 by mgoeller