Portisch/RF-Bridge-EFM8BB1 #148: Understanding how to decode from B1 to B0 RF Code

autofrank, 2020-01-24:
### Latest firmware used?
I;m running 6.1.1 on a Rf Bridge and also have the Portisch firmware update
I don't want to upgrade as I already have a remote working from the the RF bridge and I don't want to break that 

### Information
The remote is AC-114=016. It is from an AM-15 RF blind motor
https://www.amazon.co.uk/gp/product/B07RWTYDXG/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

### Sniffed data
AA B1 05 0254 147D 0117 5190 B360 20202020202020201002200220202002022002022020022020020220022002022020200202202002200202020200 55
AA B1 05 0259 148A 010F 5190 B34C 20202020202020201002200220202002022002022020022020020220022002022020200202202002200202020200 55
AA B1 05 0254 147D 0117 5190 B360 20202020202020201002200220202002022002022020022020020220022002022020200202202002200202020200 55

Hi,
I posted some sniffed code an example but am really interested in understanding the process as I think I need to sniff more codes to eventually find a good one. I'm familiar with the process havenig gotten help to solve this https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23

I used the BitBucket converter (http://bbconv.hrbl.pl/) on a number of code but no luck but I read the the BB could be out of date or it may not work for al

I worked with this code 
AA B1 05 0254 147D 0117 5190 B360 20202020202020201002200220202002022002022020022020020220022002022020200202202002200202020200 55

and tried to follow this process - https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Decode-0xB1-sniffed-data

Starting with the buckets
Bucket 0 - 0254 (596ms)
Bucket 1 - 147D (5245ms)
Bucket 2 - 0117  (279)
Bucket 3 - 5190  (20880)
Bucket 4 - B360  (45920)

Bucket 4 is the longest bucket and there fore  the (first) sync bucket

Working with the data 
20202020202020201002200220202002022002022020022020020220022002022020200202202002200202020200

The first nibble is 2 but the sync bucket is 2 and in the example they matched so I got stuck here and couldn't determine second sync bucket

Could some body explain this part ?

After that 
Bucket 1 is longer than bucket 0 (in my case)
a 10 means bit 1 and a 01 means bit 0

Could somebody explain what this means and how it impacts the next part

And I'm then pretty much lost
`First remove the high/low bucket marking by doing a logical and by 0x77 with each data byte:
818190908181908190909090908190819081818190909091
After AND 0x77:
010110100101100110101010100110011001010110101011

RF data: 01 01 10 10 01 01 10 01 10 10 10 10 10 01 10 01 10 01 01 01 10 10 10 11
Binary:  0  0  1  1  0  0  1  0  1  1  1  1  1  0  1  0  1  0  0  0  1  1  1  1
So the result is 0x32FA8F (b001100101111101010001111), 24 bit of data.`

I'll be sending the code back through a rfraw command 

As  I said at the start if I can get to understand the process I can work with a lot of codes to find one that works

Al help appreciated 

Thanks

autofrank, 2020-01-25:
Picture of remote and model number for reference
[image]

I made a little progress

Tasmota on the RF Bridge was on 6..1.1 -I used the OTA update to bring it to 8.0.1

I then went to update the Portisch FW. I tried using the recommended way - 3.3V from mt FTDI232
and linking GPIO4 to C2Ck and GPIO5 to C2D. I kept on getting "Failed to write Flash". I tried a lot and then tried reversing the C2Ck and C2D and it worked 

This is the board I have 
[image]

 and this is the way it need to be setup (The picture isn't great so I labelled the wires
[image]

I used the following location for the Portisch FW Hex file
https://github.com/Portisch/RF-Bridge-EFM8BB1
I downloaded the project zip and extracted all files. I then used the Tasmota FW Update feature to update the RF Flash

-- I am now getting more consistent codes on capture using RFRAW 177
`15:12:01 CMD: Group 0, Index 1, Command "RFRAW", Data "177"
15:12:01 RSL: RESULT = {"RfRaw":"ON"}
15:12:01 RSL: RESULT = {"Time":"2020-01-25T15:12:01","RfRaw":{"Data":"AAA055"}}
15:12:05 RSL: RESULT = {"Time":"2020-01-25T15:12:05","RfRaw":{"Data":"AA B1 03 00E6 0276 026C 281908 55"}}
15:12:09 RSL: RESULT = {"Time":"2020-01-25T15:12:09","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:13 RSL: RESULT = {"Time":"2020-01-25T15:12:13","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:16 RSL: RESULT = {"Time":"2020-01-25T15:12:16","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:19 RSL: RESULT = {"Time":"2020-01-25T15:12:19","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:21 RSL: RESULT = {"Time":"2020-01-25T15:12:21","RfRaw":{"Data":"AA B1 03 00DC 0276 026C 281908 55"}}
15:12:24 RSL: RESULT = {"Time":"2020-01-25T15:12:24","RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
15:12:24 RSL: RESULT = {"Time":"2020-01-25T15:12:24","RfRaw":{"Data":"AA B1 04 146E 024E 010E 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
15:12:26 RSL: RESULT = {"Time":"2020-01-25T15:12:26","RfRaw":{"Data":"AA B1 03 00FA 0258 0262 281908 55"}}
15:12:28 RSL: RESULT = {"Time":"2020-01-25T15:12:28","RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
15:12:30 RSL: RESULT = {"Time":"2020-01-25T15:12:30","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:33 RSL: RESULT = {"Time":"2020-01-25T15:12:33","RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
15:12:35 RSL: RESULT = {"Time":"2020-01-25T15:12:35","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:37 RSL: RESULT = {"Time":"2020-01-25T15:12:37","RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
15:12:40 RSL: RESULT = {"Time":"2020-01-25T15:12:40","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:42 RSL: RESULT = {"Time":"2020-01-25T15:12:42","RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
15:12:43 RSL: RESULT = {"Time":"2020-01-25T15:12:43","RfRaw":{"Data":"AA B1 04 1478 0258 010E 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
15:12:50 RSL: RESULT = {"Time":"2020-01-25T15:12:50","RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
15:12:52 RSL: RESULT = {"Time":"2020-01-25T15:12:52","RfRaw":{"Data":"AA B1 03 00DC 0280 0276 281908 55"}}
15`

I tried BitBucket on all of them and none worked :-(
I think I'll have to manually convert and will try that again now so if anybody has hints or advise I'd appreciated it 

Thanks

autofrank, 2020-01-25:
I tried to manually convert them @Portisch 

AA B1 03 00FA 0262 0258 281908 55
Bit Bucket gives me AA B0 0B 03 08 00FA 0262 0258 281908 55

0xAA: sync start
0xB0: command
0x0B: len command (11 in decimal)
0x03: bucket count
0x08: repeats
buckets 0-2
data
0x55: sync end

but the len looks wrong as the data is 281908 
len should be 06 and I increased the number of repeats to 20 (14 in hex)

giving me 
AA B0 06 03 14 00FA 0262 0258 281908 55

but that doesn't work either ...

I'm a little lost as I'm not sure whether my approach or bit bucket in wrong or maybe the remote is supported

autofrank, 2020-01-25:
another sample of codes I'll try and work through

{"RfRaw":{"Data":"AAA055"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 04 146E 024E 010E 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 04 146E 024E 0118 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0276 026C 281908 55"}}
{"RfRaw":{"Data":"AA B1 05 0186 14B4 028A 00DC 024E C8090A38 55"}}
{"RfRaw":{"Data":"AA B1 03 00F0 0276 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 07 1450 0276 0140 00D2 006E 05C8 14BE E8192A192B1B1B19292B19292B1B192B1B19292C1A5A1A2C1A2C1A292C1C1B192B1B1B1B1B1B1B192B1B1B1B1B1B1B1B1B192B1B1B1B19292B1B192A1929292929 55"}}
{"RfRaw":{"Data":"AA B1 03 010E 0258 0258 281908 55"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0280 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 04 146E 0258 010E 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 04 146E 0258 010E 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 03 00F0 0280 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 04 146E 024E 0118 146E 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1A1A1A192A1A1A1A19 55"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0276 026C 281908 55"}}
{"RfRaw":{"Data":"AA B1 03 00E6 0276 0276 281908 55"}}
{"RfRaw":{"Data":"AA B1 03 00F0 0276 026C 281908 55"}}

autofrank, 2020-01-26:
I finally got one to work for the "Blind Up" command

Sniffed
"RfRaw":{"Data":"AA B1 04 1464 0262 012C 1496 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A1929292929292A19292929 55"}}

Converted using Bit Bucket
AA B0 4C 04 08 1464 0262 012C 1496 38192A192A1A1A19292A19292A1A192A1A19292A192A19292A192A19292A1A1A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A1929292929292A19292929 55

autofrank, 2020-01-26:
For a given remote does anybody know if you expect the timing buckets to be the same for two different commands like Up and Down ?
I had it with another remote and was wondering if it was a coincidence ...

closed 2020-01-26 by autofrank

rrlevy, 2021-01-19:
I'm trying to replicate this procedure to controle my shades with the exact same RF Remote (AC114-01B) and Sonoff RF Bridge, but I can't make it work.

I installed the most recent Tasmota firmware as of today and flashed the Portisch RF firmware too.

I have my shades motor paired with the AC114-01B remote with the UP, DOWN and STOP buttons, and I'm trying to make the Sonoff RF Bridge to command the motor by sniffing and copying the remote codes.

I sniffed the codes by pressing the DOWN button several times at the remote, after sending the "rfraw 177" command at the tasmota bridge console:

```
16:51:14 CMD: rfraw 177
16:51:14 RSL: stat/tasmota_C23443/RESULT = {"RfRaw":"ON"}
16:51:14 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:14","RfRaw":{"Data":"AAA055"}}
16:51:27 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:27","RfRaw":{"Data":"AA B1 03 023A 0122 024E 28181909 55"}}
16:51:27 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:27","RfRaw":{"Data":"AA B1 04 1414 024E 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:28 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:28","RfRaw":{"Data":"AA B1 03 0244 010E 024E 28181909 55"}}
16:51:28 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:28","RfRaw":{"Data":"AA B1 04 1414 0244 010E 141E 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:29 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:29","RfRaw":{"Data":"AA B1 03 0244 0118 024E 28181909 55"}}
16:51:29 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:29","RfRaw":{"Data":"AA B1 04 1414 0258 0118 141E 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:30 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:30","RfRaw":{"Data":"AA B1 05 0172 1478 029E 00B4 02B2 C090A380 55"}}
16:51:30 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:30","RfRaw":{"Data":"AA B1 03 00FA 0258 0258 281818 55"}}
16:51:30 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:30","RfRaw":{"Data":"AA B1 03 0244 010E 0244 28181909 55"}}
16:51:30 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:30","RfRaw":{"Data":"AA B1 04 140A 024E 010E 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:31 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:31","RfRaw":{"Data":"AA B1 04 0212 0136 00DC 0258 38181A0A 55"}}
16:51:31 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:31","RfRaw":{"Data":"AA B1 03 023A 0122 024E 28181909 55"}}
16:51:32 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:32","RfRaw":{"Data":"AA B1 04 1414 024E 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:32 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:32","RfRaw":{"Data":"AA B1 03 0244 010E 0244 28181909 55"}}
16:51:33 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:33","RfRaw":{"Data":"AA B1 04 1414 0258 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:33 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:33","RfRaw":{"Data":"AA B1 03 010E 0244 024E 281818 55"}}
16:51:33 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:33","RfRaw":{"Data":"AA B1 03 0244 010E 024E 28181909 55"}}
16:51:33 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:33","RfRaw":{"Data":"AA B1 04 1414 024E 010E 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:34 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:34","RfRaw":{"Data":"AA B1 03 023A 0118 0244 28181909 55"}}
16:51:35 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:35","RfRaw":{"Data":"AA B1 04 1414 024E 010E 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:35 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:35","RfRaw":{"Data":"AA B1 03 0244 0118 0244 28181909 55"}}
16:51:36 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:36","RfRaw":{"Data":"AA B1 04 1414 024E 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:36 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:36","RfRaw":{"Data":"AA B1 03 023A 0118 0244 28181909 55"}}
16:51:37 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:37","RfRaw":{"Data":"AA B1 04 1414 024E 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55"}}
16:51:38 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:51:38","RfRaw":{"Data":"AA B1 03 023A 0122 024E 28181909 55"}}
```

I got the most common code:
[... cut ...]
But nothing happens, the shade won't move. I only get this message at the console:

```
16:57:18 CMD: rfraw AA B0 4C 04 08 1414 024E 0118 1428 38192A192A1A1A19292A19292A1A19292A1A192929292A1A19292A1A1A19292A192A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A192A1A192A1A1A1929 55
16:57:18 RSL: stat/tasmota_C23443/RESULT = {"RfRaw":"ON"}
16:57:18 RSL: tele/tasmota_C23443/RESULT = {"Time":"2021-01-19T16:57:18","RfRaw":{"Data":"AAA055"}}
```

Am I missing something? Are there any additional configuration that must be done for it to work? Any tips on what I might try next?

peteakalad, 2021-01-19:
Which portisch rf bridge firmware are you using?  Know tasmota repo had old version which didnt work for some devices.  This is where to get latest:  https://github.com/Portisch/RF-Bridge-EFM8BB1/releases

Failing that try converting some of the other codes you received?  Or try the ones mentioned in this issue above if you havent?

rrlevy, 2021-01-19:
I used the version from Tasmota repo, but it seems to be the same as the latest Portisch:
The file name I used was RF-Bridge-EFM8BB1-20190220.hex and the date matches the latest version.

I tried the other codes, too... But the big one I used are more in line with what is expected for this kind of remote, I guess.

peteakalad, 2021-01-19:
Give the release 0.4 hex here a go - I diff-ed it and the tasmota version (https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffRfBridge_efm8bb1/RF-Bridge-EFM8BB1-20190220.hex)  and they do differ even though they are seemingly the same date.

rrlevy, 2021-01-19:
I'll try that later today!

I just checked the version by typing rfraw 255 and it tells me it's version 03:
{"Data":"AA0355"}

rrlevy, 2021-01-19:
No luck...

I flashed the new firmware from Portisch repo, but nothing changed. 

Even the version number shown with rfraw 255 is still AA0355, but I looked it up and it seems to be an issue with the release anyway.

But the new version also didn't work, same behavior.

I tried all codes that I get when listening (event the short ones), and none of them work...

peteakalad, 2021-01-19:
Oh well..  no other ideas here