Portisch/RF-Bridge-EFM8BB1 #151: Sonoff RF Bridge, A6 sniffing does nothing

Fettkeewl, 2020-02-05:
### Latest firmware used?
Keil 8051 V9.53
Tasmota 8.0.1

### Information
Sonoff RF Bridge

### Sniffed data
None......... thats the issue, read on.

### Setup
1 Sonoff Bridge
4 Sonoff DW1 door/window sensors
2 Random 433 mhz thermometers (noname, looks like Digoo DG-TH8380, 3 channel temp/RF)
1 433 mhz doorbell

### Operational modes and results

- Mode: RfRaw 0 (normal)
    1.  Picks up and decodes signals from DW1
    2. Nothing else, as expected, other devices do not follow standard protocol.

- Mode: RfRaw 1 (Portisch enabled)
    1. Picks up and decodes signals from DW1 (presented in another format)
    2. Nothing else is picked up that is on 433 Mhz,

- Mode: RfRaw 166 / AAA655
   1. Nothing is picked up .... not even the DW1

- Mode: RfRaw 177 / AAB155
   1. Everything is picked up in some way or another

###  To the issue...
Why is A6 advanced mode not doing anything? I mean I've read the Wiki 
it states: **A6  advanced sniffing by predefined protocols in RF_Protocols.h**
Looking at RF_Protocols.h the Sonoff original protocol is supported?
As per the comment copied below
```
/*
 * PT2260, EV1527,... original RF bridge protocol
 * http://www.princeton.com.tw/Portals/0/Product/PT2260_4.pdf
 */
```

So at the very least mode A6/166 should pickup the same data that mode 1 does. 
What gives?

Fettkeewl, 2020-02-07:
@Portisch would you mind responding? :)

hongtat, 2020-03-14:
Rfraw 0 (normal)
"RfReceived":{"Sync":11080,"Low":410,"High":1020,"Data":"**_603E1F_**","RfKey":"None"}}

RfRaw 1 (same as A4)
"RfRaw":{"Data":"AAA42B48019A03E8**_603E1F_**55"}}

RfRaw 166 (A6)
The PT226X protocol defined in A6 is using a **fixed** "Sync" of 10850, so it won't capture sync time that is not 10850 (+-). The standard Rfraw 0 (or rfraw 1 / A4) uses a **dynamic** "Sync" range.

If you have devices that use the standard protocol and non-standard protocol, then A6 will not work (if you are trying to capture the state). I'm not sure why A6 cannot use dynamic range..
 
See - https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/125#issuecomment-505404568