Portisch/RF-Bridge-EFM8BB1 #152: Sonoff Rf bridge can't transmit the right code

Svenbosma, 2020-02-17:
Hi all,
I've recently bought a sonoff rf bridge and flashed it with sonoff, after doing this, I found out that my 433MHz plug transmitter was not compatible with the standard software so I downloaded and installed the Portisch software. I got it all setup, but when I try to send the code copied from the original transmitter, my plug doesn't respond. 
Here are my steps:
-I inserted `RfRaw 177` in the console of the webpage
-I clicked the transmitter button several times and combined the right code
-I converted this code to a B0 code
-I send this B0 to the plug with the code: `backlog rfraw AAB023040801A404420B7C1C5C0110101010011001101010100101010110011001101001012355; rfraw 0`
Somehow sending the code works, but the plug doesn't pick it up and change it's state to 'on'.
Does anyone know why and knows how to help me?

Fettkeewl, 2020-02-19:
Best guess, do as I did:
* Collect about 15-20 sniffs from a single button 
* Convert all to B0
* test one by one intill you hit a working one 
* save working one and continue with rest 

Get at least two working codes. Repeat for all buttons on your remote.
It's an annoying procedure I know but it was worth it.. All 8 buttons.. Later... And I have 2-3 remotes more that I could mimic.. Sigh
Sure wish I knew how to decode the data ^^

Taxxian, 2020-02-21:
I am always using the whole thing like this:
rfraw AA B0 2C 04 08 1A90 010E 029E 271A 38192A192A1A19292A1929292A192A1929292A1929292A192A19292A19292A1A1929 55
That never failed me so far...

So far I found only 1 device that had a code I could not sniff, but that does not seem to be your problem.

Svenbosma, 2020-02-21:
Thank you, I did not try enough codes. After about 6 different codes, it worked. What a process for just one button, but it works now, so thanks again.

closed 2020-02-21 by Svenbosma