Portisch/RF-Bridge-EFM8BB1 #154: Convert RTL_433 decoding into Portisch firmware?

bkenobi, 2020-03-05:
There are many transmitters that have already been decoded and implemented in various projects.  I have not had much luck getting my Accurite weather station or temperature sensors to detect correctly with the Sonoff Bridge using Tasmota/Portisch latest versions.  I confirmed that these are easily detected and decoded in RTL_433 using a basic usb SDR stick.  Is there any way to leverage the the existing knowledge of timings and enter them into the Portisch firmware?  I have all the information but do not know how that feeds into the firmware.

thermseekr, 2020-03-06:
Hello @bkenobi

I have made more than a few tries to implement the timings of known protocols for lacrosse temperature & humidity sensors. I reached partial success only, with a couple sensors being correctly decoded with a very few minor glitches every now and then, but a couple other sensors weren’t detected no matter what. All sensors same model. I had to invest a lot of hours figuring out the whole buckets & timings structure based only on the limited documentation available. Installing and running Simplicity Studio was another challenge, that thing has nothing about simple, at least on a Mac. I made several tries to contact Portisch but apart for my first post when he told he was gonna take a look (that was before I found info on the protocol) he never replied again, and as far as I could notice he didn’t reply to any other calls from other people as well, which makes me think this has become a halted project.

What I did was grab a RPi I had lying around, plugged a $10 SDR dongle, installed Raspbian and rtl_433 and now I am very happy seeing all my four sensors in my home assistant setup. Simple, fast, reliable. And the Sonoff Bridge is buried under a pile of home automation leftovers.

From a few releases back rtl_433 includes MQTT as an output format so you don’t even need to pipe its output to mosquito transmitter anymore. So apart from the small extra cost I don’t see a reason for keep struggling with the sonoff bridge + portisch solution anymore. I suggest moving on to this simpler solution with plenty of support as I did.

Cheers,
Tales
 
Enviado do meu iPhone

bkenobi, 2020-03-06:
I have a usb SDR already and the only reason I didn't use it a few years ago was RTL_433 didn't exist.  I used the SDR to detect the pulses and decode a few sensors for use in an Arduino project.  While that was a success, the range was insufficient.  I tried to build a variant of the project using a better 433 receiver but had issues with the drivers.  I switched to an ESP8266 based approach but again software issues.  After shelving the project I watched for someone else to solve my issues by asking google once every few months.  Although 433 to WiFi MQTT gateways are shown on Instructables and other sites, nobody ever shared their full project for recreation.  The Sonoff Bridge + Tasmota + Portisch firmware looked like a perfect option and it's been sitting half working for over a year on my bench.  It is pretty ironic that the first module I purchased is now fully working in my original vision only 5 years down the road.

The only reason the 433_RTL + RPi option isn't ideal for me right now is that my soon-to-be spare RPi is a 1B which doesn't have WiFi.  I can get a new one certainly but will want to test that RTL_433 is a full option this time around.  Fool me once, twice, three, four time perhaps...but not five!

bkenobi, 2020-03-06:
FWIW, the main reason I've been looking at Arduino, ESP8266, and Sonoff is that they are all hardware based setups meaning if power fluctuates, I don't have to worry about rebuilding an OS like I would with an RPi.  My HA system is running on a RPi1B running HomeGenie and has been great for years.  I have a RPi3B+ that I was going to upgrade to Home Assistant last year.  But, since I use X10 primarily and RF is not really well supported, I'm still on the old system.

Working with these boards, it's rare but a power fluctuation can wipe out the SD card.  I have my main system on a PC UPS but the 433 gateway would be located wherever it needs to be for best reception of all devices.  Meaning, I don't want to install a UPS under a dining room table potentially.

TimothyGaray, 2020-03-06:
This is a real shame...  I just went through all the excitement of flashing Tasmota on my RF Bridge and then Portisch and none of my RF devices are detected.  I can get the raw data but doesn't appear to be much I can do with that.  Same as you: Acurite 592TX plus generic F007_TH and ancient Oregon Scientific temp sensors (formerly RadioShack).  I even brought home a generic motion sensor module to test out and nothing.

I put RFLink on my Arduino Mega 2560 and everything was detected perfectly right "out of the box" (except for that generic motion sensor).  Temperature sensors all work exactly as I hoped.

I just ordered  an SDR USB and going to try the rtl_433 & rtl-sdr on a Raspberry Pi and see how that works.  Already installed the apps in the RPi so should be able to just plug it in and see.

I've been reading all your various postings @bkenobi .  I too am running Home Assistant so the ultimate goal is integration into that which MQTT seems to be the shortest path.

I'll put the RF Bridge on the shelf for now and focus on other routes.

Thank you!

-TimG

bkenobi, 2020-03-06:
It is a shame since the form factor is so appealing!  I hadn't even gotten to testing range, so who knows if it's even capable of being centrally located.  It seems if you know the timings of the target signal, you should be able to manually enter it rather than try to capture it and convert using bitbucketconverter.  But without knowledge of how to add things (even if I could get Simplicity Studio to work), I'm at a dead end.  Sounds like I'm not the only one.

On an up side, the suggestion to look at RTL_433 on an RPi has spurred me to realize there are ways to configure the RPi OS such that the SD is read-only and the write portion is in ram or on another drive (e.g., USB stick).  That would mean it *should* be power outage tolerant.  I have not tested this, but it sound's like a possibility that could make RPi3B+ with usb SDR dongle feasible.

TimothyGaray, 2020-03-06:
The green light is cool too.  :-)

I have been fortunate and haven't had a corruption of the RPi OS because of a power outage.  I guess I should do more backups...

bkenobi, 2020-03-09:
I'm going to try one more time requesting the protocol in the format that Portisch has listed.  I know everything about the protocol sufficient to capture it in my own code (derived from other's work).  I haven't seen any evidence that Portisch is still lurking around as far as comments or updates, but I'm hoping asking in a way that he needs for inclusion will work.  I have not been to figure out how to use Simplicity Studio 4 so I'm lost how to try building my own protocol.  I have it installed and the EFM8BB1 loaded and Keil is definitely one of the packages they installed.  But, I see nothing in the repository that SS finds as a project file so I'm not sure how to compile without code...

TimothyGaray, 2020-03-10:
I bought a NooElec RTL-SDR, FM+DAB, DVB-T USB Stick Set with RTL2832U & R820T and plugged it into a RPi3B+ and it worked as well.  I haven't configured any data to MQTT for it yet.  It also did not detect my generic motion sensor so I'm guessing that is one of those devices that doesn't conform to any standard protocols.  I didn't plan to use it for HA anyway - was just playing around with it.

Right now RFlink is working perfectly for me on an Arduino Mega 2560 with NodeMCU WiFi running OpenMQTTGateway.

I'll keep watching here for updates.

Thanks!

halfbakery, 2020-10-03:
When I first tried to use the RF-Bridge I just realized that many of the useful protocols were missing. I managed to add my wt0124 pool thermometer, but I had to realize that the EFM8BB1 MCU is so powerless. Upgrading the firmware is not an easy process, and most of the users have to compile their own firmware to select the required protocols. That was the point when I found the rtl_433 project.

My idea is to create a bridge between theese two processes, this [PR](https://github.com/merbanan/rtl_433/pull/1491) is the rtl_433 side. I hope I could release the first version of the Portisch side on the following days.

thermseekr, 2020-10-06:
Hello,

I reached the point where I compiled the firmware to receive data from my
Lacrosse temperature & humidity sensors, but I only managed to receive data
from 3 out of 4 sensors, and of those 3, data from one of them eventually
came in corrupted. And that's when I learned about SDR receivers and
rtl_433. The sonoff bridge is a nice little device, and the effort put in
from the project owner to reprogram the EFM8BB1 has to be recognized, but
at this point I don't think it makes a lot of sense to keep putting energy
into this. Apart from the EFM chip being kinda powerless there is also the
fact that the antenna in the sonoff bridge is far from being a
technological marvel. SDR dongles are cheap, use external antennas, there's
rtl_433 being actively developed, and on goes the advantages. The only
advantage I see in using the sonoff bridge is that you can put it anywhere,
no need to be plugged into your server or any device capable of running an
operating system.

Regards,
Tales