Portisch/RF-Bridge-EFM8BB1 #155: Detect Yale HSA6010/HSA6020 transmissions

gRoberts84, 2020-03-08:
### Latest firmware used?

Latest firmware as suggested by https://github.com/arendst/Tasmota/wiki/Sonoff-RF-Bridge-433, which is https://github.com/arendst/Tasmota/blob/master/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181127.hex

Based on the Sonoff RF Bridge (V2)

### Information
HSA6000 series devices, specially the HSA6010 and HSA6020 Door and PIR Sensors.

### Sniffed data

Door sensor 1:

```
Open
20:21:48 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:48","RfRaw":{"Data":"AA B1 03 01EA 053C 140A 0110011001010101010101011002 55"}}
20:21:49 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:49","RfRaw":{"Data":"AA B1 03 01EA 0546 15EA 0110011001100110100110100102 55"}}
20:21:49 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:49","RfRaw":{"Data":"AA B1 03 01EA 0528 140A 0110011001010101010101011002 55"}}
Open
20:21:58 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:58","RfRaw":{"Data":"AA B1 03 01EA 053C 140A 0110011001010101100101010102 55"}}
20:21:58 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:58","RfRaw":{"Data":"AA B1 03 01F4 0532 1400 0110011001010110011001011002 55"}}
20:21:58 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:58","RfRaw":{"Data":"AA B1 03 01E0 0582 1428 0110011010010110101010010112 55"}}
20:21:58 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:58","RfRaw":{"Data":"AA B1 03 01EA 053C 15EA 0110011001100110100110100102 55"}}
20:21:59 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:21:59","RfRaw":{"Data":"AA B1 03 01EA 053C 140A 0110011010010110101010010112 55"}}
Open
20:22:05 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:05","RfRaw":{"Data":"AA B1 03 01FE 0532 13F6 0110011001010101010101011002 55"}}
20:22:06 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:06","RfRaw":{"Data":"AA B1 03 01FE 0528 140A 0110011001010101100101010102 55"}}
20:22:06 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:06","RfRaw":{"Data":"AA B1 03 01EA 053C 16BC 0110011001010101100101010102 55"}}
Open
20:22:17 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:17","RfRaw":{"Data":"AA B1 04 0190 056E 02E4 13F6 0112011201010101010101011203 55"}}
20:22:17 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:17","RfRaw":{"Data":"AA B1 03 01D6 0550 1414 0110011001010101100101010102 55"}}
20:22:18 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:18","RfRaw":{"Data":"AA B1 03 01E0 0546 1400 0110011001010101010101011002 55"}}
20:22:18 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:18","RfRaw":{"Data":"AA B1 03 01F4 0528 13F6 0110011001010101010101011002 55"}}
20:22:18 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:18","RfRaw":{"Data":"AA B1 03 01EA 0532 1400 0110011001010101010101011002 55"}}
Open
20:22:26 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:26","RfRaw":{"Data":"AA B1 03 01E0 053C 13F6 0110011001010101010101011002 55"}}
20:22:26 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:26","RfRaw":{"Data":"AA B1 03 0212 0528 15E0 0110011001100110100110100102 55"}}
20:22:26 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:26","RfRaw":{"Data":"AA B1 03 01EA 0532 141E 0110011001101001100110010102 55"}}
20:22:26 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:26","RfRaw":{"Data":"AA B1 03 01E0 056E 140A 0110011001010101010101011002 55"}}
20:22:26 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:22:26","RfRaw":{"Data":"AA B1 03 01F4 053C 15EA 0110011001100110100110100102 55"}}
```

Door Sensor 2:

```
Open
20:24:11 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:11","RfRaw":{"Data":"AA B1 04 01CC 0546 02E4 15E0 0112011201120112120112120103 55"}}
20:24:12 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:12","RfRaw":{"Data":"AA B1 03 01F4 0532 1400 0110011001010101010101011002 55"}}
20:24:12 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:12","RfRaw":{"Data":"AA B1 03 01EA 053C 15E0 0110011001100110100110100102 55"}}
20:24:12 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:12","RfRaw":{"Data":"AA B1 03 01E0 0550 140A 0110011001010101010101011002 55"}}
20:24:12 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:12","RfRaw":{"Data":"AA B1 03 01EA 053C 1414 0110011001010101010101011002 55"}}
Open
20:24:17 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:17","RfRaw":{"Data":"AA B1 03 01E0 053C 15D6 0110011001100110100110100102 55"}}
20:24:17 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:17","RfRaw":{"Data":"AA B1 04 01C2 0550 02EE 140A 0112011201010101010101011203 55"}}
Open
20:24:23 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:23","RfRaw":{"Data":"AA B1 03 01EA 053C 15E0 0110011001100110100110100102 55"}}
20:24:23 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:23","RfRaw":{"Data":"AA B1 04 01D6 0550 02F8 140A 0112011001010101010101011203 55"}}
20:24:23 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:23","RfRaw":{"Data":"AA B1 04 01D6 055A 02F8 15D6 0110011201120110120112120103 55"}}
20:24:23 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:23","RfRaw":{"Data":"AA B1 04 01C2 0550 02DA 140A 0112011201010101010101011203 55"}}
20:24:23 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:23","RfRaw":{"Data":"AA B1 03 01F4 0532 15EA 0110011001100110100110100102 55"}}
20:24:24 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:24","RfRaw":{"Data":"AA B1 03 0212 0514 1400 0110011001101010011010100102 55"}}
20:24:24 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:24","RfRaw":{"Data":"AA B1 03 01FE 0528 140A 0110011001010101010101011002 55"}}
20:24:24 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:24","RfRaw":{"Data":"AA B1 03 01EA 053C 16EE 0110011001010101100101010102 55"}}
Open
20:24:34 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:34","RfRaw":{"Data":"AA B1 04 01CC 0550 02E4 15F4 0112011201120112120112120103 55"}}
20:24:34 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:34","RfRaw":{"Data":"AA B1 04 01CC 0546 02EE 140A 0110011201101012011212120103 55"}}
20:24:34 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:34","RfRaw":{"Data":"AA B1 03 01FE 0528 140A 0110011001010101010101011002 55"}}
Open
20:24:41 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:41","RfRaw":{"Data":"AA B1 04 01D6 0550 02EE 15D6 0112011201120110100112120103 55"}}
20:24:42 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:42","RfRaw":{"Data":"AA B1 03 01F4 0528 140A 0110011001010101010101011002 55"}}
20:24:42 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:42","RfRaw":{"Data":"AA B1 03 0208 051E 140A 0110011001010101100101010102 55"}}
20:24:42 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:42","RfRaw":{"Data":"AA B1 03 0208 0528 1400 0110011001010110011001011002 55"}}
20:24:43 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:43","RfRaw":{"Data":"AA B1 04 01B8 0550 02D0 16D0 0112011201010101010101011003 55"}}
Open
20:24:53 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:53","RfRaw":{"Data":"AA B1 03 01E0 0546 15EA 0110011001100110100110100102 55"}}
20:24:53 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:53","RfRaw":{"Data":"AA B1 03 01E0 0546 13F6 0110011001010110011001011002 55"}}
20:24:54 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:54","RfRaw":{"Data":"AA B1 03 01FE 0528 140A 0110011001010101010101011002 55"}}
20:24:54 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:54","RfRaw":{"Data":"AA B1 03 01FE 0528 15EA 0110011001100110100110100102 55"}}
20:24:54 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:54","RfRaw":{"Data":"AA B1 03 01EA 053C 140A 0110011001010101010101011002 55"}}
20:24:54 MQT: tasmota-4594/tele/RESULT = {"Time":"2020-03-08T20:24:54","RfRaw":{"Data":"AA B1 04 01D6 0550 02EE 16F8 0112011201010101100101010103 55"}}
```

Not quite sure why, but you can see some times it will transmit 1-2 messages and another time, 4-5. 

I've covered the PIR sensors to avoid them interfering and I made a point of open/closing multiple times (close does not transmit) just in case. 

My head is fried (took me long enough to get the bridge flashed :( 

### Goal

I have MQTT sending messages to Home Assistant whenever a message is received. To avoid pulling my alarm to bits, it would be good to identify the protocol so I can send a notification when the sensors are triggered and I am away. 

RFLink is able to detect these however their source code is closed off so I cannot find the protocol they are using and reverse engineer it :( 

Anyone able to help out?

roscoecairney, 2020-05-17:
I've been playing with rtl_433 on the hsa6010. I've managed to detect door open events as well as open and close events for the tamper switch. I'm also not seeing door close events.

Robmak73, 2020-07-31:
hi
I am also trying to use the rf bridge with tasmota and porticsh to get notifications and execute other scenes from yale alarm door sensor {and pir sensors eventually}. There is little or no information on the web how to do this. I have had some success. Thought if we could share our findings we may be able to progress with our project.
If you would like to get in touch and let me know how far you have got since may.  I have been using mqtt node-red with the switch function with a regex to then trigger alexa routines. This does work but with some interference from other rf transmissions sometimes giving false alerts. I had given up with yale door sensor and made a contact sensor from a rf remote control and reed switch that is recognised without rf 177. But I would prefer to use the yale sensor.

PeterHaban, 2021-06-30:
Long story short, the door/window sensors only trigger when opened but you don't get a signal when they are closed again so not useable to detect if doors/windows are open. 

I've been playing with rtl_433 <- hsa6010 and have managed to figure out what some of the data means, details over here: https://groups.google.com/g/rtl_433/c/1yVOs7yoLw0/m/dkC3xLFfBwAJ?pli=1

I've figured out how to detect the different device type (door sensor and PIR to start with but also works for fob, siren etc) as well as how to decode trigger/tamper/bind (there should be a low battery alert as well but not got that one yet). I was contemplating to pump some data into Home Assistant from rtl_433+RTL-SDR on a RasPi when I found this so will now try to get Portisch onto my Sonoff RF Bridge to see if I can pick up the sensors.

PeterHaban, 2021-07-02:
Just in case someone find this useful, I couldn't find an easy way to decode with Sonoff RF so had a go at RFlink which supports the HA6000 system OOTB and works very well. RFlink bridge hooked up to HA and it decodes the door/window sensors as well as other components like the remote and I can link the incoming events to switches.
Obviously doesn't make it any more useable for detecting if a door/window is closed but works very well to pick up if a sensor gets triggered e.g. to send alerts via HA or trigger other automations. I have yet to figure out how these things send their "low battery" signals which is what I'm really after.