Portisch/RF-Bridge-EFM8BB1 #156: Livolo switches

msslava, 2020-03-08:
Hi,
I have Livolo RF switches. I can control them through arduino, but there I have many inconveniences. I don’t have a Livolo remote control.

I have Sonof RF bridge (2.0) -  (tasmota + your firmware without hardware hack).

Will I be able to configure the RF bridge without a Livolo remote control?

gRoberts84, 2020-03-09:
I haven't had any experience with the switches, but found this when having a look around for possible answers: https://kind3r.github.io/livolo-rf-codes/

msslava, 2020-03-09:
Thanks for answer!
What command should I enter to learn the switch?
```
RfKeyLivolo<id remote> <button code>
RfKey<id remote> <button code>

```???

gRoberts84, 2020-03-09:
Really wish I knew, I don't have the devices to test :(

helenclarko, 2020-08-12:
Can't believe I stumbled upon this, Exactly the information I needed.

https://kind3r.github.io/livolo-rf-codes/