Portisch/RF-Bridge-EFM8BB1 #157: New protocol request - Acurite 5-in-1 weather station

bkenobi, 2020-03-09:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!
Tasmota 8.1.0
Portisch Keil 9.53 (Mar 26, 2019)

### Information
What device you try to sniff data from? Description? Brand and type?
Acurite 5-in-1 weather station VN1TXC

### Sniffed data
Post your sniffed data by command [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) or [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6).

```
17:44:43 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:44:43","RfRaw":{"Data":"AA B1 03 017C 00F0 01A4 281819 55"}}
17:45:19 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:45:19","RfRaw":{"Data":"AA B1 03 017C 00E6 019A 281819 55"}}
17:45:55 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:45:55","RfRaw":{"Data":"AA B1 03 0172 00F0 01A4 281819 55"}}
17:46:31 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:46:31","RfRaw":{"Data":"AA B1 03 015E 0096 01C2 28090909 55"}}
17:47:07 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:47:07","RfRaw":{"Data":"AA B1 03 0140 0078 01EA 280809 55"}}
17:47:25 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:47:25","RfRaw":{"Data":"AA B1 04 0082 01E0 0140 01C2 381A281A2A 55"}}
17:47:25 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:47:25","RfRaw":{"Data":"AA B1 03 0118 0064 0208 280809 55"}}
17:47:43 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:47:43","RfRaw":{"Data":"AA B1 04 017C 00E6 00A0 0190 381818181A081A 55"}}
17:47:43 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:47:43","RfRaw":{"Data":"AA B1 04 008C 01C2 0104 01C2 38181818181928 55"}}
17:48:19 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:48:19","RfRaw":{"Data":"AA B1 03 014A 0082 01E0 28090909 55"}}
17:48:37 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:48:37","RfRaw":{"Data":"AA B1 03 0136 006E 01CC 280809 55"}}
17:48:37 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:48:37","RfRaw":{"Data":"AA B1 03 0190 00DC 019A 281819 55"}}
17:48:55 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:48:55","RfRaw":{"Data":"AA B1 03 0172 00E6 01AE 281819 55"}}
17:49:13 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:49:13","RfRaw":{"Data":"AA B1 04 008C 01CC 0104 01CC 381928 55"}}
17:49:31 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:49:31","RfRaw":{"Data":"AA B1 03 0172 00F0 01A4 281819 55"}}
17:49:49 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:49:49","RfRaw":{"Data":"AA B1 03 0190 00DC 0190 281819 55"}}
17:50:07 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:50:07","RfRaw":{"Data":"AA B1 03 0118 0064 0208 280809 55"}}
17:50:25 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:50:25","RfRaw":{"Data":"AA B1 03 017C 00DC 0190 281909 55"}}
17:50:25 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:50:25","RfRaw":{"Data":"AA B1 03 0186 00DC 0190 281819 55"}}
17:50:43 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:50:43","RfRaw":{"Data":"AA B1 03 0172 00E6 01A4 281819 55"}}
17:51:01 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:51:01","RfRaw":{"Data":"AA B1 03 0186 00E6 019A 281819 55"}}
17:51:19 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:51:19","RfRaw":{"Data":"AA B1 03 014A 008C 01EA 2809090909 55"}}
17:51:37 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:51:37","RfRaw":{"Data":"AA B1 04 00A0 01B8 0104 01CC 381928 55"}}
17:51:55 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:51:55","RfRaw":{"Data":"AA B1 04 015E 00FA 00A0 01C2 381A0A0A081A0A 55"}}
17:52:13 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:52:13","RfRaw":{"Data":"AA B1 04 0172 00FA 00A0 019A 381A0A 55"}}
17:52:13 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:52:13","RfRaw":{"Data":"AA B1 03 00AA 01B8 01C2 281908 55"}}
17:52:31 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:52:31","RfRaw":{"Data":"AA B1 03 017C 00E6 01A4 281819 55"}}
17:53:07 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:53:07","RfRaw":{"Data":"AA B1 03 0172 00E6 01A4 281819 55"}}
17:53:25 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:53:25","RfRaw":{"Data":"AA B1 03 00A0 01B8 01C2 281818 55"}}
17:53:43 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:53:43","RfRaw":{"Data":"AA B1 04 0172 00FA 00AA 01AE 38181A 55"}}
17:54:01 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:54:01","RfRaw":{"Data":"AA B1 03 014A 006E 01C2 280809 55"}}
17:54:01 MQT: tele/tasmota/RESULT = {"Time":"2020-03-09T17:54:01","RfRaw":{"Data":"AA B1 04 0082 01E0 0140 01E0 381A28 55"}}
```

I captured data using an USB SDR at the same time:

```
Detected OOK package    2020-03-09 09:46:48
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-03-09 09:46:48
model     : Acurite-5n1  subtype   : 56            id        : 2825
channel   : A            sequence_num: 0           battery   : 0             wind_speed: 0.0 km/h
temperature: 32.7 F      humidity  : 99            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-03-09 09:46:48
model     : Acurite-5n1  subtype   : 56            id        : 2825
channel   : A            sequence_num: 1           battery   : 0             wind_speed: 0.0 km/h
temperature: 32.7 F      humidity  : 99            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-03-09 09:46:48
model     : Acurite-5n1  subtype   : 56            id        : 2825
channel   : A            sequence_num: 2           battery   : 0             wind_speed: 0.0 km/h
temperature: 32.7 F      humidity  : 99            Integrity : CHECKSUM
Analyzing pulses...
Total count:  207,  width: 135.16 ms            (33790 S)
Pulse width distribution:
 [ 0] count:   12,  width:  632 us [632;644]    ( 158 S)
[... 258 lines cut ...]
Pulse period distribution:
 [ 0] count:   12,  width: 1224 us [1224;1240]  ( 306 S)
 [ 1] count:  192,  width:  616 us [612;628]    ( 154 S)
 [ 2] count:    2,  width:  800 us [800;804]    ( 200 S)
Level estimates [high, low]:  15971,    219
RSSI: -0.1 dB SNR: 18.6 dB Noise: -18.7 dB
Frequency offsets [F1, F2]:    7581,      0     (+28.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 232, long_width: 424, reset_limit: 596, sync_width: 632
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=232,l=424,r=596,g=0,t=0,y=632'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 3
[00] {65} 34 f6 47 ff fa a6 9c b2 80
[01] {65} 24 f6 47 ff fa a6 9c a2 80
[02] {65} 14 f6 47 ff fa a6 9c 92 80
```

I also know the protocol is:
3 sync pulses of 600us (600 high, 600 low repeated 3 times).
Followed by data consisting of 400us and 200us pulses.
1 bit is 400 high then 200 low
0 bit is 200 high then 400 low
There are 2 types of packets that alternate.  Humidity and rainfall are sent in alternating packets.  Other data is provided in all packets.

If there is more information I can provide, let me know.  I tried to figure out how to add this to the firmware myself but could not get Simplicity Studio 4 set up in a way I could load the project.  If there is a guide on getting your project loaded, I can take another look.  Also, 2 other known protocols I would like to get incorporated.  If this Acurite weather station work with the bridge, I'll be able to provide these 2 other protocols as well.

bkenobi, 2020-05-14:
Apparently this project is dead if basic protocol additions sit endlessly.  If I could compile the source I would, but simplicity studio is not very simple.  <sigh>

Cabalist, 2020-05-14:
I'm sorry you aren't getting what you want.   Coming to open source projects with that sense of entitlement is rarely helpful and contributes to a lot of project owners burning out.   

If something is beyond your skill level it is not anyone's responsibility to provide it for you.

bkenobi, 2020-05-14:
That's certainly one way to look at it.  I have all of my code I've written shared for others and don't ask anything in return.  Writing code is not beyond my capability but apparently installing the specific compiler this project requires is.  I have commented in other issues here about this and others seem to have similar issues.  If I could get Simplicity Studio installed, I'd be able to contribute the solution which I've already done for Arduino.  FWIW, this is perhaps the first time someone has suggested I'm entitled, but cool.

thermseekr, 2020-05-16:
@bkenobi, at first I too found simplicity studio very very hard to install. I wrote about this already. I was trying on a Mac and one day I decided to try on Windows. Maybe the opsys has nothing to do with it and I was on a better day but fact is I managed to make it work on Win 10 and I was able to compile the sources to include my Lacrosse sensors.

After some time it even started to make sense, I mean the way Simplicity Studio is organized and how it works.

What happened at the end is that one of four sensors was never detected and the bridge eventually missed transmissions from a second sensor. Probably needed some tweaking on the timings but my energy and love for the sonoff bridge had all been wasted fighting with Simp Studio, and I just dropped it in the failure box and set up a SDR receiver on a RPi running rtl_433 under Raspbian.

You then have two choices, either keep fighting Simplicity Studio - it’s doable - or take the rtl_433 route. The latter works like a charm and inserting new protocols is far easier.

Cheers
Tales

Enviado do meu iPhone

coogle, 2021-03-13:
FWIW, I would also love to see this particular protocol (and other similar Acurite devices) implemented. I haven't tried my hand at Simplicity Studio yet, but I might sometime because I rather not run a full RPi and rtl_433 just to pull signals.

One thought that occurs to me, is I wonder how difficult it would be to simply translate the protocols already defined in rtl_433 to Portisch? I mean, ultimately, it's the same basic specs I think -- it's just a matter of transcribing.

Being an OSS developer for well over 20 years I can appreciate the idea that people shouldn't make demands. Consider this  however a big ole' thumbs-up on this protocol if someone has time to implement it (and I don't at least try).