Portisch/RF-Bridge-EFM8BB1 #158: RF Bridge constant beeping + Failed to erase RF chip

slamkalukas, 2020-03-11:
### Latest firmware used?
Be sure you tried the [latest release](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases) or the [newest binary](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) of the firmware!

### Information
What device you try to sniff data from? Description? Brand and type?

I am trying to install Portish on RF Bridge  R2 v1.0, tasmota is installed perfectly on the first try but I am having an issue with Portish. Each time i try to install it via Web GUI, I get an error "Failed to erase RF chip" and since one moment the buzzer of RF Bridge is constantly beeping (one continuous beep)

Does it mean that my RF Bridge is dead? I tried to power it from USB and UART as well, still the same issue.

Thanks for comments.
Lukas

bkenobi, 2020-03-12:
Make sure the switch is in the right position.

slamkalukas, 2020-03-12:
Switch is in OFF position. Can you help me to explain/find some documentation what does the constat beep sound means?

rplevka, 2020-04-10:
@slamkalukas, you have to bypass GPIO4 and GPIO5 data lines while upgrading the rf chip fw as described here:

https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/

not sure about the beep - I guess since the rf chip isn't in the flash mode it treats the imcoming fw data as a stream of invalid commands and beeps (but that's just my theory).

slamkalukas, 2020-04-19:
Thanks - another hub was successfully tasmotized and portisched :) The previous one is probably dead. Thanks for help!

closed 2020-04-19 by slamkalukas