Portisch/RF-Bridge-EFM8BB1 #162: Issue with Portisch and BF-301 Transmitter with EYA25 Tubular Motor

prbtt, 2020-04-08:
So, i'm trying to control a motorized blind with the EYA25 Tubular Motor using a BF-301 (433,92 MHz) transmitter.

It was all working fine for a month but then it just stopped working...

These are the codes that were working before:

DOWN:
23:16:27 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:16:27","RfRaw":{"Data":"AA B1 05 091A 065E 0140 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3A3B2 55"}}
23:16:36 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:16:36","RfRaw":{"Data":"AA B1 05 091A 0668 014A 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3A3B2 55"}}
23:16:39 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:16:39","RfRaw":{"Data":"AA B1 05 091A 0668 014A 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3A3A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3A3B2 55"}}
STOP:
23:18:42 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:42","RfRaw":{"Data":"AA B1 05 091A 0668 014A 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3B2A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3B2A3 55"}}
23:18:44 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:44","RfRaw":{"Data":"AA B1 04 024E 0186 00D2 0302 38181A 55"}}
23:18:44 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:44","RfRaw":{"Data":"AA B1 05 0910 0668 014A 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3B2A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3B2A3 55"}}
23:18:47 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:47","RfRaw":{"Data":"AA B1 04 024E 019A 00D2 0302 38181A 55"}}
23:18:47 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:47","RfRaw":{"Data":"AA B1 05 091A 065E 0140 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3B2A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3B2A3 55"}}
23:18:49 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:18:49","RfRaw":{"Data":"AA B1 05 0910 0668 014A 02B2 15D6 C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3B2A3B2A3B2A3A3A3A3A3B2B2A3A3A3B2A3A3B2A3 55"}}
UP:
23:19:12 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:12","RfRaw":{"Data":"AA B1 05 091A 065E 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
23:19:12 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:12","RfRaw":{"Data":"AA B1 03 0136 02A8 02A8 28181818 55"}}
23:19:15 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:15","RfRaw":{"Data":"AA B1 05 091A 0668 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
23:19:15 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:15","RfRaw":{"Data":"AA B1 03 0136 02A8 029E 28181818 55"}}
23:19:17 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:17","RfRaw":{"Data":"AA B1 05 0910 0668 014A 02B2 15EA C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
23:19:17 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:17","RfRaw":{"Data":"AA B1 03 012C 029E 02A8 28181818 55"}}
23:19:19 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:19","RfRaw":{"Data":"AA B1 05 091A 0654 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
23:19:19 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:19","RfRaw":{"Data":"AA B1 03 012C 02B2 029E 28181818 55"}}
23:19:21 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:21","RfRaw":{"Data":"AA B1 04 023A 019A 00DC 0316 38181A 55"}}
23:19:21 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:21","RfRaw":{"Data":"AA B1 03 0118 02B2 02B2 28181818 55"}}
23:19:22 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:22","RfRaw":{"Data":"AA B1 05 091A 065E 0140 02B2 15EA C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
23:19:22 MQT: sonoff-bridge-module/tele/RESULT = {"Time":"2020-03-04T23:19:22","RfRaw":{"Data":"AA B1 03 0136 02A8 02A8 28181818 55"}}

I tryed all types of hex files and used many different bitbucketconverter scripts. These were some of the results:

###### Processing line {"RfRaw":{"Data":"AA B1 05 091A 065E 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
 ######
Filtered 0xB1 data: AAB105091A065E014A02B215CCC092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B255
Resulting 0xB0 data: AA B0 36 05 08 091A 065E 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55
###### Processing line {"RfRaw":{"Data":"AA B1 05 091A 0668 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}}
 ######
Filtered 0xB1 data: AAB105091A0668014A02B215CCC092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B255
Resulting 0xB0 data: AA B0 36 05 08 091A 0668 014A 02B2 15CC C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55
###### Processing line {"RfRaw":{"Data":"AA B1 05 0910 0668 014A 02B2 15EA C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55"}} ######
Filtered 0xB1 data: AAB10509100668014A02B215EAC092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B255
Resulting 0xB0 data: AA B0 36 05 08 0910 0668 014A 02B2 15EA C092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B2 55

typed `rfraw AAB036050809100668014A02B215EAC092A3A3B2B2A3A3B2A3B2B2A3B2B2A3B2A3B2A3A3A3A3A3B2B2B2A3A3A3A3A3B2B2A3A3A3B2B2A3B2B255` got no response. Any suggestions?

derFrickler, 2020-10-13:
Did you find any solution to your problem? 
struggling with the same remote.

prbtt, 2020-12-31:
> Did you find any solution to your problem?
> struggling with the same remote.

Nope, still stuck. Might have to buy a new remote afterall...

rrlevy, 2021-01-19:
Any updates?

I'm also trying to copy codes from a remote that control my blinds. The remote (AC114-01B) already had a history of working with tasmota, and it also doesn't work. 

[I posted about it here at the issue regarding the specific remote](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/148)

I also don't know if it is an issue regarding a recent firmware version, or if I'm missing something. I did the same procedure as you did for my blinds/remote.

jakass202, 2021-02-21:
@prbtt - I have some success with ERB25 Tubular Motors and a BF-315 remote.
I found using RfRaw 177 and capturing ALOT of button presses and the pushing it through the BitBuckterConverter with send to device option (via Sonoff Rf Bridge), as and when a B0 was calculated you could then test it immediately. If it didn't work move to next convert code.
 I was able to get some working B0 codes - it did take a while.
 I can successfully send Backlog RfRaw "B0 Result"; RfRaw 0 and the blinds do as instructed.
 
 I was trying to see if a Protocol could be made up but as yet not figured out Sync Buckets etc as per Wiki
 
 #if EFM8BB1_SUPPORT_BOFU_ERB25_PROTOCOL == 1
#define Bofu_Erb25
//SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(Bofu_Erb25)[], static uint16_t, SI_SEG_CODE) = { 2460, 1640, 320, 640, 5580 }; //MOST USED ALL
//SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(Bofu_Erb25)[], static uint16_t, SI_SEG_CODE) = { 2470, 1630, 320, 650, 5570 }; //AVERAGES
SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(Bofu_Erb25)[], static uint16_t, SI_SEG_CODE) = { 2470, 1630, 320, 640, 5580 }; //MOST USED IND
//SI_SEGMENT_VARIABLE(PROTOCOL_START(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(4), LOW(0), HIGH(1), LOW(2) }; //nope
SI_SEGMENT_VARIABLE(PROTOCOL_START(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(4), LOW(0), HIGH(1), LOW(2) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(3), LOW(2) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(2), LOW(3) };
//SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(2), LOW(3) };
//SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(3), LOW(2) };
//SI_SEGMENT_VARIABLE(PROTOCOL_END(Bofu_Erb25)[], static uint8_t, SI_SEG_CODE) = { HIGH(4), LOW(0) };
#endif

jakass202, 2021-02-21:
I got a semi working protocol;

For the Rf_Config.h
`// more protocols, enable here!                 Enable  Remarks
#define EFM8BB1_SUPPORT_ERB25_1_PROTOCOL		1		//`

For the RF_Protocols.h
`#if EFM8BB1_SUPPORT_ERB25_1_PROTOCOL == 1
#define ERB25_1
SI_SEGMENT_VARIABLE(PROTOCOL_BUCKETS(ERB25_1)[], static uint16_t, SI_SEG_CODE) = { 2464, 1630, 321, 645, 5536 };
SI_SEGMENT_VARIABLE(PROTOCOL_START(ERB25_1)[], static uint8_t, SI_SEG_CODE) = { HIGH(1), LOW(2) };
SI_SEGMENT_VARIABLE(PROTOCOL_BIT0(ERB25_1)[], static uint8_t, SI_SEG_CODE) = { HIGH(3), LOW(2) } ;
SI_SEGMENT_VARIABLE(PROTOCOL_BIT1(ERB25_1)[], static uint8_t, SI_SEG_CODE) = { HIGH(2), LOW(3) };
//SI_SEGMENT_VARIABLE(PROTOCOL_END(ERB25_1)[], static uint8_t, SI_SEG_CODE) = { LOW(0) };
#endif

#if EFM8BB1_SUPPORT_ERB25_1_PROTOCOL == 1
		{
			{ &PROTOCOL_BUCKETS(ERB25_1), ARRAY_LENGTH(PROTOCOL_BUCKETS(ERB25_1)) },
			{ &PROTOCOL_START(ERB25_1), ARRAY_LENGTH(PROTOCOL_START(ERB25_1)) },
			{ &PROTOCOL_BIT0(ERB25_1), ARRAY_LENGTH(PROTOCOL_BIT0(ERB25_1)) },
			{ &PROTOCOL_BIT1(ERB25_1), ARRAY_LENGTH(PROTOCOL_BIT1(ERB25_1)) },
			//{ &PROTOCOL_END(ERB25_1), ARRAY_LENGTH(PROTOCOL_END(ERB25_1)) },
			{ NULL, 0 },
			40
		},
#endif
`

The RfRaw 166 picks up the results for the STOP and DOWN commands however no UP commands show on results.
The LED however does Blink, so I think its able to decode but no output result is printed.

Here the the stop command via RfRaw 177
{"RfRaw":{"Data":"AA B1 05 09A6 064A 014A 0294 157C C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 014A 028A 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0140 0280 15B8 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09B0 0654 014A 0280 159A C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09B0 0654 014A 0294 1586 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09B0 0654 014A 0294 159A C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0654 0140 028A 15B8 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 0140 028A 15B8 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0140 028A 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 0654 014A 028A 15A4 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 028A 15A4 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09B0 0654 014A 0294 1586 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 0294 159A C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 0294 15A4 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0140 028A 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
Successful Decode
Via RfRaw 166 RESULT = {"Time":"2021-02-21T22:50:27","RfRaw":{"Data":"AAA6060192F8057F2F55"}}

Here the the Down command via RfRaw 177
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 028A 15AE C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0140 028A 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 0140 028A 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0136 0280 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0140 0280 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0136 0280 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 028A 15A4 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0136 0280 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 0654 0154 0294 1572 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0136 0280 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0140 0280 15CC C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 0668 0136 0280 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3A3A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3B2A3 55"}
[... cut ...]
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 0294 159A C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 09A6 065E 014A 0294 15A4 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
{"RfRaw":{"Data":"AA B1 05 099C 065E 0140 028A 15C2 C092A3B2B2A3B2B2A3B2A3A3A3A3A3B2B2B2B2B2B2B2B2A3B2A3B2A3A3A3A3A3A3A3B2B2A3B2A3A3A3A3 55"}
RfRaw 166 shows the Sonoff RF Bridge blinking however no RESULT
Only difference I can see visible is that the STOP/DOWN commands end in a 2 for RF Data and UP is a 3.

prbtt, 2022-02-14:
I have given up on trying get this to work through my RF Bridge. Luckly there was an alterative way to integrate this tubular motor into home assistant. I've soldered the UP and DOWN pins of the remote controller to GPIO pins of an ESP8266 device with tasmota, so I can trigger them in home assistant. There are better ways to do this with esphome.
If anyone is interested, here's some usefull info:
[image]
[image]
[image]
[image]
[image]