Portisch/RF-Bridge-EFM8BB1 #166: BitBucketConverter.py python3 error

alekob, 2020-05-04:
Hello,
running the latest version of BitBucketConverter.py on a Ubuntu 18.04 (Python 3.6.9) fails

Traceback (most recent call last):
  File "./BitBucketConverter.py", line 335, in <module>
    strInput = getInputStr()
  File "./BitBucketConverter.py", line 64, in getInputStr
    auxStr = raw_input("Enter B1 line: ")
NameError: name 'raw_input' is not defined

pgScorpio, 2020-07-09:
Same here on Python 3.8.1

Traceback (most recent call last):
  File "BitBucketConverter.py", line 335, in <module>
    strInput = getInputStr()
  File "BitBucketConverter.py", line 64, in getInputStr
    auxStr = raw_input("Enter B1 line: ")
NameError: name 'raw_input' is not defined

MrRimmer, 2021-08-30:
I'm a complete python novice, but got it working by replacing raw_input() with input() - raw_input is deprecated in Python 3.