Portisch/RF-Bridge-EFM8BB1 #167: Issue with Aldi Quigg DMV-7009S RF-sockets

Stefan291, 2020-05-05:
Hello,

I have old (year 2005) RF-socket from Aldi /Quigg type DMV-7009S and I use Tasmota 8.2.0 and the latest Portisch-Release "RF-Bridge-EFM8BB1-20181127.hex". So far everything works good. I activate sniffing by "RfRaw AAB155" and I see the rf-codes when I presse a button on my remote control.

But I can only switch the socket off, not on. I try it with the raw codes and the converted codes but every time I send the codes to the socket, I can´t switch it on, Only when it is on, I can switch it off.

For example, this should be the command to switch the socket on:
Backlog RfRaw AA B0 60 05 08 0320 01C2 05B4 1F80 B0E4 012121212121212120202020200000000000000003012121212121212120202020200000000000000003012121212121212120202020200000000000000003012121212121212120202020200000000000000003 55; RfRaw 0

Here are some raw codes I catched in sniffing mode "RfRaw AAB155":

Button 1 - on:
"2020-05-05T14:20:24","RfRaw":{"Data":"AA B1 05 0320 01C2 05B4 1F58 B09E 012121212121212120202020202000020020000203012121212121202020202020202000000020000003012121212121202020202020202020000020000003012121212121202020202020202020000020000004 55"}}
14:20:26 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:26","RfRaw":{"Data":"AA B1 05 0320 01C2 05AA 1F4E B09E 012121212121212120202020202000000020000003012121212121202020202020202000000020000003012121212121202020202020202000000020000003012121212121202020202020202020000020000004 55"}}
14:20:28 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:28","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F58 B0A8 012121212121212120202000000000000000000003012121212121202020202020200000000000000003012121212121212020202020200000000000000003012121212121212020202020200000000000000004 55"}}
14:20:30 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:30","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F4E B09E 012121212121212120202000200000000000000003012121212121202020202020200000000000000003012121212121212020202020200000000000000003012121212121212020202020200000000000000004 55"}}
14:20:32 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:32","RfRaw":{"Data":"AA B1 05 0320 01C2 05AA 1F4E B09E 012121212121212120202020202000000020000203012121212121202020202020202000000020000003012121212121202020202020202000000020000003012121212121202020202020202020000020000004 55"}}
14:20:34 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:34","RfRaw":{"Data":"AA B1 05 0334 01B8 05AA 1F58 B09E 012121212121212120202000000000000000000003012121212121212020202020200000000000000003012121212121212120202020200000000000000003012121212121212020202020200000000000000004 55"}}
14:20:36 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:36","RfRaw":{"Data":"AA B1 05 032A 01C2 05AA 1F4E B09E 012121212121212120202020000000000000000003012121212121202020202020200000000000000003012121212121212020202020200000000000000003012121212121202020202020200000000000000004 55"}}
14:20:39 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:39","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F58 B0A8 012121212121212120202020200000000000000003012121212121212020202020200000000000000003012121212121202020202020200000000000000003012121212121212020202020200000000000000004 55"}}
14:20:41 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:41","RfRaw":{"Data":"AA B1 05 0320 01C2 05B4 1F58 B0B2 012121212121212120202020202000000000000003012121212121202020202020202000000020200203012121212121202020202020202020000020000003012121212121202020202020202000000020000004 55"}}
14:20:45 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:45","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F62 B0B2 012121212121212120202020200000000000000003012121212121212020202020200000000000000003012121212121212020202020200000000000000003012121212121202020202020200000000000000004 55"}}
14:20:47 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:47","RfRaw":{"Data":"AA B1 05 0320 01C2 05AA 1F58 B0A8 012121212121212120202020202000000020000203012121212121202020202020202000000020000203012121212121202020202020202000000020000003012121212121202020202020202020000020000004 55"}}
14:20:50 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:50","RfRaw":{"Data":"AA B1 05 0320 01C2 05B4 1F58 B0A8 012121212121212120202020202000000020000203012121212121202020202020202000000020000003012121212121202020202020202000000020000003012121212121202020202020202000000020000004 55"}}
14:20:52 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:20:52","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F58 B0A8 012121212121212120202020000000000000000003012121212121212120202020202000000000000003012121212121212120202020200000000000000003012121212121212120202020200000000000000004 55"}}

Button 1 - off:
14:21:29 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:29","RfRaw":{"Data":"AA B1 05 0320 01C2 05AA 1F62 B094 01212121212121212020202020202020202020202301212121212120202202020202020202020202023 012121212121202020202020202020202020202023 012121212121202020202020202020202020202024F 55"}}
14:21:31 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:31","RfRaw":{"Data":"AA B1 05 032A 01B8 05B4 1F62 B094 012121212121212120202020200000000000000003012121212121212020202020202020202000000003012121212121212120202020202020200000000003012121212121212020202020202020200020000004 55"}}
14:21:33 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:33","RfRaw":{"Data":"AA B1 05 0320 01C2 05B4 1F6C B094 012121212121212120202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202024 55"}}
14:21:35 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:35","RfRaw":{"Data":"AA B1 05 032A 01C2 05AA 1F62 B094 012121212121212120202020200000000000000003012121212121212020202020202020000000000003012121212121202120202020202020000000000003012121212121212020202020202020000000000004 55"}}
14:21:37 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:37","RfRaw":{"Data":"AA B1 05 0320 01C2 05AA 1F62 B094 012121212121212120202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202024 55"}}
14:21:38 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:38","RfRaw":{"Data":"AA B1 05 032A 01C2 05AA 1F62 B094 012121212121212120202020202000000000000003012121212121202020202020202020202000000003012121212121212020202020202020202020200003012121212121202020202020202020200020000004 55"}}
14:21:40 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:40","RfRaw":{"Data":"AA B1 05 0320 01C2 05B4 1F6C B094 012121212121212120202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202024 55"}}
14:21:42 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:42","RfRaw":{"Data":"AA B1 05 0320 01B8 05B4 1F62 B09E 012121212121212120202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202023012121212121202020202020202020202020202024 55"}}
14:21:46 RSL: tele/tasmota_B2EE64/RESULT = {"Time":"2020-05-05T14:21:46","RfRaw":{"Data":"AA B1 05 032A 01C2 05AA 1F62 B09E 012121212121212120202020202020000000000003012121212121212020202020202020202000000003012121212121212020202020202020200000000003012121212121202020202020202020200020000004 55"}}

Thanks in advance for your help!

Regards

Stefan

Sepp0815, 2020-05-06:
Hi Stefan,
I have a Gobaltronics GT7000 from Aldi. And I also had problems with getting the plugs to run with the sniffed codes. As far as I remember, I finally modified the RF_Protocols.h and compiled the project to get a new hex-File. Modification was made in 2018 or so. So I don't remember all the details of the action at the moment.
Found some information about the protocoll here: [https://manual.pilight.org/protocols/433.92/switch/quigg_7000.html](url)
But I have no idea whether your DMV-7009 has the same protocol as the 7000.

Kind regards
Sepp

Stefan291, 2020-05-06:
Hello Sepp,

thanks for your help and for the information. I found the manual on pilight.org too but I have no idea when I compare the raw sniffed codes with this manual.
Do you still have your modified and compiled hex-file? I think you are right and the protocol should be the same. It would be grat to test this hex-file.

Kind regards

Stefan

Sepp0815, 2020-05-07:
Hi Stefan,

I'm not shure which one of the two attached hexfiles was finally used. 

Good luck.

[Hexfiles.zip](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/4594807/Hexfiles.zip)

Stefan291, 2020-05-07:
Hi Sepp,

thank you very much for the hex-files. It seems that the file "RF-Bridge-EFM8BB1.hex" work basically, but it is more like a beta. Sometimes I can switch on and off my device with:
ON:
Backlog RfRaw AAA8040EBFF11055; RfRaw 0
OFF:
Backlog RfRaw AAA8040EBFF00055; RfRaw 0

And sometimes it doesn´t work. And I can´t switch on a second device.

So no luck, but nevermind.

Kind regards

Stefan

Sepp0815, 2020-05-08:
Hi Stefan,
here's the code code that I use with first socket:
AAA80401B0011055 on
AAA80401B0000055 off

And here's my definition in RF_Protocols.h
```
 		/*
			 * Quigg GT7000
			  */
			{
				//???,		// SYNC_LOW
				//1,		// SYNC_HIGH
				{  78, 1 },	// SYNC HIGH_LOW
				0,			// SYNC_BIT_COUNT
				670,		// PULSE_TIME
				{  1,  2 },	// BIT0 HIGH_LOW
				{  2,  1 },	// BIT1 HIGH_LOW
				20,			// BIT_COUNT
				60,			// TOLERANCE
				80,			// REPEAT_DELAY
				true		// INVERSE
			},

```
The pulse time I used is 670 instead of nominal 700. I remember that I grabed the signal of my transmitter and this was the exact timing that I could measure.
If your system works with slightly different timing, this could explain why it sometimes works or not.
Maybe using other timing value in RF_Protocols.h and compiling the project could solve the problem. 

Sepp

Stefan291, 2020-05-08:
Hi Sepp,

okay but this goes beyond my limits. I will replace the old sockets and buy others.

But thank you very much for your kind help!

Kind regards

Stefan

closed 2020-05-08 by Stefan291

m7thon, 2020-06-27:
Hi Sepp. I have Quigg7000 sockets, too. I got as far as decoding the signal and being able to control them using raw "B0" commands. According to [this](https://github.com/d-a-n/433-codes/blob/master/database.md), your timing of 670 for the pulse width is correct (even though I measured 700, 1400 timings using `rfraw 177`).

In any case, I tried your firmware `.hex` and it works. I would have liked to compile this myself, but I couldn't even get the Simplicity Studio 4 downloaded :-(.

Is there any advantage using a defined protocol over using raw bucket sending?