Portisch/RF-Bridge-EFM8BB1 #171: Sending the same B0 code only works SOMETIMES (simple RF socket)

JulianNaumann, 2020-06-09:
### Latest firmware used?
Tasmota reports
´´´00:40:24 MQT: tele/sonoff/RESULT = {"Time":"2020-06-10T00:40:24","RfRaw":{"Data":"AA0355"}}´´´ == v03
(kinda strange, I used the latest version from ./releases/ , so it should be 04 instead? Was the version string updated in the firmware?!)

### Information on RF device
Trying to control a "Bearware indoor remote socket"

### Sniffed data
B1 Code
```AA B1 04 0136 047E 00B4 227E 3818282818281818282818182818281828182828181828281828182818281818281828182818281828182818281828182818282818281818281828182818281828 55```

B0 Code (converted with BitBucketConverter.py)
```AA B0 4B 04 08 0140 047E 00B4 227E 3818282818281818282818182818281828182828181828281828182818281818281828182818281828182818281828182818282818281818281828182818281828 55

So sending the B0 code only works SOMETIMES. The command always gets acknowledged (AAA055 code), the red LED on the sonoff bridge blinks (to show RF data was transmitted). If it works, it works on first try and keeps working flawlessly. But just sometimes, it does not work, and continues to not work...
As it is a simple remote socket the code should not change. When switching the socket on through the sonoff does not work, pressing the button on the original RF remote always works right away.

Anybody got any ideas??

Arganox, 2021-12-27:
I've got similar issues with a simple doorbell. Anyone still on this project?

JulianNaumann, 2021-12-28:
I got rid of my Sonoff bridge as nobody seems to be active around this project anymore. I got a Broadlink RM Pro, can recommend.

Arganox, 2021-12-29:
@JulianNaumann I managed to code a custom protocol for my extractor. But the project does indeed seems to be dead.

I'll have a look at the Broadlink RM Pro as well. Looks like it can send IR as well. Very nice!
Thanks for the pointer!