Portisch/RF-Bridge-EFM8BB1 #174: Proposal for moving decoder up into tasmota firmware.

kenthinson, 2020-07-21:
The EFM8BB1 being limited to 8KB means we are limited to a handful of protocols. the latest release being 7.76KB seems the limit it at hand. 

What do you devs think about moving the protocol decoding up to the esp? just send the raw data over and process there?

haegar33, 2020-08-12:
The software has not been updated for over a year and probably reached an end point. It would be to some other programmer to implement this good idea but I have no clue whether this is feasible.

halfbakery, 2020-10-03:
I'm working on feeding raw sniffing data into the rtl_433 decoder, see [this PR](https://github.com/merbanan/rtl_433/pull/1491).

thermseekr, 2020-10-03:
This would be awesome 👏🏻👏🏻👏🏻👏🏻👏🏻

Enviado do meu iPhone

halfbakery, 2020-10-04:
Here's my branch with my new bucket sniffer (firmware included under the Keil directory): https://github.com/halfbakery/RF-Bridge-EFM8BB1/tree/rebaked_sniffer